circuitikz/circuitikz #655: [Feature] Arrow heads LEDs

myzinsky, 2022-10-16:
LEDs have a serif arrow head. Would it be possible to change this to a 60 degree straight arrow head?

Rmano, 2022-10-16:
This is, indeed, a nice idea. 

In the following code, I am patching `circuitikz` to use the generic arrow selectors (see page 52 of the current manual) to create a new arrow class "opto" that you can use to tweak the arrows in LEDs, photodiodes, and laser diodes.  I need to add support for the phototransistor, though, before releasing, if at all possible (transistors code is ancient, and I fear a bit touching it).

You can use this code for now, but the mechanism will be in the next release out of the box (and then you can remove the code between `makeatletter` and `makeatother`). 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{etoolbox}

\makeatletter
\pgf@circ@declare@family@arrows{opto}
\def\pgf@circ@draw@pdarrows{%
    \pgfsetlinewidth{\pgfstartlinewidth}
    \pgf@circ@fill@strokecolor
    \pgfcirc@set@arrows{opto}{}{latexslim}
    % \pgfsetarrowsend{latexslim}
    \ifpgf@pd@fliparrows
        \pgfpathmoveto{\pgfpoint{-0.6\pgf@circ@res@right}{1.8\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{0pt}{0.8\pgf@circ@res@up}}
        \pgfusepath{draw}
        \pgfpathmoveto{\pgfpoint{0pt}{1.6\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{0.6\pgf@circ@res@right}{0.6\pgf@circ@res@up}}
        \pgfusepath{draw}
    \else
        \pgfpathmoveto{\pgfpoint{0.6\pgf@circ@res@right}{2\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{-0.4\pgf@circ@res@right}{\pgf@circ@res@up}}
        \pgfusepath{draw}
        \pgfpathmoveto{\pgfpoint{1.2\pgf@circ@res@right}{1.8\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{0.2\pgf@circ@res@right}{0.8\pgf@circ@res@up}}
        \pgfusepath{draw}
    \fi
}
\patchcmd\pgf@circ@draw@ledarrows
    {\pgfsetarrowsend{latexslim}}
    {\pgfcirc@set@arrows{opto}{}{latexslim}}
    {\typeout{patching LED arrows ok}}
    {\typeout{patching LED arrows FAILED}}
\patchcmd\pgf@circ@draw@laserarrows
    {\pgfsetarrowsend{latexslim}}
    {\pgfcirc@set@arrows{opto}{}{latexslim}}
    {\typeout{patching LED arrows ok}}
    {\typeout{patching LED arrows FAILED}}
\makeatother
\begin{document}
\begin{tikzpicture}[]
    \begin{scope}
        \draw (0,2) to[leD] ++(2,0) to[pD] ++(2,0) to[lasD] ++(2,0);
        \ctikzset{led arrows from cathode}
        \ctikzset{pd arrows to cathode}
        \draw (0,0) to[leD] ++(2,0) to[pD] ++(2,0) to[lasD] ++(2,0);
    \end{scope}
    \begin{scope}[color=blue, yshift=-4cm]
        \ctikzset{opto end arrow={Triangle}}
        \ctikzset{opto start arrow={Hooks[red]}}
        \draw (0,2) to[leD] ++(2,0) to[pD] ++(2,0) to[lasD] ++(2,0);
        \ctikzset{led arrows from cathode}
        \ctikzset{pd arrows to cathode}
        \draw (0,0) to[leD] ++(2,0) to[pD] ++(2,0) to[lasD] ++(2,0);
    \end{scope}

\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2022-10-17:
[image]

myzinsky, 2022-11-08:
[image]
How about other components like vR and photoresistor?

It would also be nice to change the arrow heads there.

Rmano, 2022-11-09:
In `vR` you can already.

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}[european]
    \draw (0,0) to[vR] ++(2,0) 
        to[vR, tunable end arrow={Triangle[]}] ++(2,0);
\end{tikzpicture}
\end{document}
```

[image]

...but I think I missed the photoresistor, will look at it!

myzinsky, 2022-11-09:
Then also look for wiper pR :-D

Rmano, 2022-11-09:
> Then also look for wiper pR :-D

I don't understand, sorry --- you can change the arrow of the wiper with `wiper end arrow`, ¿no? 

```
\draw (0,2) to[pR] ++(2,0)
        to[pR, wiper end arrow={Triangle[]}] ++(2,0);
```

[image]

Rmano, 2022-11-09:
> ...but I think I missed the photoresistor, will look at it!

It results that the patch is quite simple. I'll commit as soon as I can! (Thanks for noticing) 

```
diff --git a/tex/pgfcircbipoles.tex b/tex/pgfcircbipoles.tex
index 1c87505..d40a432 100644
--- a/tex/pgfcircbipoles.tex
+++ b/tex/pgfcircbipoles.tex
@@ -365,7 +365,7 @@
     \endpgfscope
 
     \pgfscope
-        \pgfsetarrowsend{latexslim}
+        \pgf@circ@set@optoarrow@style
         \pgfpathmoveto{\pgfpoint{.7\pgf@circ@res@right}{\pgf@circ@res@up}}
         \pgfpathlineto{\pgfpoint{.3\pgf@circ@res@right}{-1.2\pgf@circ@res@down}}
         \pgfusepath{draw}
```

commit beba71ba (2022-10-16): Added styling for the "opto" arrows

Now you can change the tips of LEDs, photodiodes, laser diodes
and phototransistors.

Thanks to @ for the idea.
Fixes #655
commit 06eef7f2 (2022-10-16): Added styling for the "opto" arrows

Now you can change the tips of LEDs, photodiodes, laser diodes
and phototransistors.

Thanks to @myzinsky for the idea.
Fixes #655
PR #656 (2022-10-16): Added styling for the "opto" arrows
Now you can change the tips of LEDs, photodiodes, laser diodes and phototransistors.

Thanks to @myzinsky for the idea.
Fixes #655

[image]
PR #658 (2022-11-09): Apply the "opto" arrow style to photoresistors too.
Noticed by @myzinsky in #655
Fixes #655

[image]
closed 2022-11-09 by Rmano