circuitikz/circuitikz #660: [Bug] Varistor caption cut-off

myzinsky, 2022-11-12:
The varistor image gets cut-off in standalone mode:

[image]

```latex
\documentclass[convert=false]{standalone}
\usepackage[straightvoltages, europeanresistors, european inductor]{circuitikz}
\usepackage{tikz}

\begin{document}
    \begin{circuitikz}[european]
        \draw (0,0) to [varistor] (2,0);
    \end{circuitikz}
\end{document}
```

Rmano, 2022-11-12:
Yes, external labels are not in the component bounding box. This is normal, it happens with more or less all the "internal" text label of `circuititz` (transistors text, for example). 

Normally in complete circuits is not a problem, and it's better not to put the labels into the geographical coordinates to maintain alignment with components with no or different labels. 

If you need the label into the bounding box, just add it manually, like in

```latex
\documentclass[convert=false]{standalone}
\usepackage[straightvoltages, europeanresistors, european inductor]{circuitikz}
\usepackage{tikz}
\begin{document}
    \begin{circuitikz}[european]
        \draw (0,0) to [sR, name=myS] (2,0);
        \node [below left] at(myS.label) {\tiny U};
    \end{circuitikz}
\end{document}
```

[image]

Rmano, 2022-11-12:
Or you can manually override the bounding box (this trick is useful in several occasions)

```latex
\documentclass[convert=false]{standalone}
\usepackage[straightvoltages, europeanresistors, european inductor]{circuitikz}
\usepackage{tikz}

\begin{document}
    \begin{circuitikz}[european]
        \path [use as bounding box](0,-0.5) rectangle (2.0,0.5);
        \draw (0,0) to [varistor] (2,0);
    \end{circuitikz}
\end{document}
```

[image]

closed 2022-11-12 by Rmano