circuitikz/circuitikz #663: [Feature] Anchorpoint for Variable Resistors

myzinsky, 2022-11-13:
It would be also nice to have an anchor point at the beginning of the arrow, like here: 

[image]

The pR hast this, called wiper. But the vR has not ...

Rmano, 2022-11-13:
Yes, this could be a useful addition. In the meanwhile, you can always add the arrow by hand... ;-)

Rmano, 2022-11-13:
BTW, the reason the potentiometer has the wiper contact and the variable resistance has not, is because in the first case that is an *electrical* connection, and in the latter it isn't. But the switches have a mid-switch anchor, which is a mechanical one, so... why not? I'll look into it.

myzinsky, 2022-11-13:
But in many documents you can also see this: 

[image]

Therefore, it would be nice to have it :-)

Rmano, 2022-11-13:
I admit I really can't understand that circuit... I suppose P1 is a potentiometer, not a variable resistor. A varistor is, for what I know, a two-terminal device with a knob or a screw head or whatever to change the resistance. I would use a `pR` there, to remove any ambiguity.

myzinsky, 2022-11-13:
It's often used in Germany:

[image]

[image]

Rmano, 2022-11-13:
Nice to know. I will add the anchor(s) when I can... (In the middle of a paper submission now, sorry...)

PR #669 (2022-11-16): Added `wiper` and `tip` anchors to variable bipoles.
Fixes #663

[image]

[image]
commit fce00048 (2022-11-16): Added `wiper` and `tip` anchors to variable bipoles.

Fixes #663
closed 2022-11-16 by Rmano