circuitikz/circuitikz #664: [Feature] Longer Transmission Lines

myzinsky, 2022-11-13:
It would be nice to extend the length of a transmission line like in the following picture:

[image]

Rmano, 2022-11-13:
You basically can enlarge them, but the shape is deformed. 

```latex
\documentclass[margin=2.178mm]{standalone}
\usepackage[RPvoltages]{circuitikz}
\usepackage{tikz}

\begin{document}
    \begin{circuitikz}[european]
        \draw (0,0) to[TL, bipoles/tline/width=2, l=TL\textsubscript{5}] ++(3,0);
    \end{circuitikz}
\end{document}
```

[image]

it could probably be fixable, but I am not sure if to want to touch that part of the code. It's an old, legacy one: my personal opinion is that this symbol is not a schematic one, it is more a drawing. An RF schematic is found in the manual, page 122.

[image]

myzinsky, 2022-11-13:
Yeah but I think this RF is more for a pcb view... Would be really nice to have a line like this! 

I found this: https://tex.stackexchange.com/questions/402631/draw-transmission-line-circuits maybe it helps? Sorry I'm just a TeX user and not a TeX-Programmer :-D

Rmano, 2022-11-13:
There is a standard cylinder shape in Ti*k*Z: https://tikz.dev/library-shapes#pgf.cylinder

myzinsky, 2022-11-13:
Nice, unfortunately, it does not have a connector in the middle of the circle :-( 

[image]

Rmano, 2022-11-13:
Ah, but that would be just `($(s.before top)!0.5!(s.after top)$)` - you can put that in a macro or in a sub circuit.

Rmano, 2022-11-14:
...anyway the deformation when changing the width in TL could be thought as a (veeeeery long standing) bug, I'll have a look.

closed 2022-11-16 by Rmano

afcuttin, 2024-04-04:
thanks for the fix!
Nonetheless, the approach of https://tex.stackexchange.com/a/527729/177 would be more preferable, because it stretches the transmission line automatically, rahter than requiring one to adjust bipoles/tline/width with an absolute value.

PR #666 (2022-11-16): Fix shape for long TL component
Fixes #664

[image]

```latex
\documentclass[margin=2.178mm]{standalone}
\usepackage[RPvoltages]{circuitikz}
\usepackage{tikz}

\begin{document}
    \begin{circuitikz}[european]
        \draw (0,2) to[TL, bipoles/tline/width=0.2, l=TL\textsubscript{0.2}] ++(3,0);
        \draw (0,1) to[TL, l=TL] ++(3,0);
        \draw (0,0) to[TL, bipoles/tline/width=2, l=TL\textsubscript{2}] ++(3,0);
    \end{circuitikz}
\end{document}
```
commit 3d58349f (2022-11-16): Fix shape for long TL component

Fixes #664