circuitikz/circuitikz #665: More examples of IC based circuits? like IC pin layout, breadboard diagrams

yashpalgoyal1304, 2022-11-14:
### Current:
I want to be able to do:
* draw IC pin diagrams - quite different task than directly using ICs, example: <br/>
[![icpl image]][ref icpl link]
* draw breadboard diagrams, or atleast use a breadboard image, and overlay the CircuitTi*k*z diagram over it, example: <br/>
[[image]][ref bbd link]

[icpl image]: https://user-images.githubusercontent.com/19423063/201618559-8a6cd53f-ba0b-4d46-a56e-46207b3a2237.png
[ref icpl link]: https://duckduckgo.com/?q=pin+layout+for+74x00+in+latex&t=vivaldi&iax=images&ia=images&iai=http%3A%2F%2F3.bp.blogspot.com%2F-hLlu_bSlrbU%2FTxmzCwAueAI%2FAAAAAAAAAIk%2FWZTKQHjcBdQ%2Fs1600%2F74ls00.jpg
[bbd image1]: https://user-images.githubusercontent.com/19423063/201621490-634c1542-430a-493d-8d20-786c8aee155d.png
[ref bbd link]: http://www.mathworks.com/help/examples/arduino/win64/xxarduinomega2560_pid_connections1.png

### Background:
* I don't have any prior experience with Ti*k*z or with CircuiTi*k*z, 
* and I can't get enough time to like fully grasp the gist of things 
<details><summary>* however i have read following things (<i>click me</i>) </summary>

* i have read the instructions at following 2-3 times
    + 2.4 Tutorials/a logic circuit
    + 4.23 Flip-Flops
    + 4.25 Chips (Integrated Circuits)
* i have read the instructions at following 1-2 times
    + 3.2 node-style components
    + 3.3 styling circuits and components
    + all the 3 Ti*k*z tutorials mentioned: 2. Picture for Karl's Student, 13 Specifying coordinates, 17.2 Nodes and their shapes

</details>

Rmano, 2022-11-14:
I found the code, pasting here with no comments...

```latex
documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{mathpazo}
\usetikzlibrary{shapes.symbols, backgrounds, fit}
\usepackage{stackengine}
\DeclareSIUnit{\uF}{\micro\farad}
\DeclareSIUnit{\nF}{\nano\farad}
\ctikzsetstyle{romano}
\begin{document}

\begin{tikzpicture}[]
        \draw (0,0) node[dipchip, rotate=90, anchor=pin 8, circuitikz/chips/scale=1.5,
            no topmark, draw only pins={1,2,4,5,7,8}](AD835){}
            (AD835.north) node[above, rotate=90]{\texttt{AD835ANZ}};
        \begin{scope}[on background layer={color=gray}]
            \node [mixer, circuitikz/blocks/scale=0.7](M) at([xshift=-0.3cm]AD835.center) {};
            % 2.25 is 1.5*1.5 (chip scale * NL/2
            \node [muxdemux, muxdemux def={Lh=2.25, NL=3, Rh=0.8, NR=1, w=0.8, NB=1},
                external pins width=0, rotate=-90, anchor=blpin 3](DX)
                at ([yshift=-3mm]AD835.bpin 8){};
            \node [font=\tiny, below=-1mm] at (DX.blpin 3) {$+$};
            \draw (DX.blpin 3) -- (AD835.bpin 8) (DX.blpin 1) -- (AD835.bpin 7);
            \node [font=\tiny, below=-1mm] at (DX.blpin 1) {$-$};
            \node [muxdemux, muxdemux def={Lh=2.25, NL=3, Rh=0.8, NR=1, w=0.8, NB=1},
                external pins width=0, rotate=90, anchor=blpin 1](DY)
                at ([yshift=3mm]AD835.bpin 1){};
            \node [font=\tiny, above=-1mm] at (DY.blpin 1) {$+$};
            \node [font=\tiny, above=-1mm] at (DY.blpin 3) {$-$};
            \draw (DY.blpin 1) -- (AD835.bpin 1) (DY.blpin 3) -- (AD835.bpin 2);
            \node [muxdemux, muxdemux def={Lh=3, NL=3, Rh=1, NR=1, w=1, NB=1}, external pins width=0](A)
                at([xshift=0.7cm]AD835.center) {};
            \draw (DX.brpin 1) -- ++(0,-0.2) -- (M) (DY.brpin 1) -- ++(0,0.2) -- (M) (M.e) -- (A.left)
                (A.right) -| (AD835.bpin 5) (A.bbpin 1) -- ++(0,-0.2) -| (AD835.bpin 4);
        \end{scope}
    \end{tikzpicture}
\end{document}
```
[image]

Rmano, 2022-11-14:
I'll add the previous example to the manual, reopening as a documentation issue.

yashpalgoyal1304, 2022-11-14:
> For the second problem you can start from here: https://tex.stackexchange.com/questions/493239/alignment-of-dip-chips-in-circuitikz

> I found the code, pasting here with no comments...

> I'll add the previous example to the manual, reopening as a documentation issue.

wow, thanks a lot for these

> but you can ask in https://tex.stackexchange.com/questions/tagged/circuitikz 

* i tried there, found that to be pretty outdated: \
[8-pinned-ic-using-circuit-tikz](https://tex.stackexchange.com/questions/416631/8-pinned-ic-using-circuit-tikz) - see, the answer doesn't even contain the `dipchip` part
* while starting with LaTeX some months ago too, SO was severely lacking the quality and elegance.
* only thing helped was the wikibook on LaTeX combined with manuals

----
Continued:

* any book recommendation (preferably gratis and offline digital copy) for learning Ti*k*z & CircuiTi*k*z? As im-humble-o, books cover things in proper depth - the sweet middle ground between a tutorial and manual, and in more width (as in how to put stuff together) than both of these.
* [OPTIONAL] any thoughts on why the breadboard component is better not included?

yashpalgoyal1304, 2022-11-14:
I am making some progress using the example u provided. thanks a lot for that.

do u mind me posting my investigations on that here?

yashpalgoyal1304, 2022-11-14:
## Investigations:

### To-Do:
- [x] overlaying the circuit on the chip
- [ ] Offsetting the pin labels to move them either outwards (preferably on the pin pads like in the IC manuals), or atleast off-centering

### Input:
1. Overlaying the circuit on the chip, on inspecting/isolating the code share above, is done just using the `tikzlibrary` and `scope` environment:

```latex
\usetikzlibrary{shapes.symbols, backgrounds, fit} % Have not tried to minimize further
...

\begin{tikzpicture}
    ...
    \begin{scope}[on background layer]
        ... <overlaying circuit content goes here>
    \end{scope}
\end{tikzpicture}
```

Rmano, 2022-11-14:
> Offsetting the pin labels to move them either outwards (preferably on the pin pads like in the IC manuals), or atleast off-centering 

Like this?

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usetikzlibrary{backgrounds}
\begin{document}

\begin{tikzpicture}[]
        \draw (0,0) node[dipchip, rotate=90, anchor=pin 8, circuitikz/chips/scale=1.5,
            circuitikz/multipoles/external pins thickness=2,
            circuitikz/multipoles/external pad fraction=4, hide numbers,
            no topmark, ](AD835){}
            (AD835.north) node[above, rotate=90]{\texttt{AD835ANZ}};
        \foreach \i in {1,...,4} \node[font=\tiny\ttfamily, below] at(AD835.bpin \i){\i};
        \foreach \i in {5,...,8} \node[font=\tiny\ttfamily, above] at(AD835.bpin \i){\i};
        \begin{scope}[on background layer={color=gray}]
            \node [mixer, circuitikz/blocks/scale=0.7](M) at([xshift=-0.3cm]AD835.center) {};
            % 2.25 is 1.5*1.5 (chip scale * NL/2
            \node [muxdemux, muxdemux def={Lh=2.25, NL=3, Rh=0.8, NR=1, w=0.8, NB=1},
                external pins width=0, rotate=-90, anchor=blpin 3](DX)
                at ([yshift=-3mm]AD835.bpin 8){};
            \node [font=\tiny, below=-1mm] at (DX.blpin 3) {$+$};
            \draw (DX.blpin 3) -- (AD835.bpin 8) (DX.blpin 1) -- (AD835.bpin 7);
            \node [font=\tiny, below=-1mm] at (DX.blpin 1) {$-$};
            \node [muxdemux, muxdemux def={Lh=2.25, NL=3, Rh=0.8, NR=1, w=0.8, NB=1},
                external pins width=0, rotate=90, anchor=blpin 1](DY)
                at ([yshift=3mm]AD835.bpin 1){};
            \node [font=\tiny, above=-1mm] at (DY.blpin 1) {$+$};
            \node [font=\tiny, above=-1mm] at (DY.blpin 3) {$-$};
            \draw (DY.blpin 1) -- (AD835.bpin 1) (DY.blpin 3) -- (AD835.bpin 2);
            \node [muxdemux, muxdemux def={Lh=3, NL=3, Rh=1, NR=1, w=1, NB=1}, external pins width=0](A)
                at([xshift=0.7cm]AD835.center) {};
            \draw (DX.brpin 1) -- ++(0,-0.2) -- (M) (DY.brpin 1) -- ++(0,0.2) -- (M) (M.e) -- (A.left)
                (A.right) -| (AD835.bpin 5) (A.bbpin 1) -- ++(0,-0.2) -| (AD835.bpin 4);
        \end{scope}
    \end{tikzpicture}
\end{document}
```

[image]

(I reduced the dependencies, too).

> any book recommendation for learning Tikz & CircuiTikz?

I learnt by perusing StackExchange and the Ti*k*Z manual. About `circuitikz`, I am pretty sure that the only book existing is the manual (which I heavily author, but it's not just mine).

yashpalgoyal1304, 2022-11-14:
Code for drawing the pin layout diagram of IC-`74x08`:

```latex
% Refer:
% 2.4 A logic circuit
% 4.25 Chips (IC)
% Code shared by @Rmano
% Tikz: 2.20, 88 Repeating Things: For-loops

\documentclass[border=10pt]{standalone}
\usepackage{circuitikz}
\usetikzlibrary{backgrounds}

\begin{document}
	
% Define Sizes
\ctikzset{
	multipoles/thickness=4,
	multipoles/external pins thickness=2,
	multipoles/dipchip/pin spacing=0.8,
	multipoles/dipchip/width=2,
	multipoles/external pad fraction=4,
	multipoles/external pins width=0.3,
	logic ports=ieee,
	logic ports/scale=0.4
}

% Define the IC for 2-in, 1-out gates
\newcommand*{\gateic}[3]{
	%	\draw (0,0) 
	node[dipchip, rotate=90, num pins=14, hide numbers]%
	(ic08){\rotatebox{-90}{\texttt{#1}}};
	
	% Draw Pin Numbers outside the bpin
	\foreach \i in {1,...,7} \node[font=\scriptsize\ttfamily, below] at(ic08.bpin \i){\i};
	\foreach \i in {8,...,14} \node[font=\scriptsize\ttfamily, above] at(ic08.bpin \i){\i};
	
	% Label VCC & GND
	\node[font=\small\ttfamily, below] at(ic08.pin 7){GND};
	\node[font=\small\ttfamily, above] at(ic08.pin 14){VCC};
	
	% Begin Overlay Circuit
	\begin{scope}[on background layer={color=gray}]
		\def\gate{#2 }
		
		% Specify the parameters for gate
		% a = in 1, b = in 2, y = out, s = sign, g = gate
		\foreach \a / \b / \y / \s / \g in {%
			{1/2/3/+/\gate},
			{4/5/6/+/\gate},
			{9/10/8/-/\gate},
			{12/13/11/-/\gate}}{%
			
			\draw (ic08.bpin \y) |- ++(-0.1,\s 0.7) node[\g port, anchor=out](1){};
			\draw (ic08.bpin \b) |- (1.in 2);
			\draw (ic08.bpin \a) |- (1.in 1);
		}
	\end{scope}
	% End Overlay Circuit
} % End IC definition

\begin{tikzpicture}[]
	\draw (0,0) \gateic{74x08}{and};
	\draw (0,-5) \gateic{74xxx}{xnor};
\end{tikzpicture}

\end{document}
```

Output: <br/>
[image]

Comparing with the reference/target image, it's awesome: 
_(ofc, the reference image is for NANDs, but I deliberately made for the ANDs)_<br/>
![icpl image]

[icpl image]: https://user-images.githubusercontent.com/19423063/201618559-8a6cd53f-ba0b-4d46-a56e-46207b3a2237.png

----
### Further thoughts:

Any suggestions (code, implementation, or graphic, or anything else) is welcome

In the previous attempt with parametrization, 
* it required arithmetic on the pin selection (for anchors and such) 
* and i haven't yet figured out how to do arithmetic properly in LaTeX. 
* The `\*expr` (numpexpr, dimexpr etc) are always shown invalid in horizontal environment lol
* so, do you know how can i do arithimetic in latex?

PS:
1. How did you get the output in the image format - like did you convert the pdf into png, or directly output to png? In either case, how (just a brief summary like)?
2. Remaining major things from my side is just the breadboard, and the alignment of these obtained figures on the page (the need might arise to use it with multi-column layout) - but i am positive that i'll figure those things out now.

----
### For Inclusion in the documentation as example:

* if this one seems good to you (i.e. worthy of addition in documentation - examples),
* then i will be happy to discuss it with you as i am thinking it's scope can be expanded a bit more if it were to be included there like:
    + highlighting one gate (just to showcase that as an example)

Rmano, 2022-11-15:
I will get back to you with comments as soon as I finish the paper submission I'm in (for not having time, this is quite impressive 😉 on your side).

About the arithmetic, if you show me an example I can help. Often the problem is expandability... You can check `\pgfmathsetmacro` and/or the new LaTeX3 `\fpeval`, `\inteval`...

yashpalgoyal1304, 2022-11-15:
> for not having time, this is quite impressive 😉 on your side

haha, thanks :smiley:

for clarification, i meant that - i can not get enough **relaxed** time to calmly dig through stuff for solutions, and do experiments with their validity and applicability to my use. Mental Bandwidth would have pro'lly been a better term there instead of just time - but i guess that sounds a bit showoff i guess :sweat_smile:

yashpalgoyal1304, 2022-11-15:
plus, i gotta say, the documentation of Ti*k*z is sometimes a bit - umh - ambiguous or unclear, for example, 
* i understood the difference in the relative coordinates only from CircuitTi*k*z documentation. 
* for `\foreach` at section "2.21 Adding Text", it felt a bit unclear. And lack of any redirects to the detailed one ("88 Repeating Things") made it worse.
* detailed citations/quotes below:

----
Quotes:

> **Ti*k*z 2.15 Specifying Coordinates:**
> +(0cm,1cm) means “1cm upwards from the previous specified position” 
> ++(2cm,0cm) means “2cm to the right of the previous specified position, making this the new specified position”
>
> **CircuiTi*k*z 3.1.3 Relative coordinates:**
> the +(x,y)-style coordinates (which are supposed to set a temporary relative coordinate and then going back to the starting point).

1. The circuitikz one made it immediately clear - _"going back to the starting point"_ , just the lack of _"making this the new specified position"_ does not make it clear at all in case of tikz ones.

> **T*k*z 2.21 Adding Text:**
> ... two (or even more) variables separated by a slash as in \x / \xtext
> ```
> \foreach \x/\xtext in {-1, -0.5/-\frac{1}{2}, 1}
>```

2. My brain got confused by a lot of things there: `/-\` why minus in between?, `/` used only at one place here?, etc. 
    * I admit that i was not relaxed, in a hurry, while reading this, but also, this is a, umh, quite direct application example
    * if there were redirect to more basic examples as provided aptly in Ti*k*z manual itself: **88 Repeating Things - Multiple variables (pg 1003)** - than I'd have no problem grabbing that.
    * On this matter, i'd like to applaud CircuiTi*k*z documentation - as it always mentions relevant section containing more details (in both directions: next/previous chapters) - it was only because of those redirects/hyperlinks in CircuiTi*k*z manual that I was able to read multiple sections relevant to me w/o much additional effort in scrolling.

yashpalgoyal1304, 2022-11-15:
_self hiding this comment as off topic_:

----
Quotes:

> **Ti*k*z 2.15 Specifying Coordinates (pg 40):**
> +(0cm,1cm) means “1cm upwards from the previous specified position” 
> ++(2cm,0cm) means “2cm to the right of the previous specified position, making this the new specified position”
>
> **CircuiTi*k*z 3.1.3 Relative coordinates (pg 32):**
> the +(x,y)-style coordinates (which are supposed to set a temporary relative coordinate and then going back to the starting point).

1. The circuitikz one made it immediately clear - _"going back to the starting point"_ , just the lack of _"making this the new specified position"_ does not make it clear at all in case of tikz ones.

> **T*k*z 2.21 Adding Text (pg 46):**
> ... two (or even more) variables separated by a slash as in \x / \xtext
> ... the elements in the set over which \foreach iterates must also be of the form ⟨first⟩/⟨second⟩. In each iteration, \x will be set to ⟨first⟩ and \xtext will be set to ⟨second⟩
> ```
> \foreach \x/\xtext in {-1, -0.5/-\frac{1}{2}, 1}
>```
>.
>
> **Ti*k*z 88 Repeating Things - Multiple variables (pg 1003):**
> ```
> \foreach \x / \y / \r in {0 / 0 / 2mm, 1 / 1 / 3mm, 2 / 0 / 1mm}
>```

2. My brain got confused by a lot of things there: `/-\` why minus in between?, `/` used only at one place here? and that with one more command (which further confused me)? _<tt>\<angle brackets></tt>_?, etc. 
    * I admit that i was not relaxed, in a hurry, while reading this 2-4 days ago, but also, this is a, umh, quite direct application example
    * if there were redirect to more basic examples as provided aptly in Ch-88 in Ti*k*z manual itself:  - than I'd have no problem grabbing that.
    * On this matter, i'd like to applaud CircuiTi*k*z documentation - as it always mentions relevant section containing more details (in both directions: next/previous chapters) - it was only because of those redirects/hyperlinks in CircuiTi*k*z manual that I was able to read multiple sections relevant to me w/o much additional effort in scrolling.

Rmano, 2022-11-15:
I'd love to follow up more, but you got me in a crossfire of deadlines in my job.... 

Yes, the Ti*k*Z manual is somewhat cryptic sometimes, but lately development is quite active so you can PR suggestions.... Once you untangle the source format, that is. 

About the PNG conversion: mine are just bitmap copy and paste from `okular` (or screenshots from `evince`, when the antialiasing of the first one become unbearable).

yashpalgoyal1304, 2022-11-16:
> I'd love to follow up more, but you got me in a crossfire of deadlines in my job....

it's okay, take your time to participate in real life. 
* i don't have any problem in communicating with long gaps in between.
* in fact, when i post some stuff here on github, I generally don't expect any immediate reply.
* .... my posting just means that yeah, i got some time on this issue, and this is my input on that.
* so, don't feel any urgency to reply to my posts/comments.

> Definitely I'd like to add your example to the manual!

thanks, but there's some more cleanup which i wish to do in this example - most notably, defining the sizes in some style so that this definition is optional, rather than being forced onto; so that this can be used flexibly with normal dipchips too

Rmano, 2022-11-16:
> > Definitely I'd like to add your example to the manual!
> 
> thanks. awesome.

...or if you want and you are in the mood, you can even do it as an "advanced tutorial". Whatever you prefer. I can just take the code (and attribution --- please tell me exactly how you'd like to be cited) and add to the examples, or wait for a PR on your side.

yashpalgoyal1304, 2022-11-20:
> LaTeX3 `\fpeval`, `\inteval`
> \- https://github.com/circuitikz/circuitikz/issues/665#issuecomment-1314950801

wow, they worked out of the box: `\inteval{#1/2}` , i thought it would require some additional settings to use LaTeX3

yashpalgoyal1304, 2022-12-04:
Some more involved abstractions/cleanups per the remark mentioned at comment: https://github.com/circuitikz/circuitikz/issues/665#issuecomment-1314236121

[TEX.SE: custom-node-in-tikz]: https://tex.stackexchange.com/questions/366802/#comment904352_366804

- [x] Inside/Outside positioning : #672 
done using partitioning by coordinate calculation
- [ ] Abstracting/defining the code as `node` or `pic` rather than LaTeX command
keyword: "new shape" 
    - **IX.106.5** : The Basic Layer > Nodes and Shapes : Declaring New Shapes ( via: [TEX.SE: custom-node-in-tikz] )
    - **III.18** : Ti*k*Z > Pics

yashpalgoyal1304, 2022-12-19:
I have gather quite a lot of good points on tackling a kinda good amount of logic symbols.

am very excited to put them in a tutorial. however, have exams at the moment at hand which will continue uptill 1 month at least. will do it thereafter.

yashpalgoyal1304, 2022-12-19:
just for records, i was thinking about TikZ styled tutorial - the walkthrough in a third person story type. it allows incorporating various "loose" tips and "reason behinds" as well.

but i was a bit unsure about how deep it will be, as i was thinking of starting from start, so before section 2.1 a current shunt (of Circuitikz tutorials), but its body will not touch much the contents of sections 2.1 to 2.3 transistor based amplifier, but then it will contain much more depth content than section 2.4 A logic circuit

the thing i am wondering about is how we will communicate regarding the order - ideation will require more commits, which will require cleanup. but i just know how to commit - that's pretty much all i know in git.

so, when we get to it, will you handle the commit cleanup part? the rebase / squash commit etc 😅 as i don't want to pollute the commit history...

----
anyways, that's enough rambling for now 😅 , i just wanted to put these thoughts out here

Rmano, 2022-12-19:
> so, when we get to it, will you handle the commit cleanup part? the rebase / squash commit etc sweat_smile as i don't want to pollute the commit history...

Yes, I can help --- as long as you do the changes on you branch, you can even reorganize it and force push --- until the thing is merged, no harm is done (I force push quite a lot when I am developing, you can see that). I am not a git-uber-user too, I just do the basic things (the interactive rebase is the best tool ever), especially if you want to "follow" the development. 

One suggestion: you can develop it in a separate file --- I do the same when preparing part of the manual. You can add a file in your clone with the following content:

```latex
documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}

\makeatletter
\makeatother

\begin{document}
\section{Dummy sec}
\subsection{Dummy subsec}
\subsubsection{Dummy subsubsec}

\begin{groupdesc}
    \circuitdesc*{bnc}{BNC connector}{}(left/135/0.6, right/45/0.6, center/-45/0.6, hot/0/0.6, zero/-135/0.6, shield/-90/0.4)
    \circuitdescbip[cuteopenswitch]{cute open switch}{Cute open switch}{cosw}(out/45/0.2)[out.s/-90/0.2]
\end{groupdesc}

\geolrcoord{bnc}

\begin{lstlisting}[varwidth=true]
    A one
    B two
    C three
\end{lstlisting}

\begin{LTXexample}[varwidth=true, basicstyle=\small\ttfamily]
\begin{circuitikz}[european]
    \draw (0,3) node[nand port](A){}
        (A.out) to[short, *-o] ++(0.5,0);
\end{circuitikz}
\end{LTXexample}

\end{document}
```

which load the same setup used in the manual, and then merge it when it is more or less finished in the main `circuitikzmanual.tex`.

Rmano, 2024-04-26:
Hi @goyalyashpal, @gro1m --- I was thinking about all this, especially after seeing #784. While I think that adding a background image to a generic chip (like I implemented for `muxdemux`es) could be a nice thing¹  for specific circuits, I now think that the full-fledged breadboard thing is quite out-of-scope for the package. 

On the other hand, a package that extends TikZ for that specific usage would be a great thing. I do not know how much of the internals of `circuitikz` it would need (I hope very little), but I am willing to add interfaces where and when needed for the new package. Another advantage of a separate package is that, if the author(s) is(are) willing so, it can be made just LaTeX-specific, and so it could use the power of the `latex3` programming layer, a thing that I can't do in `circuitikz` because it must be usable also in `ConTeXt` (or at least, it *should*). 

Notes:
 
¹ really, it should be improved to be able to use generic, high-level tikz code for the drawings that can use the anchors of the component. Moreover, it should be added to chips. Again, I have little time on my hands now, and I am not sure on how to do it (`tikzpictures` can't be nested...).