circuitikz/circuitikz #672: [Feature] Having relative anchors for chips will help a lot in many cases

yashpalgoyal1304, 2022-12-03:
**Is your feature request related to a problem? Please describe.**
Current lack of relative anchors severly hampers flexibility while abstracting code to define custom components

* Most customisation on chips are done relatively, i.e.:
    * circuit goes inside
    * labelling goes outside
* and achieving that requires lots of fragmentation of otherwise simple logic ...
* and hardcoding the absolute anchors (`above`, `below`, ... ; `north`, `south`, etc) in those fragments

**Describe the solution you'd like**
Some way to define the relative direction, so:
* for `dipchip` `in` or `inside` will be equivalent to `above` if the pin number is less than half, and `below` if pin number is greater than half
* the inverse goes for `outside`
* similar corresponding logic applies to `qfpchip`: just a bit more elaborate and in set of 4:
* say for a 8 pin qfpchip: the set of 4 will be: `{ {1, ... , 2} , {3, ... , 4} , {5, ... , 6} , {7, ... , 8} }`:
    * `inside` will be equivalent to `{ above, left, down, right }`
    * and `outside` will be opposite, and so on

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Umh, i don't think any alternate exists for this solution. of course there can be multiple ways to implement this

**Additional context**
Add any other context or screenshots about the feature request here.

Yes, in the comments below

Rmano, 2022-12-03:
This is a good idea, but --- I do not know how easy would be to achieve it. The problem is what happens when you rotate the chip, say, 90 degrees. Now for pin 1 `outside` should change from `above` to `right`...

yashpalgoyal1304, 2022-12-03:
The right hand side in this screenshot of diff is what would become possible if the proposed solution were there:
* Describing the complete pinout names will become a piece of cake 
* ... with direct correspondence in the mental image of IC pinouts, the source, and the output

[image]

Sharing the output with `outside` replaced by `right`:

[image]

yashpalgoyal1304, 2022-12-03:
> The problem is what happens when you rotate the chip, say, 90 degrees. Now for pin 1 `outside` should change from `above` to `right`...

yeah, good point:
* i haven't yet grasped the depth of transforms, so can't say about that
* i started working in `node[dipchip, rotate=90, ...]` , and thought to remove the rotate once, the whole diagram went haywire, haven't touched rotate again ever since 😆

yashpalgoyal1304, 2022-12-03:
also, aside from the feature request - i.e for the package, do u have any ideas on any even hackish or non-elegant way on how can this be done?

this will make my current task on hand so much more manageable

Rmano, 2022-12-03:
> also, aside from the feature request - i.e for the package, do u have any ideas on any even hackish or non-elegant way on how can this be done?

Unfortunately, no; but I'll think about it. The only thing I can think of is to use a `path` from `bpin` to `pin`, extending it with `calc`, and then use some kind of label along the slope (`sloped`)... difficult.

yashpalgoyal1304, 2022-12-03:
one way which came to my mind was using some kind of offset (similar to the commonly used margin/padding). so:

* use the center anchor, rather than any of those specific anchors
* then applying the offset

the only caveat will be the direction of offset... which for `dipchip` is the sign....

Rmano, 2022-12-03:
You can have a line extending in the direction "outside" by saying something like `($(chip.bpin 1)!3!(chip.pin 1)$)` (not tested), and inside with a negative "stride" instead of 3...

Rmano, 2022-12-03:
```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}[]
    \draw (0,0) node[dipchip](chip){};
    \foreach \i in {1,...,8} {
        \node [circ, red] at ($(chip.bpin \i)!3!(chip.pin \i)$) {};
        \node [circ, blue] at ($(chip.bpin \i)!-1!(chip.pin \i)$) {};
    }
\end{tikzpicture}
\end{document}
```

[image]

yashpalgoyal1304, 2022-12-03:
wow, awesome.

section of the pgf/Ti*k*zmanual corresponding to above shared solution: 
### Part-I, Ch-4 "Tutorial: Euclid’s Amber Version of the Elements"

* pg 63, 4.1.2 Special coordinates
* pg 67, 4.2.1 Using Partway Calculations for the Construction of D

> It turns out that the calc library allows him to do exactly this kind of computation. When this library is
loaded, you can use special coordinates that start [and end like] `($ ... $)` rather than just `( ... )`. Inside
these special coordinates you can give a linear combination of coordinates.

> First, there is a simple syntax for computing a point
that is “partway” on a line from p to q: You place these two points in a coordinate calculation – remember,
they start [and end like] `($ ... $)` – and then combine them using `!⟨part⟩!` . A `⟨part⟩` of 0 refers to the first
coordinate, a `⟨part⟩` of 1 refers to the second coordinate, and a value in between refers to a point on the line
from p to q. Thus, the syntax is similar to the xcolor syntax for mixing colors.

yashpalgoyal1304, 2022-12-03:
this issue can be closed from my side now - i don't particularly think now that there's any need of any additional anchors

thanks a lot

just posting for the records the successful results with `qfpchip` too:

```latex
\documentclass[border=10pt]{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{tikzpicture}[]
    \draw (0,0) node[qfpchip, hide numbers](chip){};
    \foreach \i in {1,...,8} {
        \node [font=\scriptsize, red] at ($(chip.bpin \i)!2!(chip.pin \i)$) {\i};
        \node [font=\tiny\ttfamily, blue] at ($(chip.bpin \i)!-0.5!(chip.pin \i)$) {\i};
    }
\end{tikzpicture}
\end{document}
```

[image]

closed 2022-12-03 by yashpalgoyal1304

Rmano, 2022-12-03:
Yes, nice. The problem could be if you want a non-centered alignment for numbers with different lengths... but maybe that's nitpicking.