circuitikz/circuitikz #679: how to use other shape for a node like triangle (for european invert/not symbol)?

yashpalgoyal1304, 2022-12-13:
* Similarly, what is the node for `tripoles/european not symbol=triangle`
* i.e. how to use other shape for a node like triangle (for european invert/not symbol)? 

I found out that:
* `Triangle[open, left]` (page 219, III.16.5.3 Ti*k*Z > Arrows > Reference: Arrow Tips > Geometric. 
* It says 'ts `An instance of a Kite with zero inset.` 
* so, I tried creating the node with both Triangle, triangle, Kite, kite, but none worked...
* `\node[kite] (B) at (A.bout)  {};` , error returned: `Package pgfkeys Error: I do not know the key '/tikz/kite' and I am going to ignore it. Perhaps you misspelled it. \node[kite]`

----
Citations:

Page 157, 4.22.7 European logic port usage
The node for using outside for `tripoles/european not symbol=ieee circle` is `ocirc` (open circle).

Page 156, 4.23 Flip-flops
a _negation_ flag ... will put an `ocirc` shape on the border of corresponding pin

Page 199, 6.1 Nodes to bipoles (and 4.12 terminal shapes)
You can use other shapes too, but at your own risk…

----
I was easily able to put a circle on the bottom left pin (`blpin`) of the muxdemux, but couldn't 

```latex
\documentclass[border=10px]{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{tikzpicture}
\path (0,0) node[muxdemux, muxdemux def={NL=2,Lh=2,Rh=2}](mdemux){};
% Page 170 anchor=0
\path (mdemux.blpin 1) node[ocirc,anchor=0](){};
\end{tikzpicture}
\end{document}
```

[image]

yashpalgoyal1304, 2022-12-13:
i will move this subcircuit definition to my custom style file, and do `\RequirePackage{circuitikz}[1.5.5]` - will that make it foolproof??

Rmano, 2022-12-13:
> 1. i will move this subcircuit definition to my custom style file, and do `\RequirePackage{circuitikz}[1.5.5]` - will that make it foolproof??
> 

I suppose it will work. Never tried, I do not think that any package depend *on* `circuitikz`

>     2. Will it hurt if i activate it right after the definition in the style file itself? (seems yes, it shows errors)

Yes. The problem is that to know the distances with the actual sizes of components, you have to call the "activate" after all the options are active. That will typeset the circuit to measure the distances...  What you could try to do is to add the activation to a hook, if you don't change styles after the preamble (something like `\AtBeginDocument`, or better one of the new hooks, you can look at `lthooks-doc.pdf` in your TeX system).

Rmano, 2022-12-19:
> I suppose it will work. Never tried; I do not think that any package depends _on_ `circuitikz`

I supposed wrong. There are only some "go back" points in the package, and 1.5.5 is not one of them. Anyway, I do not think to change the subcircuit thing anymore --- at least not in a backward compatible way (I am a bot of a sucker for backward compatibility...). I'll change the manual to reflect that in a future commit...

Rmano, 2022-12-21:
See #682 😉

yashpalgoyal1304, 2022-12-27:
hey, by any means, do you think adding this as new component would be worth it??

Rmano, 2023-01-01:
You got me... I never use European logic symbols (I think a symbol must convey its meaning graphically, and the European symbols don't do that --- just a lot of rectangles). So I am not the right one to judge it. 
Standard Ti*k*Z circuit library has options to add negation on the inputs, which I never tried even to think how to implement (I provide the anchor to trick it); maybe that would be the logical way to do it...

Rmano, 2023-01-27:
Yes, the look is quite nice. I think that's definitely possible (and maybe good) to add the shape --- it's not invasive and it can be used in a lot of place.

yashpalgoyal1304, 2023-01-27:
> Yes, the look is quite nice

glad u liked 'em...

Rmano, 2023-01-27:
What about...

[image]

this one scales with `logic port ` class (this is a bit arbitrary, I know) and the default size I chose are quite random too, but... what do you think? 

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}
\makeatletter
\ctikzset{euroinver/width/.initial = 0.2}
\ctikzset{euroinver/height/.initial = 0.1}
\pgfdeclareshape{euroinver}{
    \savedmacro{\ctikzclass}{\edef\ctikzclass{logic ports}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedanchor\northeast{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@y=\ctikzvalof{euroinver/height}\pgf@circ@scaled@Rlen
        \pgf@x=\ctikzvalof{euroinver/width}\pgf@circ@scaled@Rlen
    }
    \anchor{center}{\pgfpointorigin}
    \anchor{top}{\northeast\pgf@x=0pt\relax}
    \anchor{apex}{\northeast\pgf@y=0pt\relax}
    \anchor{base}{\pgfpointorigin}
    \anchor{mid}{\northeast\pgf@x=0.5\pgf@x\pgf@y=0.5\pgf@y}
    % not standard geoanchors, south west is the same as 0,0...
    \anchor{right}{\northeast\pgf@y=0pt\relax}
    \anchor{east}{\northeast\pgf@y=0pt\relax}
    \anchor{e}{\northeast\pgf@y=0pt\relax}
    \anchor{left}{\pgfpointorigin}
    \anchor{west}{\pgfpointorigin}
    \anchor{w}{\pgfpointorigin}
    \anchor{south}{\northeast\pgf@y=0pt\pgf@x=0.5\pgf@x}
    \anchor{s}{\northeast\pgf@y=0pt\pgf@x=0.5\pgf@x}
    \anchor{north}{\northeast\pgf@x=0.5\pgf@x}
    \anchor{n}{\northeast\pgf@x=0.5\pgf@x}
    \anchor{south west}{\pgfpointorigin}
    \anchor{sw}{\pgfpointorigin}
    \anchor{north east}{\northeast}
    \anchor{ne}{\northeast}
    \anchor{north west}{\northeast\pgf@x=0pt\relax}
    \anchor{nw}{\northeast\pgf@x=0pt\relax}
    \anchor{south east}{\northeast\pgf@y=0pt\relax}
    \anchor{se}{\northeast\pgf@y=0pt\relax}
    % geo anchors based on north-east (not correct, fix me)
    % \pgfcirc@northeast@symmetric@geoanchors
    \pgf@circ@draw@component{
        \northeast\pgf@circ@res@up=\pgf@y\pgf@circ@res@right=\pgf@x
        \typeout{DIMENSxy:\space\the\pgf@circ@scaled@Rlen\space \the\pgf@circ@res@right\space \the\pgf@circ@res@up}
        \pgfpathmoveto{\pgfpointorigin}
        \pgfpathlineto{\pgfpoint{0pt}{\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0pt}}
        \pgfpathclose
        \pgfusepath{draw}
    }
}
\makeatother
\begin{document}
\begin{tikzpicture}[]
    \path (0,0) node[muxdemux, muxdemux def={NL=2,Lh=2,Rh=2,NB=0},external pins width = 2](mdemux){};
    \draw (mdemux.blpin 1) node[euroinver,anchor=apex]{};
    \draw (mdemux.brpin 1) node[euroinver]{} ;
\end{tikzpicture}

\showanchors[]{euroinver, scale=6}{}(north/90/0.4, north east/45/0.4, east/0/0.4,
    south east/-45/0.4,
    south/-90/0.4, south west/-135/0.4, west/180/0.4, north west/135/0.4,
    left/160/0.4, right/30/0.4, center/-120/0.3,
    base/-90/0.6, top/90/0.6, apex/-90/0.6, mid/60/0.6
    )

\end{document}
```

yashpalgoyal1304, 2023-01-28:
i prepared a whole paragraph of reply... but thought it will require a bit more testing and filtering on my part.

Rmano, 2023-01-28:
If you can use the new shape with one of your examples I'll be glad to add it...

Rmano, 2023-01-28:
What about this:

[image]
[image]

<details><summary>CODE:</summary>

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}
\makeatletter
\ctikzset{european inversion mark/width/.initial = 0.2}
\ctikzset{european inversion mark/height/.initial = 0.1}
\pgfdeclareshape{euroinv}{
    \savedmacro{\ctikzclass}{\edef\ctikzclass{muxdemuxes}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedanchor\northeast{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@y=\ctikzvalof{european inversion mark/height}\pgf@circ@scaled@Rlen
        \pgf@x=\ctikzvalof{european inversion mark/width}\pgf@circ@scaled@Rlen
    }
    \anchor{center}{\pgfpointorigin}
    \anchor{top}{\northeast\pgf@x=0pt\relax}
    \anchor{apex}{\northeast\pgf@y=0pt\relax}
    \anchor{base}{\pgfpointorigin}
    \anchor{mid}{\northeast\pgf@x=0.5\pgf@x\pgf@y=0.5\pgf@y}
    % not standard geoanchors, south west is the same as 0,0...
    \anchor{right}{\northeast\pgf@y=0pt\relax}
    \anchor{east}{\northeast\pgf@y=0pt\relax}
    \anchor{e}{\northeast\pgf@y=0pt\relax}
    \anchor{left}{\pgfpointorigin}
    \anchor{west}{\pgfpointorigin}
    \anchor{w}{\pgfpointorigin}
    \anchor{south}{\northeast\pgf@y=0pt\pgf@x=0.5\pgf@x}
    \anchor{s}{\northeast\pgf@y=0pt\pgf@x=0.5\pgf@x}
    \anchor{north}{\northeast\pgf@x=0.5\pgf@x}
    \anchor{n}{\northeast\pgf@x=0.5\pgf@x}
    \anchor{south west}{\pgfpointorigin}
    \anchor{sw}{\pgfpointorigin}
    \anchor{north east}{\northeast}
    \anchor{ne}{\northeast}
    \anchor{north west}{\northeast\pgf@x=0pt\relax}
    \anchor{nw}{\northeast\pgf@x=0pt\relax}
    \anchor{south east}{\northeast\pgf@y=0pt\relax}
    \anchor{se}{\northeast\pgf@y=0pt\relax}
    % geo anchors based on north-east (not correct, fix me)
    % \pgfcirc@northeast@symmetric@geoanchors
    \pgf@circ@draw@component{
        \northeast\pgf@circ@res@up=\pgf@y\pgf@circ@res@right=\pgf@x
        \pgfsetbeveljoin
        \pgfpathmoveto{\pgfpointorigin}
        \pgfpathlineto{\pgfpoint{0pt}{\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0pt}}
        \pgfpathclose
        \pgfusepath{draw}
    }
}
\makeatother
\begin{document}
\section{Dummy sec}
\subsection{Dummy subsec}
\subsubsection{Dummy subsubsec}

\circuitdesc{euroinv, scale=2}{Inversion marker for European logic symbols\footnotemark}{}(base/180/0.2, top/90/0.2, apex/0/0.2, mid/45/0.2)
\footnotetext{Thanks for the contribution by \href{https://github.com/circuitikz/circuitikz/issues/679}{\texttt{yashpalgoyal1304} on GitHub}.}

It's often necessary or handy to add an inversion marker to symbols created with this class of devices; the \texttt{euroinv} shape will nicely do. It'll scale with the \texttt{muxdemuxes} class, and the length and height can be changed with the keys \texttt{european inversion mark/width} (default \texttt{0.2}) and \texttt{height} (default \texttt{0.1}), with the same units that are used for the \texttt{external pins width} and similar keys.

\begin{LTXexample}[varwidth=true]
\begin{tikzpicture}[]
    \path (0,3) node[muxdemux, muxdemux def={NL=2,Lh=2,Rh=2,NB=1},
        external pins width = 0.5](mdemux){};
    \draw (mdemux.blpin 1) node[euroinv,anchor=apex]{};
    \draw (mdemux.brpin 1) node[euroinv]{};
    \draw (mdemux.bbpin 1) node[euroinv, anchor=apex,
        rotate=90]{};
    \ctikzset{european inversion mark/.cd, width=0.1}
    \ctikzset{muxdemuxes/.cd, fill=yellow, thickness=4, scale=0.8}
    \path (0,0) node[muxdemux, muxdemux def={NL=2,Lh=2,Rh=2,NB=0},
        external pins width = 0.5](mdemux){};
    \draw (mdemux.blpin 1) node[euroinv,anchor=apex]{};
    \draw (mdemux.brpin 1) node[euroinv]{};
\end{tikzpicture}
\end{LTXexample}

%\showanchors[]{euroinv, scale=6}{}(north/90/0.4, north east/45/0.4, east/0/0.4,
%   south east/-45/0.4,
%    south/-90/0.4, south west/-135/0.4, west/180/0.4, north west/135/0.4,
%    left/160/0.4, right/30/0.4, center/-120/0.3,
%    base/-90/0.6, top/90/0.6, apex/-90/0.6, mid/60/0.6
%    )

\end{document}
```
</details>

yashpalgoyal1304, 2023-01-28:
> _"and the length and height can be changed with ..."_

yeah, that's what i was thinking for. I was thinking more about angle... as when modifying, it may be convenient to be able to specify the apex angle? what do you think? although, yeah, the height and width gives better control...

yashpalgoyal1304, 2023-01-28:
> the default size I chose are quite random too

For dimensions:
- is this specified in IEEE too?
- how aboout picking the one already used in european logic ports in circuitikz
- is it consistent with TikZ circuit library

yashpalgoyal1304, 2023-01-28:
Found one more interesting document while [searching]: 
http://www.wakerly.org/DDPP/DDPP3_pdf/IEEEsyms.pdf | [archival link]

It contains nice terminology which will come in handy while writing tutorials.

[archival link]: https://web.archive.org/web/20221222094950/http://wakerly.org/DDPP/DDPP3_pdf/IEEEsyms.pdf
[searching]: https://duckduckgo.com/?q=ieee+european+symbol&t=vivaldi&ia=web

At page 2, it says the following:

> ANOTHER KIND OF BUBBLE
In addition to the familiar bubble, the IEEE standard also allows an external,
triangular “polarity symbol” to be used to specify active-low inputs and outputs,
for which the external `LOW` level corresponds to the internal 1-state. However,
under a positive-logic convention, the bubble and the triangular polarity symbol
are equivalent, so we use the more traditional bubble in this appendix.

Rmano, 2023-01-28:
> wow, was searching for the IEEE standard, and it requires sign in to view eyes

I have it, and 1973 one in some place,  but I can't find it now. I used them to define IEEE shapes(see #383 for more details). There is no specific size there, but you can see here that the 2:1 proportion is more or less the correct one:

[image]

Notice that in IEEE the triangle is normally meant to indicate an *electrical* condition (negative logic versus positive logic), not logical; but it's true that now everything is in positive logic, so it became equivalent to the "negaation ball".

yashpalgoyal1304, 2023-01-28:
> Notice that in IEEE the triangle is normally meant to indicate an _electrical_ condition (negative logic versus positive logic), not logical; but it's true that now everything is in positive logic, ...

i don't really understand these terms then 😅, will learn someday later. any recommended resource suggestion regarding that are welcome :)

Rmano, 2023-01-28:
> i don't really understand these terms then sweat_smile, will learn someday later. any recommended resource suggestion regarding that are welcome :)

No surprise, I know it because I'm old. If you consider "1" low voltage and "0" high voltage, a NAND swaps meaning with a NOR, and so on.  But I think I saw a negative logic circuit last time in an exam when I was a student at Uni, so we can probably we can consider it historical.

So do you think I can merge it like it is?

yashpalgoyal1304, 2023-01-29:
> So do you think I can merge it like it is?

reviewing there itself

yashpalgoyal1304, 2023-02-18:
Just sharing for reference:

```
U+2300 Miscellaneous Technical
⌲ U+2332 CONICAL TAPER
⌳ U+2333 SLOPE
```

PR #688 (2023-02-01): Add `wedgeinv` and `circleinv` shape to muxdemuxes
Thanks to @yashpalgoyal1304
Fixes #679

[image]
commit 3b3fcdde (2023-02-01): Add `wedgeinv` and `circleinv` shapes

Thanks to @yashpalgoyal1304
Fixes #679

Minor optimization/fix for the notcirc, too