circuitikz/circuitikz #680: Feature request: International and english symbol for direct current (DC) symbol

dbstf, 2022-12-13:
I'ld like to customize the direct current (DC) symbol of the circuitikz library to match the international or English-speaking definition ([https://de.wikipedia.org/wiki/Gleichstrom](https://de.wikipedia.org/wiki/Gleichstrom)) as shown below.
[image]
I personally favor variant c). In previous versions (v1.4.6 was the latest one I had installed before updating to v1.6.0) I could simply edit the file "C:\texlive\2022\texmf-dist\tex\generic\circuitikz\pgfcircquadpoles.tex" and change the section
```
%draw equal sign 
\pgfpathmoveto{\pgfpoint{.2\pgf@circ@res@step}{-.375\pgf@circ@res@step}}
\pgfpathlineto{\pgfpoint{0.7\pgf@circ@res@step}{-0.375\pgf@circ@res@step}}
\pgfusepath{draw}
\pgfpathmoveto{\pgfpoint{.2\pgf@circ@res@step}{-.625\pgf@circ@res@step}}
\pgfpathlineto{\pgfpoint{0.7\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfusepath{draw}
```
to
```
%draw equal sign 
\pgfpathmoveto{\pgfpoint{0.2\pgf@circ@res@step}{-0.375\pgf@circ@res@step}}
\pgfpathlineto{\pgfpoint{0.7\pgf@circ@res@step}{-0.375\pgf@circ@res@step}}
\pgfusepath{draw}
\pgfpathmoveto{\pgfpoint{0.2\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfpathlineto{\pgfpoint{0.3\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfusepath{draw}
\pgfpathmoveto{\pgfpoint{0.4\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfpathlineto{\pgfpoint{0.5\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfusepath{draw}
\pgfpathmoveto{\pgfpoint{0.6\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfpathlineto{\pgfpoint{0.7\pgf@circ@res@step}{-0.625\pgf@circ@res@step}}
\pgfusepath{draw}
```
However, the 'blocks' class seems to have changed in newer revisions of the package.
Now the question: Where did the definition of the DC symbol a.k.a. equal sign move to? I.e., how can I change it?

If one wants to get really fancy, one could implement this as package option, e.g., a suggestion here:
- olddcsymbol --> leaves the DC symbol defined as "equal sign" --> package default
- intdcsymbol --> defines the DC symbol as variant b) shown in the picture above
- endcsymbol --> defines the DC symbol as variant c) shown in the picture above

Many thanks in advance

Rmano, 2022-12-13:
Yes, the fact is that I factored out the common code. Now the "dc symbol" is created by the code in `\pgf@circ@twoport@converter@dc`. You can change that and change all the symbols together... put in your preamble:

```
\makeatletter
\def\pgf@circ@twoport@converter@dc#1#2{%
    \pgfscope
    \pgftransformshift{\pgfpoint{#1\pgf@circ@res@step}{#2\pgf@circ@res@step}}
    \pgfpathmoveto{\pgfpoint{-0.25\pgf@circ@res@step}{0.125\pgf@circ@res@step}}
    \pgfpathlineto{\pgfpoint{0.25\pgf@circ@res@step}{0.125\pgf@circ@res@step}}
    \pgfusepath{draw}
    \pgfpathmoveto{\pgfpoint{-0.25\pgf@circ@res@step}{-0.125\pgf@circ@res@step}}
    \pgfpathlineto{\pgfpoint{0.25\pgf@circ@res@step}{-0.125\pgf@circ@res@step}}
    \pgfusepath{draw}
    \endpgfscope
}
\makeatother
```

after loading `circuititk` and modifying the code there (the arguments are for shifting the symbol around, ignore them and just modify the second block of moveto-lineto).

But yes, it could be an interesting option to add it to the code. I'll look at it.

dbstf, 2022-12-13:
Many thanks for the quick reply! That's exactly what I was looking for!
If more people are interested in this I'ld be happy to see this implemented in some future release :)

Rmano, 2022-12-18:
What about this:

[image]

Please comment (also about the names of the keys... I changed them several times)

Rmano, 2022-12-18:
The code still have a bit of problems if the inner block is dashed --- I have to figure out what to do.

dbstf, 2022-12-19:
Looks very promising to me. I especially appreciate the options to globally set the the dashing with one global key, while still preserving the ability to modify the "in dashing" individually from the "out dashing" if needed.
The key names sound good to me, as well. Setting the number of dashes on the bottom line with the corresponding numerical value feels intuitive to me.
I would also like to add the supported values to the documentation, e.g., something like this:
"... key blocks dc segments (1 for a continuous line (default); 2 for two bottom dashes (international symbol); 3 for three bottom dashes (English symbol))."
There also is a minor copy-paste mistake in the last sentence of the documentation:
"... the key blocks dc in segments and blocks dc **_in_** segments (see the following example)". The second "dc in" should write "dc **out**".

Rmano, 2022-12-19:
Like this?

[image]

dbstf, 2022-12-20:
Looks very good to me. I like it.

Rmano, 2022-12-20:
Merged!

PR #681 (2022-12-20): Add configurable dashes to the dc symbols in converter blocks
suggested by user @dbstf on GitHub
Fixes #680

[image]
commit 3c5f83b5 (2022-12-20): Add configurable dashes to the dc symbols in converter blocks

suggested by user @dbstf on GitHub
Fixes #680
closed 2022-12-20 by Rmano