circuitikz/circuitikz #684: A question about the alignment of american voltage signs when using other fonts

EEpchi, 2023-01-09:
I don't know if this counts as a bug😂, maybe it's the effect of the font package. that is, when I use the package "newtxtext" and "nextxmath" to draw a circuit, the american voltage signs can't well aligned. The "minus" looks like it's moved up a bit. The MWE is shown below. I draw a line using anchor to show their position offset.

[image]

```
\documentclass{article}
\usepackage{newtxtext}
\usepackage{newtxmath}
\usepackage[americanvoltages]{circuitikz}
\begin{document}
	\begin{circuitikz}[raised voltages]
		\draw (0,0) to[R, v=$v_1$, name=R] ++(2,0);
		\draw[red] (R-Vto) -- (R-Vfrom);
	\end{circuitikz}
\end{document}
```

Rmano, 2023-01-09:
I think it's a problem with the font, namely the bounding boxes of the chars. Look:

```latex
\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage{newtxtext}
\usepackage{newtxmath}
\begin{document}
\Large \fbox{$+$} \fbox{$-$}
\end{document}
```

[image]

Maybe there is a workaround (I have the sensation I have already seen this somewhere, but I am unable to find the reference).

Anyway, you can cheat:

```latex
\documentclass{article}
\usepackage{newtxtext}
\usepackage{newtxmath}
\usepackage[americanvoltages]{circuitikz}
\ctikzset{voltage/american plus={${+}$}}
\ctikzset{voltage/american minus={$\vphantom{+}-$}}
\begin{document}
	\begin{circuitikz}[raised voltages]
		\draw (0,0) to[R, v=$v_1$, name=R] ++(2,0);
		\draw[red] (R-Vto) -- (R-Vfrom);
	\end{circuitikz}
\end{document}
```

[image]

Rmano, 2023-01-09:
Ah, I found it:

https://tex.stackexchange.com/questions/609201/circuitikz-american-voltage-signs-not-vertically-aligned-with-unicode-math

EEpchi, 2023-01-10:
Oh, thank you very much for your answer, which is very helpful to me.🌹

closed 2023-01-10 by Rmano