circuitikz/circuitikz #689: [Feature]

myzinsky, 2023-02-01:
I want to draw such a block diagram: 

[image]

Right now I'm drawing the mic with a circle. However, I would just use micshape. However, the micshape has the connection lines attached to the shape. However, the loudspeakershape does not! It would be nice if mic and loudspeaker could have the same behaviour. Any thoughts?

Rmano, 2023-02-01:
Yep, it could be a worthwhile addition to the block thing.

Would you try a shot at it? I do have not a lot of time now, although I use `circuitikz` development to relax a bit...

myzinsky, 2023-02-01:
I wanted to try it, however, I do not even find the micshape in the code :-D Any pointers?

Rmano, 2023-02-01:
Anchors: 

[image]

myzinsky, 2023-02-03:
Will you bring that to main?

Rmano, 2023-02-03:
Yes, this weekend I'll add and merge it. I was thinking to add an option to control the thickness of the "bar" independently, in both `mic` and `tlmic`.

myzinsky, 2023-02-03:
I made also some other block components. Shall I open a pull request for each of them individually? I really want to learn this by myself, since I have a huge project with a lot of custom Building Blocks, mainly from German DIN.

Rmano, 2023-02-03:
Well, let's try with one or a first group of them, and then when the structure/details are ok, you can do the rest in a block.

One of the problem I have is that `circuitikz` is becoming quite heavy; I have to find a way to create a "sublibraries" system to avoid loading everything all the time. But this is quite a big job, and I do not think I'll be able to tackle that before this summer...

Rmano, 2023-02-03:
[image]

closed 2023-02-03 by Rmano

myzinsky, 2023-02-03:
Question, when will this be available in the CPAN tex?

Rmano, 2023-02-03:
CTAN? Yes, as soon as I make a release... I had *almost* done one last weekend; then things started coming in. 😉

As soon as things calm down a bit, I'll make a release.

PR #692 (2023-02-03): Add tail-less microphone
Also, add the option to change the microphone bar thickness.

Fixes #689

[image]
commit 5cdcbc00 (2023-02-03): Add tail-less microphone

Also, add the option to change the microphone bar thickness.

Fixes #689