circuitikz/circuitikz #691: [New Component] Better GaN HEMT Symbol

epsilon-phi, 2023-02-02:
Hello,

i would like to propose a better symbol styling for the GaN HEMT symbol. Unfortunately I am quite unfamiliar with git, github and anything deeper than the very top level of latex. Aditionally this is my first time using github for anything else than downloading so please excuse me if i do not know the usual practises and conventions.

**--- The Component ---**

[image]

This exact component was also requested here: https://tex.stackexchange.com/questions/523418/draw-gan-e-mode-hemt-transistor-with-circuitikz

**--- The Code ---**

This is the code that wrote directly into my local tex distribution. 

File pgfcirctripoles.tex:

        \ctikzset{tripoles/pchemt/width/.initial=.7}
        \ctikzset{tripoles/pchemt/gate height/.initial=.65}
        \ctikzset{tripoles/pchemt/base height/.initial=.75}
        \ctikzset{tripoles/pchemt/conn height/.initial=.35}
        \ctikzset{tripoles/pchemt/gate channel height/.initial=.1}
        \ctikzset{tripoles/pchemt/channel sep/.initial=.05}
        \ctikzset{tripoles/pchemt/height/.initial=1.1}
        \ctikzset{tripoles/pchemt/base width/.initial=.5}
        \ctikzset{tripoles/pchemt/gate width/.initial=.62}
        \ctikzset{tripoles/pchemt/bodydiode scale/.initial=.3}
        \ctikzset{tripoles/pchemt/bodydiode distance/.initial=.3}
        \ctikzset{tripoles/pchemt/bodydiode conn/.initial=.6}
        \ctikzset{tripoles/pchemt/curr direction/.initial=1}

and (this is basically copy-paste from the existing hemt transistor symbol)

      %% Philipp Czerwenka HEMT FET Transistor
      \pgfcircdeclaretransistor{pchemt}{
              \anchor{inner up}{
                  \northeast
                  \pgf@y=\ctikzvalof{tripoles/pchemt/gate height}\pgf@y
              }
              \anchor{inner down}{
                  \northeast
                  \pgf@y=-\ctikzvalof{tripoles/pchemt/gate height}\pgf@y
              }
          }{%
          \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@up+\pgfverticaltransformationadjustment*.5*\pgflinewidth}}
          \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}
          {\ctikzvalof{tripoles/pchemt/gate height}\pgf@circ@res@up}}
          \pgfpathlineto{\pgfpoint
              {\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}
          {\ctikzvalof{tripoles/pchemt/gate height}\pgf@circ@res@up}}
      
          \pgfpathmoveto{\pgfpoint
              {\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}
          {\ctikzvalof{tripoles/pchemt/gate height}\pgf@circ@res@down}}
          \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}
          {\ctikzvalof{tripoles/pchemt/gate height}\pgf@circ@res@down}}
          \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@down-\pgfverticaltransformationadjustment*.5*\pgflinewidth}}
          \pgfusepath{draw}
	      
	      \pgfscope
	      \pgfpathmoveto{\pgfpoint{\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}{\ctikzvalof{tripoles/pchemt/base height}\pgf@circ@res@down}}
          \pgfpathlineto{\pgfpoint{\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}{\ctikzvalof{tripoles/pchemt/conn height}+2*\ctikzvalof{tripoles/pchemt/gate channel height})*\pgf@circ@res@down}}
	      \pgf@circ@setlinewidth{tripoles}{\pgflinewidth}
	      \pgfusepath{draw}
	      \endpgfscope
      
	      \pgfscope
	      \pgfpathmoveto{\pgfpoint{\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}{(\ctikzvalof{tripoles/pchemt/conn height}-\ctikzvalof{tripoles/pchemt/gate channel height})*\pgf@circ@res@down}}
          \pgfpathlineto{\pgfpoint{\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}{(\ctikzvalof{tripoles/pchemt/conn height}+\ctikzvalof{tripoles/pchemt/gate channel height})*\pgf@circ@res@down}}
          \pgf@circ@setlinewidth{tripoles}{\pgflinewidth}
	      \pgfusepath{draw}
	      \endpgfscope
      
	      \pgfscope
	      \pgfpathmoveto{\pgfpoint{\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}{\ctikzvalof{tripoles/pchemt/conn height}-2*\ctikzvalof{tripoles/pchemt/gate channel height})*\pgf@circ@res@down}}
          \pgfpathlineto{\pgfpoint{\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}{\ctikzvalof{tripoles/pchemt/base height}\pgf@circ@res@up}}
	      \pgf@circ@setlinewidth{tripoles}{\pgflinewidth}
          \pgfusepath{draw}
	      \endpgfscope
	      
	      % arrows
          \pgfscope
              \pgfslopedattimetrue
              \pgfallowupsidedownattimetrue
              \pgfresetnontranslationattimefalse
          %    \ifpgf@circuit@trans@arrowatend
          %            \ifpgf@circuit@trans@ntype
          %                \edef\@@anchor{tip}
          %                \edef\@@pos{1.0}
          %            \else
          %                \edef\@@anchor{btip}
          %                \edef\@@pos{0.0}
          %            \fi
          %    \else
                  \edef\@@anchor{center}\edef\@@pos{0.6}
          %    \fi
              \pgftransformlineattime{\@@pos}{%
                  \pgfpoint
                  {\pgf@circ@res@right}%
                  {\ctikzvalof{tripoles/pchemt/gate height}\pgf@circ@res@down}%
                  }{%
                  \pgfpoint%
                  {\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}%
                  {\ctikzvalof{tripoles/pchemt/gate height}\pgf@circ@res@down}%
              }
          %    \ifpgf@circuit@trans@ntype
          %    \else
                  \pgftransformrotate{180}
          %    \fi
              \pgfnode{trarrow}{\@@anchor}{}{}{\pgfusepath{stroke}}
          \endpgfscope
	      
	      \pgfpathmoveto{\pgfpoint
              {\ctikzvalof{tripoles/pchemt/base width}\pgf@circ@res@left}
          {\ctikzvalof{tripoles/pchemt/conn height}\pgf@circ@res@down}}
          \pgfpathlineto{\pgfpoint{\pgf@circ@res@left-\pgfhorizontaltransformationadjustment*.5*\pgflinewidth}{\ctikzvalof{tripoles/pchemt/conn height}\pgf@circ@res@down}}
	      \pgfusepath{draw}
      }

To use in a latex document, it can just be used like any other node transistor symbol.

        \documentclass[border=1mm]{standalone}
        \usepackage[siunitx, RPvoltages]{circuitikz}
        \begin{document}
            \begin{circuitikz}[european]
                \draw (0,0) node[pchemt] (Q1) {$Q_{\mathrm{1}}$};
            \end{circuitikz}
        \end{document}

If you do choose to include the proposed symbol in circuittikz, you might want to assign a better name than "pchemt". I suggest "ehemt" as this is a symbol for an e-mode GaN HEMT.
Along these lines: If you do choose include the proposed symbol, also including a d-mode version with the vertical bar drawn connected would be awesome.

**--- Regarding Standards ---**

Regarding standards i believe that something like the IEEE Std 315-197 does not contain a symbol for the 2D electron gas yet, which gives e-mode GaNs their characteristics. So I am citing papers: 

 * GaN Manufacturer GaNSystems uses it in their paper: [10.1109/TTE.2020.2989036](https://doi.org/10.1109/TTE.2020.2989036)
 * We use it inhouse and have published with it: [978-9-0758-1539-9](https://ieeexplore.ieee.org/document/9907587)

I am aware that many other sources just use the regular N-channel MOSFET symbol. There are also sources that use the GaN HEMT symbol already existing in circuittikz, however i believe they are just as sparse as the examples i mentioned. I am also not suggesting to exchange one symbol for the other, rather to offer both and let the author decide.

Thank you kindly for considering my suggestion.
Greetings from Reutlingen, Germany
Philipp Czerwenka

Rmano, 2023-02-02:
Hmmm... nice, I will give it a shot when I have time. I have a suggestion for the time being and a question. 

**Suggestion**: do not modify `pgfcirctripoles.tex` in your distribution; it will be overwritten on any update. You can just put your code after loading `circuitikz` in a `\makeatletter ... \makeatother` block. See, for example https://github.com/circuitikz/circuitikz/issues/679#issuecomment-1406972921

**Question**: apart from the depletion-enhancement version, should there be the `n` channel also (with reversed arrow)?

Rmano, 2023-02-02:
And... Which would be the size with respect to the other transistors? It is meant to be taller of just the gate position is lower?

epsilon-phi, 2023-02-03:
Thank you so much for the feedback! I will absolutely follow your suggestion in the future (makes it a lot easier for me indeed).

Rmano, 2023-02-04:
I will have a look at it. The transistor "set" of the package is one of the parts that I like less. There is a lot of code repetition (which in great part is my fault) and really little generalization. Add this to the fact that the graphic representation of transistors is one of the most variable out there... 
Anyway, let's see if it's possible to generalize the `hemt` thing to get the asymmetric things (or maybe the `*fet*` group...) are you sure that the gate is not separated? I had a problem with IEEEexplore and couldn't see the papers.

Rmano, 2023-02-04:
So, this is my proposal: I enhanced the `hemt` shape so that it can have an asymmetric and split gate and an additional arrow that follows (more or less) the standard option, although it will not show up unless set. This way, the circle, more or less, works, as well as the bodydiode.
Then we can add the style options for a default "GaN type asymmetric hemt", and if someone want different shapes they just change the style. 

What do you think? 

[image]

[image]

The full code:

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}

\makeatletter
\ctikzset{tripoles/hemt/width/.initial=.7}
\ctikzset{tripoles/hemt/gate height/.initial=.35}
\ctikzset{tripoles/hemt/base height/.initial=.5}
\ctikzset{tripoles/hemt/conn height/.initial=0}
\ctikzset{tripoles/hemt/height/.initial=1.1}
\ctikzset{tripoles/hemt/base width/.initial=.5}
\ctikzset{tripoles/hemt/gate width/.initial=.5}% the horizontal position is the same
\ctikzset{tripoles/hemt/bodydiode scale/.initial=.3}
\ctikzset{tripoles/hemt/bodydiode distance/.initial=.3}
\ctikzset{tripoles/hemt/bodydiode conn/.initial=.6}
\ctikzset{tripoles/hemt/curr direction/.initial=1}
\ctikzset{tripoles/hemt/gate asym/.initial=0}
\newif\ifpgf@circ@hemt@split
\ctikzset{tripoles/hemt/split gate/.is if=pgf@circ@hemt@split}
\ctikzset{tripoles/hemt/source arrow/.initial=0}

\pgfcircdeclaretransistor{hemt}{
        \anchor{inner up}{
            \northeast
            \pgf@y=\ctikzvalof{tripoles/hemt/gate height}\pgf@y
        }
        \anchor{inner down}{
            \northeast
            \pgf@y=-\ctikzvalof{tripoles/hemt/gate height}\pgf@y
        }
    }{%
    % add the circle if requested (before everything else, so we can fill it)
    \pgfcirc@transistorcircle
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@up+\pgfverticaltransformationadjustment*.5*\pgflinewidth}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}
    {\ctikzvalof{tripoles/hemt/gate height}\pgf@circ@res@up}}
    \pgfpathlineto{\pgfpoint
        {\ctikzvalof{tripoles/hemt/base width}\pgf@circ@res@left}
    {\ctikzvalof{tripoles/hemt/gate height}\pgf@circ@res@up}}

    \pgfpathmoveto{\pgfpoint
        {\ctikzvalof{tripoles/hemt/base width}\pgf@circ@res@left}
        {\ctikzvalof{tripoles/hemt/gate height}\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}
        {\ctikzvalof{tripoles/hemt/gate height}\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint
        {\pgf@circ@res@right}{\pgf@circ@res@down-\pgfverticaltransformationadjustment*.5*\pgflinewidth}}
    \pgfusepath{draw}
    \pgfscope
        \edef\@@tmp{\ctikzvalof{tripoles/hemt/source arrow}}
        \ifnum\@@tmp=0 \else
            \ifpgf@circuit@trans@arrowatend
                \ifnum\@@tmp > 0 %
                    \pgftransformshift{\pgfpoint
                        {\pgf@circ@res@right}
                        {\ctikzvalof{tripoles/hemt/gate height}\pgf@circ@res@down}}
[... 89 lines cut ...]
        ]  at (2,0) {Q};
\end{circuitikz}
\end{LTXexample}

\begin{LTXexample}[varwidth=true]
\begin{circuitikz}[]
    \path (-1,-1) rectangle (3,1); % BB
    \node  [hemt, tr circle, bodydiode](H) at (0,0) {Q};
    \node  [asy hemt, tr circle, bodydiode,
        ]  at (2,0) {Q};
\end{circuitikz}
\end{LTXexample}
% \showanchors[]{micshape}{}(north/90/0.4, north east/45/0.4, east/0/0.4,
%     south east/-45/0.4,
%     south/-90/0.4, south west/-135/0.4, west/180/0.4, north west/135/0.4,
%     left/160/0.4, right/30/0.4, center/-120/0.3
%     )
% \showanchors[]{tlmicshape}{}(north/90/0.4, north east/45/0.4, east/0/0.4,
%     south east/-45/0.4,
%     south/-90/0.4, south west/-135/0.4, west/180/0.4, north west/135/0.4,
%     left/160/0.4, right/30/0.4, center/-120/0.3
%     )

\end{document}
```

Rmano, 2023-02-04:
PS notice that the shape is not complete, I still need to adjust the gate anchors... But I just wanted to share the idea.

Rmano, 2023-02-05:
Currently, I have this:

[image]

would you think it would be better to call the style `gan hemt` (or `GaN hemt`)?

Rmano, 2023-02-06:
@epsilon-phi, if you are ok with this solution, I'll add it. I think I'll call the style `GaN hemt`, the other names are too generic.

epsilon-phi, 2023-02-06:
Sorry for my delay answering, I was away during the weekend.

Regarding disconnected gate: I have in fact not seen this representation yet, but i wont rule out its use. Better to have it than not :)

Regarding the new integration into the hemt class: I think this is perfect! In fact, this is more than i could have ever asked for. I am very exited that my suggestion was of actual use. On a more technical side: i think that your proposal is quite intuitive and easy for a user to understand with the documentation.

Regarding body diode: To my knowledge, it is inherent to a GaN HEMT that they do not have a body diode because of their different semiconductor layers compared to e.g a silicon MOSFET. So I think the diode option will see rare use, but then again: Better to have it than not.

The symmetrical version with split gate is especially thoughtful of you. Very nice.

There is one last thing I would request would be an asymmetrical d-mode version. However, this is easily possible with just

        \begin{circuitikz}
                \path (-1,-1) rectangle (5,1); % BB
                \node  [hemt](H) at (0,0) {Q};
                \node  [asy hemt,]  at (2,0) {Q};
                \node  [asy hemt,circuitikz/tripoles/hemt/split gate=false]  at (4,0) {Q};
        \end{circuitikz}

which yields
[image]
So this is amazing.

But in general: I am not only VERY ok with this solution, i am super happy! This will make our work quite a lot easier and I hope the works of some other people as well. Thank you so much for your support, I highly appreciate it and I am looking forward to the new version.

Rmano, 2023-02-06:
Thank you for the feedback. I'll add  the example for the asymmetrical d-mode too, and add this feature to the next release (1.6.1, which I hope to be able to do during this or the next week).

PR #695 (2023-02-07): Enhance `hemt` to allow, for example, GaN transistors.
Fixes #691

[image]
commit 6bfa4bd4 (2023-02-07): Enhance `hemt` to allow, for example, GaN transistors.

Fixes #691
closed 2023-02-07 by Rmano