circuitikz/circuitikz #694: [Feature] Correct Anchors for TL

myzinsky, 2023-02-05:
As you know, I'm using a lot of transmission lines: 

[image]

Sometimes I have to connect things to south west and south east. However, it connects to the rectangular boarding box. 

I tried to overwrite this, however, I know where the \anchor{...} stuff should go, however, I don't know how to adress the right coordinates:

``` LaTeX
 412 %% Node shapes for RF bipoles%<<<
 413 \ctikzset{bipoles/tline/height/.initial=.3}
 414 \ctikzset{bipoles/tline/width/.initial=.6}
 415
 416 \pgfcircdeclarebipolescaled{RF}
 417 { ###### INSERT ANCHOR STUFF HERE??? ######}
 418 {\ctikzvalof{bipoles/tline/height}}
 419 {tline}
 420 {\ctikzvalof{bipoles/tline/height}}
 421 {\ctikzvalof{bipoles/tline/width}}
 422 {
 423     \pgf@circ@res@step=.4\pgf@circ@res@up % the size of the ellipsis is proportional to the height
 424     \pgfscope
 425         \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
 426         \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{\pgf@circ@res@up}}
 427         \pgfpathlineto{\pgfpoint{\pgf@circ@res@left+\pgf@circ@res@step}{\pgf@circ@res@up}}
 428         \pgfpatharc{-90}{90}{-\pgf@circ@res@step and -\pgf@circ@res@up}
 429         \pgfpathlineto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{\pgf@circ@res@down}}
 430         \pgfpatharc{-90}{90}{\pgf@circ@res@step and \pgf@circ@res@up}
 431         \pgf@circ@draworfill
 432         \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{\pgf@circ@res@up}}
 433         \pgfpatharc{-90}{90}{-\pgf@circ@res@step and -\pgf@circ@res@up}
 434         \pgfusepath{stroke}
 435     \endpgfscope
 436     \pgfsetlinewidth{\pgfstartlinewidth}
 437     \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{0pt}}
 438     \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0pt}}
 439     \pgfusepath{stroke}
 440 }
```

any ideas how I could achieve a proper placement, like this: 

[image]

Rmano, 2023-02-05:
Like those?

[image]

```
\makeatletter
\pgfcircdeclarebipolescaled{RF}
{
    \anchor{center right}{\northeast \advance\pgf@x by -0.4\pgf@y\pgf@y=0pt}
    \anchor{top right}{\northeast \advance\pgf@x by -0.4\pgf@y}
    \anchor{bottom right}{\northeast \advance\pgf@x by -0.4\pgf@y\pgf@y=-\pgf@y}
    \anchor{center left}{\northeast \advance\pgf@x by -0.4\pgf@y\pgf@x=-\pgf@x\pgf@y=0pt}
    \anchor{top left}{\northeast \advance\pgf@x by -0.4\pgf@y\pgf@x=-\pgf@x}
    \anchor{bottom left}{\northeast \advance\pgf@x by -0.4\pgf@y\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
}
{\ctikzvalof{bipoles/tline/height}}
{tline}
{\ctikzvalof{bipoles/tline/height}}
{\ctikzvalof{bipoles/tline/width}}
{
    \pgf@circ@res@step=.4\pgf@circ@res@up % the size of the ellipsis is proportional to the height
    \pgfscope
        \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
        \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@left+\pgf@circ@res@step}{\pgf@circ@res@up}}
        \pgfpatharc{-90}{90}{-\pgf@circ@res@step and -\pgf@circ@res@up}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{\pgf@circ@res@down}}
        \pgfpatharc{-90}{90}{\pgf@circ@res@step and \pgf@circ@res@up}
        \pgf@circ@draworfill
        \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{\pgf@circ@res@up}}
        \pgfpatharc{-90}{90}{-\pgf@circ@res@step and -\pgf@circ@res@up}
        \pgfusepath{stroke}
    \endpgfscope
    \pgfsetlinewidth{\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@step}{0pt}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0pt}}
    \pgfusepath{stroke}
}
\makeatother
```

Rmano, 2023-02-05:
...and probably something like that should be added for the `tlinestub`...

Rmano, 2023-02-05:
...and by the way, `tlinestub` has wrong anchors since... well, forever, and nobody noticed...

[image]

myzinsky, 2023-02-05:
I did ... I also would need it in the other direction, will post a picture tomorrow

Rmano, 2023-02-05:
`xscale= -1` ?

myzinsky, 2023-02-05:
No, just the flat end without connection, and the connection hidden behind...

Rmano, 2023-02-06:
Ah, yes, I see... Anyway, the stub code is beyond salvation, probably the only way out is to add two "stubs" with correct anchors and with the same size parameters (height mainly, and the ellipse shape).

Rmano, 2023-02-06:
@myzinsky  ...and what about an option to *not* draw the short line on `tline`? That should solve everything with a couple of styles (rotating and mirroring, basically) with a lot of flexibility.

myzinsky, 2023-02-06:
Yes that would help a lot! Or just a parameter to normal tline, and I could use tlineshape 🤓

Rmano, 2023-02-06:
> Yes that would help a lot! Or just a parameter to normal tline, and I could use tlineshape nerd_face

Yes, this would be ok I think. I'll give it a shot. The only trick would be to move the `left` anchor to `center left` when the wire is not drawn, but that should be straightforward...

Rmano, 2023-02-06:
[image]

myzinsky, 2023-02-06:
Exactly that What I need. It works charming. Will that be part of the next release too?

Rmano, 2023-02-06:
> Exactly that What I need. It works charming. Will that be part of the next release too?

Yes!

Rmano, 2023-02-07:
After merging this, I have only #693  pending before releasing 1.6.1 --- do you prefer to let it for the next release?

PR #696 (2023-02-08): Add features to `tline`
Fixes #694

[image]
commit 9f6daa24 (2023-02-08): Add features to `tline`

Fixes #694
closed 2023-02-08 by Rmano