circuitikz/circuitikz #697: Alternative VarCap Style

myzinsky, 2023-02-11:
[image]

It would be nice to have this alternative varcap symbol. I tried to do it by myself, but there are very many ??? in the LED implementations. Would it make sense to take the base LED shape and base C shape and combine that?

Rmano, 2023-02-11:
Hmm. Why don't you use a subcircuit? Should be easy for this one.

Rmano, 2023-02-13:
```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage{circuitikz}

\begin{document}
\begin{tikzpicture}[stroke diodes, 
    mini c/.style={C, capacitors/scale=0.3}]
    \draw (0,0) to[D=$D_3$, name=d] ++(0,2) (d.se) 
        ++(0.2,0) to[mini c] ++(0,-.3);
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2023-04-04:
I will close this for now; I think it's not worth adding a shape for this (it's simple to add a macro or a subcircuit). Please feel free to re-open if needed.

closed 2023-04-04 by Rmano