circuitikz/circuitikz #703: [Bug] Body Diode size changes with "betterproportions"

sputeanus, 2023-03-18:
This is probably not a real bug but might be improved. I prefer the Diode shape of the "betterproportions" options. It think the Diodes look smoother with this. However, if I add a bodydiode to e.g. an IGBT the bodydiode is very small. (Probably becasue "betterproportions" renders all Diodes smaller.)

With betterproportions on
[image]

With betterproportions off
[image]

Thanks for making this Circuitikz package it already stayed with me long time and will continue to stay with me :)

Rmano, 2023-03-18:
`betterproportions` is really not fully supported. Have you tried using the "class" mechanism to change relative sizes? (Manual pages 38 and following). 

Anyway, *you* can change the factor used to reduce the diode size in the body diode implementation. It is not documented, and maybe I should add it to the "body diode customization" (not difficult at all; I just need the time to do it...). So I'll change this to an enhancement request!

```latex
\documentclass[border=2.72mm,preview]{standalone}

\usepackage[betterproportions]{circuitikz}

\begin{document}
\begin{circuitikz}
    \node [nigbt, bodydiode] at (0,0){};
    \ctikzset{tripoles/nigbt/bodydiode scale=.6}% default 0.3
    \node [nigbt, bodydiode] at (2,0){};
    \ctikzset{tripoles/nigbt/bodydiode scale=1}% default 0.3
    \node [nigbt, bodydiode, tr circle] at (4,0){};
\end{circuitikz}
\end{document}
```
[image]

Rmano, 2023-03-18:
As a reminder:
1. notice in the documentation that `betterproportion` is not supported anymore, one should use the class mechanism.
2. add a generic `bodydiode scale` option to the body diode customization.

sputeanus, 2023-03-18:
Reading in the style section i realised, that the "Romano" style also has more beautiful results for the Diode ;).
With the scaling of the Bodydiode size I get what I want, thank you very mucht!

Warm regards
Simon

sputeanus, 2023-03-18:
Hmmm is there also already a way to change the style of the body Diode to "Stroke Diode"?

Rmano, 2023-03-19:
> Hmmm is there also already a way to change the style of the body Diode to "Stroke Diode"?

Basically the body diode will follow your chosen style:

```latex
\documentclass[border=2.72mm,preview]{standalone}

\usepackage{circuitikz}

\begin{document}
\begin{circuitikz}[stroke diodes]
    \node [nigbt, bodydiode] at (0,0){};
    \ctikzset{tripoles/nigbt/bodydiode scale=.6}% default 0.3
    \node [nigbt, bodydiode] at (2,0){};
\end{circuitikz}
\end{document}
```

[image]

sputeanus, 2023-03-20:
Perfetto!
Merci beaucoup :)

PR #709 (2023-04-04): Add `scale` to `transistor bodydiode` options
Closes #703

[image]

[image]
commit 74989681 (2023-04-04): Add `scale` to `transistor bodydiode` options

Closes #703