circuitikz/circuitikz #704: jump crossing different color for wire

lkjell, 2023-03-19:
Hi,

I am using node[jump crossing, color=blue] {}
the whole crossing has color blue. Is there a way to modify different colors for the wires? 
If it is can you please put it in doc manual?

Rmano, 2023-03-19:
I'm sorry but no, vertical and horizontal wires are of the same color by design. It could be done... Basically 2ith an option for changing, for example, ten vertical wire color.

I'll look into it when I'll have time (don't hold your breath 😉)

Rmano, 2023-03-19:
For the time being, you can use this quick hack:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\makeatletter
\tikzset{rainbow crossing/vertical color/.initial = .}
\pgfdeclareshape{rainbow crossing}
{
    \savedanchor\northwest{%
        \pgf@y=\ctikzvalof{bipoles/crossing/size}\pgf@circ@Rlen
        \pgf@y=.5\pgf@y
        \pgf@x=-\pgf@y
    }
    \anchor{center}{ \pgf@y=0pt \pgf@x=0pt }
    \anchor{east}{ \northwest \pgf@y=0pt \pgf@x=-\pgf@x  }
    \anchor{e}{ \northwest \pgf@y=0pt \pgf@x=-\pgf@x  }
    \anchor{west}{ \northwest \pgf@y=0pt }
    \anchor{w}{ \northwest \pgf@y=0pt }
    \anchor{south}{ \northwest \pgf@x=0pt \pgf@y=-\pgf@y }
    \anchor{s}{ \northwest \pgf@x=0pt \pgf@y=-\pgf@y }
    \anchor{north}{ \northwest \pgf@x=0pt }
    \anchor{n}{ \northwest \pgf@x=0pt }
    \anchor{south west}{ \northwest \pgf@y=-\pgf@y }
    \anchor{north east}{ \northwest \pgf@x=-\pgf@x }
    \anchor{north west}{ \northwest }
    \anchor{south east}{ \northwest \pgf@x=-\pgf@x \pgf@y=-\pgf@y }
    \pgf@circ@draw@component{
        \northwest
        \pgf@circ@res@up = \pgf@y
        \pgf@circ@res@down = -\pgf@y
        \pgf@circ@res@right = -\pgf@x
        \pgf@circ@res@left = \pgf@x
        % horizontal jumper
        \pgfsetbeveljoin
        \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{0pt}}
        \pgfpathlineto{\pgfpoint{0.4\pgf@circ@res@left}{0pt}}
        \pgfpatharc{0}{-180}{0.4*\pgf@circ@res@left}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0pt}}
        \pgfusepath{draw}
        % vertical, broken path
        \pgfsetcolor{\pgfkeysvalueof{/tikz/rainbow crossing/vertical color}}
        \pgfpathmoveto{\pgfpoint{0pt}{\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{0pt}{0.5\pgf@circ@res@up}}
        \pgfpathmoveto{\pgfpoint{0pt}{0.3\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{0pt}{\pgf@circ@res@down}}
        \pgfusepath{draw}
    }
}
\makeatother
\begin{document}
\begin{tikzpicture}[]
    \draw (0,0) node[jump crossing, color=blue]{}; 
    \draw (1,0) node[rainbow crossing, color=blue, rainbow crossing/vertical color=black]{}; 
\end{tikzpicture}
\end{document}
```

[image]

Notice that changing *only* the loop is not easy, because the horizontal + loop part is drawn in one path (to have nice line joins) and you can't change the color of just part of a path.

lkjell, 2023-03-19:
Thanks. This will be merged later on?

Rmano, 2023-03-19:
> Thanks. This will be merged later on?

No, not in this form. I will add an option to the `jump crossing` component; I do not want to add a new one. And possibly add a full configurability for the vertical stroke, like color/thickness/dash pattern. 

But this solution is independent, I mean: if you keep the code between `\makeatletter` and `\makeatother` in your preamble, it will not stop working.

Rmano, 2023-04-05:
In the end, this is the final solution:

[image]