circuitikz/circuitikz #705: [Feature] Clock wedge component

Mario1159, 2023-03-31:
It would be useful to have a clock wedge component to able to implement clocked custom digital blocks such as this example:

[image]

This block could be done similar to the 555 example in section 4.24.5 "Mux-Demux special usage" but it needs a clock wedge symbol which could be similar to the signal ground node but tailless.

Rmano, 2023-04-02:
Yes, it could be useful (although it is pretty trivial to add manually...). I'll add it to the list of things nice to do 😉 — I have little time now and `context` new version created a big disruption...

Rmano, 2023-04-03:
For example, you can use this: 

```latex
\documentclass[border=2.72mm,preview]{standalone}
\usepackage{circuitikz}
\begin{document}
\newcommand{\wedgedown}{+(-0.15,0) -- +(0,-0.15) -- +(0.15,0)}
\begin{circuitikz}
    \node [muxdemux, muxdemux def={Lh=3, Rh=3, NT=2}](A) at (0,0){};
    \draw (A.btpin 1) \wedgedown;
\end{circuitikz}
\end{document}
```

[image]

Rmano, 2023-04-04:
It is quite complex to define a shape that always has the same size as the wedge in flip-flops, so I am not sure if it's worth it...

PR #710 (2023-04-04): Add `clockwedge` shape
Closes #705

[image]
commit ef206520 (2023-04-04): Add `clockwedge` shape

Closes #705
closed 2023-04-04 by Rmano