circuitikz/circuitikz #706: [Bug] circuitikz sort-of stopped working with context

Rmano, 2023-04-03:
In March 2023, with TeXLive 2023, `context` went through a big change (the basic engine changed from the traditional one to a new, much more modern one).

In the process, the `circuitikz` example completely stopped working. The first problem is related to [this problem with Ti*k*Z](https://github.com/pgf-tikz/pgf/issues/1248#issuecomment-1486235591), and can be fixed by adding

     \usemodule[pgfrcs,pgfmat]

 I have no clue how to fix this; moving the following line *inside*  `t-circuitikz.tex` does not work. 

Moreover, `t-circuitikz` loads, by default, several macros that mimic some of the `siunitx` behavior. I never really liked them, but now they stopped working too, for reasons I do not follow.

For the moment, I will make the changes in the test file as a workaround so that the compilation test passes in old and new `context`, and add a notice to the manual and the release (when it happens). 

I'd really like to fix this so that it works correctly with `context`, but I haven't the knowledge needed. If there are `context` users that use `circuitikz` out there willing to help, please add to this issue.

adityam, 2023-04-28:
Not a circuitikz user but ...

> Moreover, t-circuitikz loads, by default, several macros that mimic some of the siunitx behavior. I never really liked them, but now they stopped working too, for reasons I do not follow.

I suspect that `m-circuitikz` file provided by ConTeXt is loaded instead of `t-circuitikz` file provided by the package. In this version of the module, the "manual" unit definitions are commented out with the following comment:

```
% hm, we have units since '97 so one can load the units module in mkii or
% assume it is present (mkiv and therefore lmtx)
```

I'll try to see why circuitikz is failing with latest ConTeXt.

Rmano, 2023-04-28:
Thank you for looking at it!

adityam, 2023-04-28:
Does it work if you add the following file (saved as `m-tikz.mkxl`  in the current directory):

```
%D \module
%D   [       file=m-tikz,
%D        version=2021.07.12,
%D          title=\CONTEXT\ Extra Modules,
%D       subtitle=TIKZ support,
%D         author=Hans Hagen,
%D           date=\currentdate,
%D      copyright={PRAGMA ADE \& \CONTEXT\ Development Team}]
%C
%C This module is part of the \CONTEXT\ macro||package and is
%C therefore copyrighted by \PRAGMA. See mreadme.pdf for
%C details.

%D A fixed variant of the t-tikz module distributed with tikz. For practical reasons
%D per 2021 we split the module for \MKIV\ and \LMTX. (Maybe some day I'll optimize
%D TIKZ a bit.)
%D
%D All these t-* modules in the tikz distribution make no sense as we can just load
%D them here which is way cleaner. There is also some mkii filename mapping going on
%D that is not needed on todays systems (windows has no 8.3 limitation and on unix
%D tds finally adopted a case insensitive approach.)

%D Just in case someone does a hard input ...

\ifdefined\pgfdefined    \endinput \fi
\ifdefined\pgfcoredefined\endinput \fi

%D ... but normally we end up here:

\startmodule[tikz]

\ifdefined\pdflastxpos \else
    \unprotect
        \frozen\overloaded\protected\def\pdflastxpos{\numexpr\clf_lastxpos\relax}
        \frozen\overloaded\protected\def\pdflastypos{\numexpr\clf_lastypos\relax}
    \protect
\fi

\newcatcodetable \tikzcatcodes

\startcatcodetable \tikzcatcodes
    \catcode\tabasciicode             \spacecatcode
    \catcode\endoflineasciicode       \endoflinecatcode
    \catcode\formfeedasciicode        \endoflinecatcode
    \catcode\spaceasciicode           \spacecatcode
    \catcode\endoffileasciicode       \ignorecatcode
    \catcode\circumflexasciicode      \superscriptcatcode
    \catcode\underscoreasciicode      \subscriptcatcode
    \catcode\ampersandasciicode       \alignmentcatcode
    \catcode\backslashasciicode       \escapecatcode
    \catcode\leftbraceasciicode       \begingroupcatcode
    \catcode\rightbraceasciicode      \endgroupcatcode
    \catcode\dollarasciicode          \mathshiftcatcode
    \catcode\hashasciicode            \parametercatcode
    \catcode\commentasciicode         \commentcatcode
    \catcode\atsignasciicode          \lettercatcode
    \catcode\exclamationmarkasciicode \othercatcode
    \catcode\questionmarkasciicode    \lettercatcode
    \catcode\tildeasciicode           \activecatcode
    \catcode\barasciicode             \othercatcode
\stopcatcodetable

\permanent\protected\def\starttikzinput
  {\pushoverloadmode
   \pushcatcodetable
   \setcatcodetable\tikzcatcodes
   \pushmacro\meaning
   \let\meaning\meaningless
   \autoparagraphmode\zerocount}

\permanent\protected\def\stoptikzinput
  {\autoparagraphmode\plusone
   \popcatcodetable
   \popmacro\meaning
   \popoverloadmode}

\permanent\protected\def\tikzinputfile#1%
  {\starttikzinput
   \input{#1}\relax
   \stoptikzinput}

\permanent\protected\def\tikzerrormessage#1#2#3%
  {\writestatus{#1}{#2}}

% For now we need this but we need to educate the user to wrap settings in the
% wrappers. So some day the next line will go. I need to check what commands are
% possible outside a picture.

\overloadmode\zerocount

\newtoks\everytikzpicture

% \appendtoks
%     \resetcharacterspacing
% \to \everytikzpicture

\permanent\protected\def\starttikzsettings
  {\pushoverloadmode
   \pushmacro\meaning
   \let\meaning\meaningless
   \autoparagraphmode\zerocount}

\permanent\protected\def\stoptikzsettings
  {\autoparagraphmode\plusone
   \popmacro\meaning
   \popoverloadmode}

\permanent\protected\def\starttikzpicture
  {\dontleavehmode
   \hcontainer\bgroup
   \setcatcodetable\tikzcatcodes
   \the\everytikzpicture
   \autoparagraphmode\zerocount
   \pushmacro\meaning
   \let\meaning\meaningless
 % \pushoverloadmode
   \ifdefined\PackageError\else \let\PackageError\tikzerrormessage \fi
   \tikzpicture}

\permanent\protected\def\stoptikzpicture
  {\endtikzpicture
 % \popoverloadmode
   \popmacro\meaning
   \egroup}

\let\pgfdefined    \relax
\let\pgfcoredefined\relax

\tikzinputfile{pgfutil-common.tex}
\tikzinputfile{pgfutil-context.def}
\tikzinputfile{pgfrcs.code.tex}
\tikzinputfile{pgfsys.code.tex}
\tikzinputfile{pgfkeys.code.tex}
\tikzinputfile{pgfsyssoftpath.code.tex}
\tikzinputfile{pgfsysprotocol.code.tex}
\tikzinputfile{pgfcore.code.tex}
\tikzinputfile{pgffor.code.tex}
\tikzinputfile{pgfmoduleplot.code.tex}
\tikzinputfile{tikz.code.tex}

\let\startpgfpicture             \pgfpicture              \let\stoppgfpicture            \endpgfpicture
\let\startpgfscope               \pgfscope                \let\stoppgfscope              \endpgfscope
\let\startpgflowlevelscope       \pgflowlevelscope        \let\stoppgflowlevelscope      \endpgflowlevelscope
\let\startpgfinterruptpath       \pgfinterruptpath        \let\stoppgfinterruptpath      \endpgfinterruptpath
\let\startpgfinterruptpicture    \pgfinterruptpicture     \let\stoppgfinterruptpicture   \endpgfinterruptpicture
\let\startpgfinterruptboundingbox\pgfinterruptboundinbox  \let\stoppgfinterruptboudingbox\endpgfinterruptboundingbox

\let\normalusepgfmodule  \usepgfmodule
\let\normalusepgflibrary \usepgflibrary
\let\normalusetikzlibrary\usetikzlibrary

\tolerant\protected\def\usepgfmodule[#1]#;#2% somehow both variants are used
  {\starttikzinput
   \normalusepgfmodule[#1#2]%
   \stoptikzinput}

\tolerant\protected\def\usepgflibrary[#1]#;#2% somehow both variants are used
  {\starttikzinput
   \normalusepgflibrary[#1#2]%
   \stoptikzinput}

\tolerant\protected\def\usetikzlibrary[#1]#;#2% somehow both variants are used
  {\starttikzinput
   \normalusetikzlibrary[#1#2]%
   \stoptikzinput}

\usepgfmodule[shapes]
\usepgfmodule[plot]
\usepgfmodule[matrix]
\usepgfmodule[decorations]

\definefilesynonym [pgfplots]      [tikz-pgfplots]
\definefilesynonym [pgfplotstable] [tikz-pgfplotstable]

\stopmodule
```

adityam, 2023-04-28:
If not, I'd appreciate if you could provide MWE that fails. I don't know circuitikz so it is difficult for me to test all types of usage.

Rmano, 2023-04-29:
Ok, I'll try --- tomorrow,  I hope --- I'm just now on a short holiday and I have little time connected...

Thanks!

Rmano, 2023-04-29:
The example code below **do** work if I add the code you posted above (and fails without it).

But the (commented out) code after it, using the "poor-man" simulation of `siunitx`, still does not work (it worked fine in TeXLive 2022). Not a big problem; I understand that `context` is a rapidly moving target and that the "full backward compatibility" policy of `circuitikz` can't work here.

```
\usemodule[circuitikz]

\starttext

A simple example to test the installation.

% removed all the "pseudo-SI commands". They should work, but they
% stopped working when context passed from mkiv to ltmx (March 2023)
\startcircuitikz[scale=1.2]
	\draw
  (0,2) to[I=1~mA] (2,2)
        to[R, l_=2~kΩ, *-*] (0,0)
        to[R, l_=2~kΩ] (2,0)
        to[V, v_=2~V] (2,2)
        to[cspst, l=$t_0$] (4,2) -- (4,1.5)
        to [generic, i=$i_1$, v=$v_1$] (4,-.5) -- (4,-1.5)
  (0,2) -- (0,-1.5) to[V, v_=4~V] (2,-1.5)
        to [R, l=1~kΩ] (4,-1.5)
  (5,2) node[dipchip, anchor=pin 1]{}
  (5,-2) node[flipflop JK, anchor=pin 1]{};

\stopcircuitikz

% \startcircuitikz[scale=1.2]
% 	\draw
%   (0,2) to[I=1\milli\ampere] (2,2)
%         to[R, l_=2\kilo\ohm, *-*] (0,0)
%         to[R, l_=2\kilo\ohm] (2,0)
%         to[V, v_=2\volt] (2,2)
%         to[cspst, l=$t_0$] (4,2) -- (4,1.5)
%         to [generic, i=$i_1$, v=$v_1$] (4,-.5) -- (4,-1.5)
%   (0,2) -- (0,-1.5) to[V, v_=4\volt] (2,-1.5)
%         to [R, l=1\kilo\ohm] (4,-1.5)
%   (5,2) node[dipchip, anchor=pin 1]{}
%   (5,-2) node[flipflop JK, anchor=pin 1]{};

% \stopcircuitikz

\stoptext
```

adityam, 2023-05-01:
> The example code below **do** work if I add the code you posted above (and fails without it).

Great! This (or a modified version of it) would be included in ConTeXt, so circuitikz should start working with context again.

> But the (commented out) code after it, using the "poor-man" simulation of `siunitx`, still does not work (it worked fine in TeXLive 2022). 

That should be simpler to use. I don't use context's unit module often enough. Let me look into it and suggest a workaround.

adityam, 2023-05-02:
Would it be okay to use units as follows:

    \usemodule[circuitikz]

    \starttext

    \startcircuitikz[scale=1.2]
        \draw
        (0,2) to[I=\unit{1 milli ampere}] (2,2)
              to[R, l_=\unit{2 kilo ohm}, *-*] (0,0)
              to[R, l_=\unit{2 kilo ohm}] (2,0)
              to[V, v_=\unit{2 volt}] (2,2)
              to[cspst, l=$t_0$] (4,2) -- (4,1.5)
              to [generic, i=$i_1$, v=$v_1$] (4,-.5) -- (4,-1.5)
        (0,2) -- (0,-1.5) to[V, v_=\unit{4 volt}] (2,-1.5)
              to [R, l=\unit{1 kilo ohm}] (4,-1.5)
        (5,2) node[dipchip, anchor=pin 1]{}
        (5,-2) node[flipflop JK, anchor=pin 1]{};

    \stopcircuitikz

    \stoptext

Basically, rather than providing a poor man's version of units, let ConTeXt user handle units using ConTeXt. It is a bit more verbose, but it is consistent with how units are handled everywhere else in ConTeXt, so a ConTeXt user should be used to that. You can perhaps add a link to the [context units manual](http://www.pragma-ade.nl/general/manuals/units-mkiv.pdf) in the documentation.

adityam, 2023-05-03:
> do you know when the change for the Ti_k_Z part will be sent out?

I have asked on the context mailing list for the new version of m-tikz to be merged. I expect that it should be part of the next release (not sure when that will be).

> But I still would like to know why the code here:
> 
> https://github.com/circuitikz/circuitikz/blob/4f9fd7e956aa4275890254593a23eb52a97d967f/tex/t-circuitikz.tex#LL52C1-L60C1
> 
> https://github.com/circuitikz/circuitikz/blob/4f9fd7e956aa4275890254593a23eb52a97d967f/tex/t-circuitikz.tex#LL98C1-L120C39
> 
> seems to have no effect whatsoever...

I apologize in advance about the long and convoluted state of affairs. First, this is what is happening. ConTeXt includes a `m-circuitikz.mkxl` file as part of the distribution. (the `m-` namespace is for modules distributed as part of context while the `t-` namespace is for third party modules). When you call

```
\usemodule[circuitikz]
```

context first searches for `m-circuitikz` and then for `t-circuitikz`. Since there is a `m-circuitikz.mkxl` that is found and used. 

Now, you might genuinely ask, why does context include `m-circuitikz`! It has to do with the fact that the luatex and luametatex introduce a lot of low level changes in tex primitives. So, if used without any change, `t-tikz` will not work in luametatex. So, context provides a wrapper around the popular tikz modules to make sure that they work in context (this latest issue notwithstanding). 

So, it you compare `t-circuitikz` with [`m-circuitikz`](https://github.com/contextgarden/context-mirror/blob/19750b667c23a9f276032837d6c517063376bd0c/tex/context/modules/mkxl/m-circuitikz.mkxl) (which is included as part of context), you will see that `m-circuitikz` has commented out the poor man version of the units module (my guess is because context already has a well integrated mechanism to include units). 

So hopefully this explains _why_ your code was not working. I am not sure what is the best way to _fix_ this. 

1. You could ask (and I can do this on your behalf) context maintainers not to include a patched up version of circuitikz loader in context and distribute MKIV and LMTX flavors of the module as part of circuitikz. 

2. If you really want a latex compatibility layer, it could be distributed as a separate (sub)-module (or pgf/tikz-library.

Rmano, 2023-05-05:
Thanks, now I understand. So really, including the `t-circuitikz.tex` file is not useful at all ;-) (BTW, it seems that the two m-circuitikz.*  files are identical in my distribution...)

I don't know how to proceed. On the one side, I would like to maintain control of the files, but on the other side, I have zero knowledge of  ConTeXT, so I would not be able to keep them up-to-date; in that sense, I'm happy having the files managed by knowledgable people (I can even not detect how those files pick up the current version...).

I think that for now, I'll simply say that `siunitx` option is not supported in ConTeXt, link to the `units` link you gave to me, and leave the hack in the test file so that it will not fail the checks while the new `m-tikz` percolate down.

Rmano, 2023-05-07:
Done; I'll leave this open for now. https://github.com/circuitikz/circuitikz/pull/715

adityam, 2023-05-08:
> I think that for now, I'll simply say that siunitx option is not supported in ConTeXt, link to the units link you gave to me, and leave the hack in the test file so that it will not fail the checks while the new m-tikz percolate down.

I agree that Please feel free to ping me when you update the documentation and I can take a look to verify everything. 

The new m-tikz is now part of context distribution and circuitikz works fine in the context lmtx distribution. I don't use texlive, so don't know when texlive will be updated.

Rmano, 2023-05-12:
@adityam  I think I have it sorted out with #715; I hope to release 1.6.2 soon (maybe this weekend). Suggestions welcome!

Rmano, 2023-05-12:
@adityam  I think that #716 closes it; TeXLive now works correctly too!

Thanks for the assistance!

closed 2023-05-12 by Rmano

adityam, 2023-05-12:
The description in the updated documentation is good!

PR #716 (2023-05-12): ConTeXt up and running again!
Thanks to Aditya Mahajan for assistance!

Fixes #706
commit 6e94b3a8 (2023-05-12): ConTeXt up and running again!

Thanks to Aditya Mahajan for assistance!

Fixes #706