circuitikz/circuitikz #712: [Feature] Add size of components in documentation

Firestar-Reimu, 2023-04-06:
What is the default size of various components in circuitikz?

for example:

- the length of a resistor (and other bipoles)?
- if `node[npn]{Q}` is at `(0,0)`, what is the **absolute coordinate** of `Q.b`/`Q.c`/`Q.e`?
- if `node[op amp]{OA}` is at `(0,0)`, what is the **absolute coordinate** of `OA.-`/`OA.+`/`OA.out`/`OA.up`/`OA.down`?

Thanks! So that there is no need of **MANY MANY MANY relative coordinates** when using transistors or amplifiers.

Rmano, 2023-04-06:
This is impossible, as I answered in https://tex.stackexchange.com/a/682025/38080. The position depends on tens of parameters, and it's practically impossible to have it fixed; often they are not "nice numbers". You can always find the positions, but alas, *why*? Using the anchors it's easier, faster, and make the circuit easily adjustable.

How to find the coordinate:
```latex
\documentclass[border=2.72mm,preview]{standalone}
\usepackage{circuitikz}
\makeatletter
\def\extractcoord#1#2#3{% https://tex.stackexchange.com/a/179946/38080
  \path let \p1=(#3) in \pgfextra{
    \pgfmathsetmacro#1{\x{1}/\pgf@xx}
    \pgfmathsetmacro#2{\y{1}/\pgf@yy}
    \xdef#1{#1} \xdef#2{#2}
  };
}
\makeatother
\begin{document}
\begin{circuitikz}[]
    \draw [cyan!50, ultra thin] (-1,-1) grid[step=1mm] (2,1);
    \draw [cyan, thin] (-1,-1) grid (2,1);
    \node [npn](N1) at (0,0) {Q};
    \extractcoord\nx\ny{N1.C}
    \node [above, red, font=\tiny\ttfamily] at (N1.C) {\nx,\ny};
    \ctikzset{tripoles/npn/height=0.8}% internal parameter, beware
    \node [npn](N2) at (1.5,0) {Q};
    \extractcoord\nx\ny{N2.C}
    \node [above, red, font=\tiny\ttfamily] at (N2.C) {\nx,\ny};
\end{circuitikz}
\end{document}
```

[image]

***CONCLUSION:*** Use the anchors, Luke...

Rmano, 2023-04-06:
I'll close this, as "won't fix". If you think you have a way to provide the coordinates, please re-open the issue! (And better if you can provide a PR... 😉).

closed 2023-04-06 by Rmano