circuitikz/circuitikz #722: [Feature] oosourcetrans anchors

Tipounk, 2023-05-31:
Hi everyone and first of all thanks for the work you do on the package.

Would it be possible to add anchors for the single phase round transformer shape (oosourcetrans) ? I wasn't able to find it.

[image]

For example it could be interesting to have a prim1, prim2 and sec1, sec2 anchors just like for the ooosource. I often prefer to use this rounded transformer shape rather than the one with inductors in some simple diagram.

Thanks,

Rmano, 2023-06-01:
On second thought, you really don't need anchors (although they can be handy) because you have the main one, that this, the center of the circle and the left and right positon; that enables you to reach any point on the circle with [partway modifiers](https://tikz.dev/tikz-coordinates#sec-13.5.3):

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}[]
    \path (0,0) to[oosourcetrans, name=T]++(2,0);
    % \node [circ, red] at (T.left) {};
    \draw ($(T.centerprim)!1!45:(T.left)$) -- ++(-.5,-.5) node[ground]{};
    \begin{scope}[font=\tiny\ttfamily]
        \foreach \a in {-90,-45,...,90}
            \node [circ, scale=0.5, label=\a:\a, color=red] at
                ($(T.centersec)!1!\a:(T.right)$){};
    \end{scope}
\end{tikzpicture}
\end{document}
```

[image]

What do you think? Maybe I can add this trick (or example) to the manual and avoid adding anchors that will be more or less arbitrary...

Tipounk, 2023-06-01:
Hi,
Thanks for the tip with partway modifiers, it will do the trick i think. But adding an example in the manuel seems to be a good idea for the beginners like me.

For information I haven't found this transformer symbol with four ports in a standard but i tried to measure the symbol used by simulation softwares like plecs and the anchors used is about 40 degrees.

PR #723 (2023-06-02): Document how to contact the border of sources.
Fixes #722

[image]
commit 98b09c19 (2023-06-02): Document how to contact the border of sources.

Fixes #722
closed 2023-06-02 by Rmano