circuitikz/circuitikz #725: [Feature] oosourceshape secondary grounded

lapreindl, 2023-06-05:
Hi there,

thank you for your great work!
I have the following feature request:
Sometimes I need to add a secondary neutral node to a switchgear circuit, like
[image]

I quickly solved this problem by adding the following to `pgfcircbipoles.tex`
```
\anchor{centersec2}{
    \northeast
    \pgf@y=0pt\relax
    \pgf@x=\ctikzvalof{bipoles/oosourcetrans/circlesize}\pgf@x
}
```
May you can add this as a coordinate or as a fix implemented ground symbol to `wye` and `zig`.

Sorry for my bad English ;-),
Thanks, Laurenz!

Rmano, 2023-06-06:
Hmmm... if I understand correctly, you are asking for an anchor for the center position of the optional symbol? (Your code works, but a bit due to chance, I fear 😉)

lapreindl, 2023-06-06:
Thanks,
yeah right, I'm asking for an anchor of the center position of the symbols `wye` and `zig`.

Laurenz

Rmano, 2023-06-06:
Ok, I'll have a look, and I'll implement them when I have a bit of time.

PR #727 (2023-06-18): Add `oosourcetrans` and `oosource` symbol anchors
Slightly change shape and positions for symbols

[image]

@olfline  --- ok with you?

Fixes #725
commit 718221c4 (2023-06-18): Add `oosourcetrans` and `oosource` symbol anchors

Slightly change shape and positions for symbols

Fixes #725
closed 2023-06-18 by Rmano