circuitikz/circuitikz #730: [Bug]

sputeanus, 2023-08-09:
Hey guy,

I am currently trying to draw a European Nand which an scaling of 2 and I realized that the And symbol in the middle is not scaling accordingly.

I already created a workaround but I am curious wheter this is a bug or I did not read the manual properly.

`\documentclass[english,10pt,a4paper]{article}
\usepackage[utf8]{inputenc}
\usepackage[T1]{fontenc}
\usepackage{babel}
\usepackage{tikz}
\usepackage{circuitikz}
%\ctikzsetstyle{romano}
\begin{document}
	\begin{tikzpicture}[line width=1.5pt]%,font = \huge]
		\ctikzset{
			logic ports/scale=2,
			tripoles/european not symbol=ieee circle,
			tripoles/thickness=1,
			bipoles/thickness=1,
			font=\Huge
		}
	
	% Draw the Nand with AND Symbol
	\draw(0,4cm) node [european nand port] (NAND1) {}
	;
	\draw (NAND1.out) to [european resistor, l=$R$] ++(3cm,0)
	;

	
	% Draw the Nand with big AND symbol workaround
	\draw(0,0) node [european nand port] (NAND2) {\colorbox{white}{\&}}
	;
	\draw (NAND2.out) to [european resistor, l=$R$] ++(3cm,0)
	;

	\end{tikzpicture}
\end{document}`

[image]

Thanks for your help!
Simon

sputeanus, 2023-08-09:
Ahh and maybe just another question.

Is it true that I can just change the width of the american logic ports but not the width of european logic ports?
When I use

tripoles/european nand port/port width=.4
instead of 
tripoles/american nand port/port width=.4

I get an error message that he does not know the key

Rmano, 2023-08-09:
Indeed, I didn't give the European ports a lot of love. They were (re-)implemented by John Kormylo to add multiple inputs, and I just added them to the class mechanism and made changes to implement the various "not" markers...

For the first question, it's true that the font is fixed at the standard size. I missed it when adding the scaling, but on the other side, generally, scaling fonts is not advisable. So my proposal is to add an option to set the font. In the following code, you have a temporary code that is doing the same, just as a stopgap solution.

The problem with the width is that the parameter's name is different. Maybe a should add a "class" meta-parameter for it... 

```latex
\documentclass[english,10pt,a4paper]{article}
\usepackage[utf8]{inputenc}
\usepackage[T1]{fontenc}
\usepackage{babel}
% \usepackage{tikz}
\usepackage{circuitikz}

% new font parameter, start code
\usepackage{etoolbox}
\makeatletter
\ctikzset{tripoles/european port font/.initial={}}
%% patch the internal command to add the options
\patchcmd{\pgfcircdeclareeurologicport}%
    {\pgftext{#2}}%
    {\pgftext{\ctikzvalof{tripoles/european port font}#2}}%
    {\relax}{\ERRORPACHING}
%% redeclare the ports
\pgfcircdeclareeurologicport{and}{\&}{\pgf@circ@res@count}{}
\pgfcircdeclareeurologicport{or}{$\ge 1$}{\pgf@circ@res@count}{}
\pgfcircdeclareeurologicport{xor}{$=1$}{\pgf@circ@res@count}{}
\pgfcircdeclareeurologicport{not}{$1$}{1}{not}
\pgfcircdeclareeurologicport{buffer}{$1$}{1}{}
\pgfcircdeclareeurologicport{nand}{\&}{\pgf@circ@res@count}{not}
\pgfcircdeclareeurologicport{nor}{$\ge 1$}{\pgf@circ@res@count}{not}
\pgfcircdeclareeurologicport{xnor}{$=1$}{\pgf@circ@res@count}{not}
\makeatother
% end font parameter code

\begin{document}
\begin{tikzpicture}[line width=1.5pt]%,font = \huge]
    \ctikzset{
        logic ports/scale=2,
        tripoles/european not symbol=ieee circle,
        tripoles/thickness=1,
        bipoles/thickness=1,
        % font=\Huge not used
    }

    % Draw the Nand with AND Symbol
    \draw(0,4cm) node [european nand port,
        circuitikz/tripoles/european nand port/width=2.2] (NAND1) {}
        ;
    \draw (NAND1.out) to [european resistor, l=$R$] ++(3cm,0)
        ;

    % Draw the Nand with big AND symbol workaround
    \ctikzset{tripoles/european port font=\Huge}
    \draw(0,0) node [european nand port] (NAND2) {}
        ;
    \draw (NAND2.out) to [european resistor, l=$R$] ++(3cm,0)
        ;

\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2023-08-09:
Hmmm... but yes, there is something pesky about the width --- center os not center anymore. 
I'll investigate more!

Rmano, 2023-08-10:
Nicely seen. Yes, the workaround is relatively simple, I think; can you try the following?:

```latex
\documentclass[english,10pt,a4paper]{article}
\usepackage[utf8]{inputenc}
\usepackage[T1]{fontenc}
\usepackage{babel}
% \usepackage{tikz}
\usepackage{circuitikz}

% new font parameter, start code
\usepackage{etoolbox}
\makeatletter
\ctikzset{tripoles/european port font/.initial={}}
%% patch the internal command to add the options
\patchcmd{\pgfcircdeclareeurologicport}%
    {\pgftext{#2}}%
    {\pgfscope
        \pgftransformresetnontranslations
        \pgftext{\ctikzvalof{tripoles/european port font}#2}%
    \endpgfscope}%
    {\relax}{\ERRORPACHING}
%% redeclare the ports
\pgfcircdeclareeurologicport{and}{\&}{\pgf@circ@res@count}{}
\pgfcircdeclareeurologicport{or}{$\ge 1$}{\pgf@circ@res@count}{}
\pgfcircdeclareeurologicport{xor}{$=1$}{\pgf@circ@res@count}{}
\pgfcircdeclareeurologicport{not}{$1$}{1}{not}
\pgfcircdeclareeurologicport{buffer}{$1$}{1}{}
\pgfcircdeclareeurologicport{nand}{\&}{\pgf@circ@res@count}{not}
\pgfcircdeclareeurologicport{nor}{$\ge 1$}{\pgf@circ@res@count}{not}
\pgfcircdeclareeurologicport{xnor}{$=1$}{\pgf@circ@res@count}{not}
\makeatother
% end font parameter code

\begin{document}
\begin{tikzpicture}[line width=1.5pt]%,font = \huge]
    \ctikzset{
        logic ports origin=center,
        logic ports/scale=2,
        tripoles/european not symbol=ieee circle,
        tripoles/thickness=1,
        bipoles/thickness=1,
    }

    % Draw the Nand with big AND symbol workaround
    \ctikzset{tripoles/european port font=\Huge}
    \draw(0,0) node [european nand port, rotate=180] (NAND1) {};
    \draw(0,-3) node [european nand port, rotate=90] (NAND2) {};
    \draw(3,-3) node [european nand port, xscale=-1] (NAND3) {};

\end{tikzpicture}
\end{document}
```

[image]

That way, the text in the European port should always be "straight."  And, by the way,  the problem with the center position is an old one, solved by setting `logic ports origin=center`.

Rmano, 2023-08-10:
Although maybe it would be also useful to define a couple of "blank" blocks, one negated and the other "straight", so that one can add the symbol/text/whatever in the box?

sputeanus, 2023-08-10:
Yeah that would be nice if you would be able just to fill in your own symbol. Like witch the american ones.
[image]

sputeanus, 2023-08-10:
But the workaround runs smootly :)
[image]

Rmano, 2023-08-10:
I think I'll add both. Let's see if somebody protest, but given that the change is aesthetic only, I feel I can do it.

Rmano, 2023-08-10:
@sputeanus  notice that the name of the font parameter changed a bit. I hope you like the change... (notice that the workaround will stop working when `circuitikz` will update to 1.6.4).

sputeanus, 2023-08-11:
Ah I see its plural now ;).

Alright looks nice, waiting for the Patch!

Rmano, 2023-08-11:
Ok, merged now. It's still too early to release a version, but you can try it by downloading the "bleeding edge" version from https://circuitikz.github.io/circuitikz (when Travis decides to push the changes...)

closed 2023-08-11 by Rmano

Rmano, 2023-08-11:
Ok, Travis started again, so now it's all synced.

PR #731 (2023-08-11): Unrotate text in European logic ports
In addition, make the font used configurable; add a couple of "blank" ports that can be filled with the node text.

Thanks to user @sputeanus
Fixes #730

[image]

[image]

[image]
commit dadb63a9 (2023-08-11): Unrotate text in European logic ports

In addition, make the font used configurable; add a couple
of "blank" ports that can be filled with the node text.

Thanks to user @sputeanus
Fixes #730