circuitikz/circuitikz #733: [Feature] Normally closed switched with arrows

Rmano, 2023-08-15:
Suggested by user Draux here: https://tex.stackexchange.com/questions/693446/new-switch-components-for-circuitikz

Dr4UX, 2023-08-15:
#734 Here you can find the corresponding PR. I changed the thickness of the arrows to the thickness they have in the other switch symbols. But now it seems to overlap with the switch symbol. Probably it is better to leave them on normal thickness. Maybe add a key to control the properties of the arrows, so you can control the arrow-style by yourself?

Additionally I added a closing and an opening switch in the style of the normal closed switch but as normal open switch. You can see all the variants in the image.

[image]

Rmano, 2023-08-15:
I'd like to have the most "visual match" possible, I mean, so that the existing `spst` and the new arrows match. You need to draw more of the arc to avoid overlaps. 

After adding the new switches, we can add a parameter to style the arrows of all of them (in the style of the "modifiers thickness" of for example varistors, see manual around bottom of page 53). I prefer to add the new component before, with the same thickness as the existing one, and then add the modifiers (I can do that myself).

Rmano, 2023-08-15:
BTW, the "modifiers thickness" thing is worth adding, though, to all the switches. So forget about that for now, just tune the switch without changing the arrow thickness, and then I'll add it to all the switches.

Dr4UX, 2023-08-16:
Okay, I changed the end angles of the arcs to 95° for the opening switches and to -35° for the closing switches. This seems to be the closest I can get to the original values without overlapping of the arrow and the switch.

Furthermore I reduced the length of the vertical line in the normal open Switches slightly, because the gap to the Switch itself was to narrow.

You can see a difference to the `spst`, but this should mostly be caused by the smaller arrows.

Rmano, 2023-08-17:
I have merged your PR #734  --- I will now prepare a patch to add arrow thickness configurability  (and I'll wait for that to close this issue)

Thanks for your contribution!

closed 2023-08-17 by Rmano

Dr4UX, 2023-08-18:
Thanks. I am glad I could help :)

PR #735 (2023-08-17): Configure switches arrow
[image]

Fixes #733