circuitikz/circuitikz #740: Magnetic circuits

mxxmxn, 2023-09-12:
Magnetic circuits are widely used in the teaching of basic electrical engineering.

Basically, these circuits are similar to electrical circuits. They consist mainly of coils wound around a high permeability core and resistors represented by air gaps or interposed materials.

A simple example:
```
\begin{tikzpicture}
  \draw[double=lightgray, double distance=1cm,line cap=rect]
  (3,0)-- ++(0,-.9) -- ++(-3,0) -- ++(0,3) -- ++(3,0) -- ++(0,-.9);
  \pgfpathcurvebetweentime{0}{1}
      {\pgfpoint{-1cm}{1.03cm}}
      {\pgfpoint{-.5cm}{1.03cm}}
      {\pgfpoint{-.51cm}{1.03cm}}
      {\pgfpoint{0cm}{1cm}}
  \pgfpathcurvebetweentime{.5}{.89}
      {\pgfpoint{0cm}{1.1cm}}
      {\pgfpoint{-2.2cm}{1.05cm}}
      {\pgfpoint{2.2cm}{.95cm}}
      {\pgfpoint{0cm}{.9cm}}
  \pgfpathcurvebetweentime{.11}{.89}
      {\pgfpoint{0cm}{.9cm}}
      {\pgfpoint{-2.2cm}{.85cm}}
      {\pgfpoint{2.2cm}{.75cm}}
      {\pgfpoint{0cm}{.7cm}}
  \pgfpathcurvebetweentime{.11}{.89}
      {\pgfpoint{0cm}{.7cm}}
      {\pgfpoint{-2.2cm}{.65cm}}
      {\pgfpoint{2.2cm}{.55cm}}
      {\pgfpoint{0cm}{.5cm}}
  \pgfpathcurvebetweentime{.5}{1}
      {\pgfpoint{0cm}{.5cm}}
      {\pgfpoint{-.5cm}{.47cm}}
      {\pgfpoint{-.51cm}{.47cm}}
      {\pgfpoint{-1cm}{.47cm}}
  \pgfusepath{stroke}
\end{tikzpicture}
```
 [image]

Is the implementation of these magnetic circuits desired?
If so, how could this be implemented?

Rmano, 2023-09-12:
Yes, I kind of remember those. My feeling is that what you drawn is more a physical diagram than a circuit, and so probably plain TikZ and pics are better  suited to draw it. If I am not wrong, the following step was to draw an equivalent electrical circuit, for which I *think* there are all the necessary components already.

But, as in the case of #683, I am not opposed to add them, especially as an additional loadable module, if someone contribute it. Although using the `to` for the solid part of the circuit would need a different to-path routine, which is not trivial.

mxxmxn, 2023-09-19:
Since these are components like any other circuit, I think the syntax could be used very well for this. Especially since drawing these magnetic circuits can sometimes be difficult.
In general, I just wanted to know if this is something that would fit circuitikz. It will probably take a long time to implement, if I can do it at all.
Can you possibly give me an example of a loadable module in circuitikz?

Rmano, 2023-09-19:
There is (still) no interface for a loadable module... I'd like to add it, but I never found time.

For now, you can just think of a simple input-able file (so, with all the definitions together) supposing it is read after the package is loaded.

Rmano, 2023-09-19:
Additional comment: where I see the main problem is managing the "connection" between sides of the magnetic circuit. 

When implementing a `to-path` routine, you have no look-ahead to the circuit. The to-path routine of `circuitikz` simply draws two line segments with the current width with the component in the middle, rotated by the correct quantity in order to be aligned. The poles (`-o`, etc.) are specified in the parameters because they do not depend on the rest of the circuit. If you specify them twice, in two adjacent branches, they simply overwrite each other, so the most recent "wins". 

Here you have to manage the end-of-`to` behavior, which is different if the branch continues in the same direction, or does a 90-degree "curve", or even splits into two... I don't say it's not doable, but the interface must be well thought-out (it's interesting anyway, a pity I have very little time now, at least till December I am quite class-swamped).