circuitikz/circuitikz #758: [New Component] Saturation Block

Tipounk, 2023-12-12:
Hello everyone,

I'm using circuit tikz to draw block diagrams (for electric systems control).
I wanted to define a new component myself which would be a saturation block : 
[image]

It's widely used in power electronics to model the boundaries of a signal (for instance a maximum output current that the converter cannot exceed). It is used in simulation softwares like Simulink : https://fr.mathworks.com/help/simulink/slref/saturation.html

It is my first time defining a new component for Circuitikz so here what i've managed to do : 

```
\makeatletter
 

  \ctikzset{bipoles/saturation/width/.initial=.7}
\def\pgfcirc@twoport@draw@satdraw{
    \pgfscope
        \pgfsetlinewidth{1.2\pgfstartlinewidth}
		\pgfpathmoveto{\pgfpointorigin}
		\pgfpathlineto{\pgfpoint{.4\pgf@circ@res@step}{.4\pgf@circ@res@step}}
		\pgfpathlineto{\pgfpoint{0.65\pgf@circ@res@step}{.4\pgf@circ@res@step}}
		
		\pgfpathmoveto{\pgfpointorigin}
		\pgfpathlineto{\pgfpoint{-.4\pgf@circ@res@step}{-.4\pgf@circ@res@step}}
		\pgfpathlineto{\pgfpoint{-0.65\pgf@circ@res@step}{-.4\pgf@circ@res@step}}
        \pgfusepath{draw}
    \endpgfscope
}
%% saturation
\pgfcirc@define@twoports{blocks}
{}
{\ctikzvalof{bipoles/saturation/width}}
{saturation}
{\ctikzvalof{bipoles/saturation/width}}
{\ctikzvalof{bipoles/saturation/width}}
{
    \pgfcirc@twoport@draw@satdraw{0}
    %\pgftext[x=0.45\pgf@circ@res@left]{\ctikzvalof{bipoles/twoport/text}}
}
\pgfcirc@activate@bipole@simple{l}{saturation}
 
 \makeatother
```

I just wanted to add the possibility to put value or variable name for the upper and lower boundaries of the saturation like that : 
[image]

I thought of adding new anchors but I don't know how to do that.

Also, if you're ok with that, can you tell me how I could add this new block, once finished to the new circuitikz release ?

Thank's for your help ;).

Rmano, 2023-12-12:
Thank you for the contribution! I will revise it as soon as I can, and I'll explain the process to add it to the next release.

About the anchors, I think that putting a node `above` the `ne` one (and `below` the `sw` one) should suffice. I'll show an example when I can, also (on a mobile phone now...)

Rmano, 2023-12-12:
So, the block works; the only error is that `\pgfcirc@twoport@draw@satdraw` does not take arguments, so the `{0}` is wrong. You can see it because the `{0}` is simply copied as `0` in the input stream, and you have 

```
  Missing character: There is no 0 in font nullfont!
```

in the log file. Anyway, the code below does work ok:

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}
\usepackage{amsmath}

\makeatletter
\ctikzset{bipoles/saturation/width/.initial=.7}
\def\pgfcirc@twoport@draw@satdraw{
    \pgfsetlinewidth{1.2\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpointorigin}
    \pgfpathlineto{\pgfpoint{.4\pgf@circ@res@step}{.4\pgf@circ@res@step}}
    \pgfpathlineto{\pgfpoint{0.65\pgf@circ@res@step}{.4\pgf@circ@res@step}}
    %
    \pgfpathmoveto{\pgfpointorigin}
    \pgfpathlineto{\pgfpoint{-.4\pgf@circ@res@step}{-.4\pgf@circ@res@step}}
    \pgfpathlineto{\pgfpoint{-0.65\pgf@circ@res@step}{-.4\pgf@circ@res@step}}
    \pgfusepath{draw}
}
%% saturation
\pgfcirc@define@twoports{blocks}
{}
{\ctikzvalof{bipoles/saturation/width}}
{saturation}
{\ctikzvalof{bipoles/saturation/width}}
{\ctikzvalof{bipoles/saturation/width}}
{
    \pgfcirc@twoport@draw@satdraw{0}
}
\pgfcirc@activate@bipole@simple{l}{saturation}
\makeatother

\begin{document}
\section{Dummy sec}
\subsection{Dummy subsec}
\subsubsection{Dummy subsubsec}

\begin{groupdesc}
\circuitdescbip*{saturation}{Saturation}{}
\end{groupdesc}

\begin{LTXexample}[pos=t, varwidth=t, basicstyle=\small\ttfamily]
\begin{circuitikz}[european]
    \draw (0,0) to[amp, >, t=$K_{\mathrm{p}}$] ++(2,0)
    to[saturation, >, name=sat] ++(2,0) -- ++(0.5,0) 
    to[twoport, t=$\dfrac{\tau s +1}{\tau s}$, >, bipoles/twoport/width=1.5] ++(3,0);
    \node[above] at (sat.ne) {$+I_{\mathrm{max}}$};
    \node[below] at (sat.sw) {$-I_{\mathrm{max}}$};
    \draw (0,0) to[saturation] ++(0,3);
\end{circuitikz}
\end{LTXexample}

\end{document}
```

producing:

[image]

I can add it to the package, with the attribution to you, or if you prefer, you can prepare a PR (as explained in the file https://github.com/circuitikz/circuitikz/blob/master/CONTRIBUTING.md .

Tipounk, 2023-12-13:
Hi,
Thanks for your help reviewing this code.
The anchors you suggested fit well.

I'm ok if you add it yourself in the package if it's not bothering you.
Never did a pull request before so I should learn for next time :)

Rmano, 2023-12-13:
Ok. I will re-open and close it when I merge the change.

How do you like to be attributed? "user @Tipounk on GitHub" or you prefer a name/email?

Tipounk, 2023-12-13:
I prefer my name email : P. Sacco - paul.sacco@estaca.eu

PR #760 (2023-12-15): Add `saturation`, `iamp`, and `sigmoid` blocks
Contributed by P. Sacco - paul.sacco@estaca.eu
Fixes #758

[image]
[image]
[image]
commit 7157f8b2 (2023-12-15): Add `saturation` block

Contributed by P. Sacco - paul.sacco@estaca.eu
Fixes #758
closed 2023-12-15 by Rmano