circuitikz/circuitikz #761: [Feature] Allow separate node size for circ

TheChilliPL, 2023-12-20:
**Is your feature request related to a problem? Please describe.**
For some reason
```latex
\tikzset{%
    circ/.append style={%
        circuitikz/nodes width=0.02 % or any other number I tested
    }
}
```
causes nodes not to be rendered at all instead of changing circ node size. `\ctikzset{nodes width=0.02}` works but I need it to be separate from ocirc

**Describe the solution you'd like**
Allow setting separate node size for various node types

**Describe alternatives you've considered**
-

**Additional context**
-

Rmano, 2023-12-21:
It doesn't work because `circ` is not a style but a shape name. You can obtain what you need (although I am not sure about your use case, given that you didn't post an example). You can define a style, use the styled pole, and/or define the shortcuts you need for the `to` syntax.

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\tikzset{Circ/.style={circ, circuitikz/nodes width=0.1}}
\ctikzset{C-*/.style = {bipole/nodes/left=Circ, bipole/nodes/right=circ}}
\begin{document}
\begin{tikzpicture}[]
    \draw (0,1) node[Circ]{} to[R] ++(2,0) node[circ]{};
    \draw (0,0) to[R, C-*] ++(2,0);
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2023-12-21:
Closing now; please feel free to reopen if needed.

closed 2023-12-21 by Rmano