circuitikz/circuitikz #764: [Feature] Scaling Arrow Tips (and connection Dots)

sputeanus, 2023-12-27:
Hey there, 
at the moment I am shrinking down a tikzpicture.
[image]

The last thing which misses me is, how to shrink down the arrow tips in circuitikz and maybe also to scale down the connection points. Is there already a possibility implementend in ciruitikz?

Warm regards
Simon

Rmano, 2023-12-27:
Can you share some code? The best way of scaling everything down (without affecting fonts) is with `bipoles/length`, which should scale everything...

sputeanus, 2023-12-30:
Hey Romano,

sorry for the late reply. But perfekt  ```bipoles/length``` was exactly what I was looking for. Thank you very much!

```
\documentclass{article}

\usepackage{tikz}
\usepackage{circuitikz}
\usepackage{pgfplots}
\usetikzlibrary{arrows.meta} 

\newcommand{\ind}[2]{\ensuremath{#1_{\textrm{#2}}}} %Klassiker

\begin{document}
\begin{tikzpicture}

%% Draw the electrolyzer polarization curve
\begin{axis}[%
%	grid,
	width=4cm,
	height=2cm,
	at={(-2cm,-0.4cm)},
	label style={font=\footnotesize},
	tick label style={font=\footnotesize} ,
	scale only axis,
	xmin=0,
	xmax=120,
	ymin=0,
	ymax=60,
	xtick={100},
	extra x ticks={0},
	xticklabels={\ind{I}{max}},
	ytick={18*1.23,32.2},
	yticklabels={\ind{U}{OC}, \ind{U}{stat}},
	extra y ticks={0},
	xlabel=\ind{i}{cell},
	ylabel=\ind{u}{cell},
	axis y line=center, 
	axis x line=center,
	axis background/.style={fill=white},
	samples=100, %Anzahl der Punkte für 
	axis on top %Sets the axis to the foreground
	]

	%Polarization Curve
	\addplot [domain = 0:100, color=black] {18*1.23+0.12*x+2.5*ln(x*2+1)};

	% Linear Approximation
	\addplot [domain = 0:100, color=olive] {32.2+0.153*x};

	% Triangle
	\draw [densely dotted,olive] (axis cs: 50, 39.7) to ++(axis cs: 0,8) to node[above](R) {\footnotesize $\ind{R}{a}+\ind{R} {m}+\ind{R}{c}$} ++(axis cs: 50,0);

	% Operating Point
	\addplot [color=black, only marks, mark=x, mark size=3pt]
	table[row sep=crcr]{%
		75	43.6832\\
	} node[anchor=north,yshift=-4] {\footnotesize Operating Point};
		

	\end{axis}

	% Draw a) Polarization curve
	\draw (0,-1cm) node(Beschriftung){\footnotesize a) Polarization curve};

	%% Set the the size of the Bipoles and the font
	\ctikzset{bipoles/length=0.8cm}
	\ctikzset{bipole label style/.style={font=\footnotesize}}
	\ctikzset{bipole current style/.style={font=\footnotesize},
	bipole voltage style/.style={font=\footnotesize}}

	%% Draw the messy circuit
	% Resistances and voltage sources
	\draw(-3.05,-3)to [short,i=\ind{i}{cell},o-*] ++(0.8,0) node(Anode){} to ++(0,0.4) to [R,l=\ind{R}{a}]  ++(1.2,0)to [short,-*] ++(0,-0.4) to ++(0.2,0) to  [R,l=\ind{R}{m}] ++ (0.8,0) node[below,xshift=3,yshift=-30](Circuit){\footnotesize b) Equivalent circuit} to [european voltage source, v=\ind{U}{stat}] ++(0.8,0) to [short,-*] ++(0.2,0)  node(Cathode){} to ++ (0,0.4) to [R, l=\ind{R}{c}] ++(1.2,0) to [short,-*] ++(0,-0.4) to  [short,-o] ++ (0.8,0);

	% Capacitors
	\draw(Anode.center) to ++(0,-0.4) to [C,l_=\ind{C}{a}] ++(1.2,0) to ++(0,0.4);
	\draw(Cathode.center) to ++(0,-0.4) to [C,l_=\ind{C}{c}] ++(1.2,0) to ++(0,0.4);

	% Cell voltage
	\draw
	[out=20, in=160,-{Triangle[width=1mm, length=1mm]}] (-3.05,-2.4) to node[above] (Voltage){\footnotesize \ind{u}{cell}} ++(6,0);
	

\end{tikzpicture}%

\end{document}
```

[image]

Rmano, 2023-12-30:
Nice, I'm really happy it worked for you.

Have a nice New Year's eve!

closed 2023-12-30 by Rmano