circuitikz/circuitikz #765: [Bug] Incorrect documentation of resistor

quark67, 2023-12-27:
**Describe the bug**
The manual show how to change the height of the resistor, but the text say that we "can change the width of resistor, **not the height**".

**To Reproduce**
Open the manual on page 37, which show how to change the **height**.

**Screenshots**
[image]

**But in fact (with this modified code example)**

We can change the width AND the height (but only on "american" resistor, not "european"): 

```latex
\begin{circuitikz}[american]
\tikz \draw (0,0) to[R=1<\ohm>] (2,0); 
\ctikzset{bipoles/resistor/width=3.5, bipoles/resistor/height=.6}
\tikz \draw (0,0) to[R=1<\ohm>] (5.5,0);
\end{circuitikz}
```

gives:

[image]

**Operating system, LaTeX distribution**

Version 1.6.6 of circuitikz

**Expected behavior**

Modifying the manual with an example showing that the width and the heigth of the resistor can be changed AND a code which show that.

Rmano, 2023-12-28:
Well, the problem is really the example on page 37, which I should remove. As you noticed, it works only for American resistors (you have to use the internal name of the European one, `generic`, to change it), so basically, it is accessing internal parameters, which I do not want to be forced to maintain forever(*). So I will change the docs, but probably not in the way you think... 😉

To change the resistors' width and height, you should change the scale and width locally, which will work for every style. 

For example, using styles:

```latex
\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage[siunitx]{circuitikz}
\begin{document}
    \ctikzset{american}
    \tikz \draw (0,0) to[R=1<\ohm>] (4,0);

    \ctikzset{bigR/.style={resistors/scale=4, resistors/width=1}}
    \tikz \draw (0,0) to[R=1<\ohm>, bigR] (6,0);

    \ctikzset{european}
    \tikz \draw (0,0) to[R=1<\ohm>, bigR] (6,0);

\end{document}
```

[image]

BTW, your code is wrong; you are nesting Ti*k*Z environment using `\tikz`inside `circuitikz`.  See https://tex.stackexchange.com/questions/46598/problem-with-overlay-when-a-tikzpicture-is-inside-another-tikzpicture for example

(*) if you look at the source, there are *thousands* of parameter that are all the same (for example, widths for every type of resistor or source or...), which I would like to get rid of when I have time, and putting it in the manual is a form of officially freeze them forever.

Rmano, 2023-12-28:
Like this:

\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage[siunitx]{circuitikz}
\begin{document}
    \ctikzset{american}
    \tikz \draw (0,0) to[R=1<\ohm>] (4,0);

    \ctikzset{bigR/.style={resistors/scale=4, resistors/width=1}}
    \tikz \draw (0,0) to[R=1<\ohm>, bigR] (6,0);

    \ctikzset{european}
    \tikz \draw (0,0) to[R=1<\ohm>, bigR] (6,0);

\end{document}

quark67, 2023-12-28:
> BTW, your code is wrong; you are nesting TikZ environment using `\tikz` inside `circuitikz`. 

Yes, right, my bad.

I loaded the package with `\usepackage[siunitx,europeanresistors]{circuitikz}` in order to have the european shape for resistor (**as an European, the european resistor shape seems so natural...**).

In first attempt, I only used the code:

```latex
\tikz \draw (0,0) to[R=1<\ohm>] (2,0); 
\ctikzset{bipoles/resistor/width=3.5, bipoles/resistor/height=.6}
\tikz \draw (0,0) to[R=1<\ohm>] (5.5,0);
```

and to quickly obtain the american shape, I enclosed this previous code with `\begin{circuitikz}[american]` ... `\end{circuitikz}`, which is in fact an error as you pointed out. The 2 `\tikz` are superfluous. And in fact, using `\ctikzset{american}` is effectively more quick.

Thanks.

commit c5192a28 (2023-12-28): Fix a misleading documentation about resistor's shape

Thanks to @quark67
Fixes #765
closed 2023-12-28 by Rmano