circuitikz/circuitikz #768: [New Component] Quadripoles with parameter: Z/Y/G/H, generic Thevenin/Norton.

alceu-frigeri, 2024-01-03:
Hi, first of, Happy New Year !

Some years ago, for my own use I ended creating some shapes (more of a composite) to construct standard quadripoles Z, Y, G, H parameters and associated. And now some colleagues of mine were asking me to publish it.

The most natural would be to submit it to you, though 
I have tried to look at the way your shapes are being build and classified, and I got stuck 
(the last time I took a look, there wasn't \ctikzclass ... I probably should had tried to contact you earlier, but, anyway.)

Since what I ended up making isn't a single shape:
There are two 'base boxes'
* a single port (just a black box)
* a dual port one (the base for a quadripole)
and derivatives
* from the single port
    - Black Box
    - Thevenin
    - Norton
* from the dual port
    - Quad (no circuit inside)
    - Quad Z
    - Quad Y
    - Quad G
    - Quad H

And related  to[] paths as well.

My trouble is if and how to fit this in your system, if just as it is (I doubt) or if (and then what) I would have to change in my setup.
Given the size of the code, it wouldn't be simple to just add it here (in this post), so I made a repository just for that

https://github.com/alceu-frigeri/ctikzext

as usual, doc folder with a quick documentation, code in the latex folder.

You will notice that I have ended creating my own shapes for sources and impedances, and I have an extensive use of pgfkeys.

That all said, it would be nice to make this work, but if not (or too much work), it´s fine.
Cheers,
Alceu 

**Show the component and its usage**

Add an image of the new component and typical usage in a circuit.

**Add references to standards, usage, etc.**

Add links to pages that show the component usage or specification --- official (ISO, IEC, IEEE) standards preferred.

Rmano, 2024-01-04:
Hi Alceu,

Welcome and Happy New Year to you, too!

I need to dive a bit more into your code, but from what I see, it's quite extensive (with a nice look, too!). Wouldn't the best option be to keep it as your package? I will then try not to break it when updating `circuitikz`.  I can help you with doing this work if you need it. 

I think the best option for the class is to define your own class (or classes) and use that for your package. Again, defining a new class is easy, but I can help you with it.

I think it would be nice to have a package that depends on `circuitikz`; the real thing is that it's becoming too big, and I'd wish to have the time to modularize and simplify it... but that will be a "v2.0 retirement project" for me, I fear 😉.

alceu-frigeri, 2024-01-04:
Hi again,
sorry for taking so long to come back here (long, long, work day).
To have it as a separated package is fine, and perhaps the smoothest way. I already have a few packages at CTAN.
(I would just need to write down a proper package documentation... an the blunt is done, now it would be a matter of properly completing it.). And the risk of things getting broken down is low, I basically just use ```\pgf@circ@Rlen``` as a scale reference, so things should be just fine ;)

But, just exploring the options and ideas

1.   I added another 'demo file': https://github.com/alceu-frigeri/ctikzext/blob/main/doc/ctikzext%20Demo.pdf
So you can see why I call it a shape-shifting composite, not just because it is made of internal nodes, 
but the shape DO change per parameter used.
For instance, the round and control sources are the same shape (a local flag changes their appearance).
2. Classes, I think this would be the case of a new one (after all it isn't a source, or a resistor or...)... what would be necessary to 'stablish one'? just a separated pgfkey path? all my keys are under ```\def\ctikzext@@KeysPath{/tikz/ctikzext}```
3. which reminds me of another detail: pgfkeys, my approach was to have a 'search function', {```\SearchValueOf``` based on my package ```pgfkeysearch.sty```} which searches for a key in a pgfkeyspath (but only upwards). 
That's the way I managed to have my own set of keys, valid only when using one of those blocks, as if they were /tikz/ ones
(at the root /tikz/ level I created a single key: ```QuadKeys``` which sets everything as needed.)
4. modularization: are you fluent in ```expl3```? I know it can be a pain to mix pgf with expl3 in the same file (better keep them separated), but with expl3 properties (and keys) one could 'kind of easily' modularize it, for instance instead of defining shapes the tikz way, have an auxiliary, expl3 function that just assembles parts from a properly constructed property list an generates the shape. That's something I could help ;)

Rmano, 2024-01-05:
> Hi Romano, sorry for taking so long [...]

No problem, and no need to hurry (I'm still on vacation... so I'm taking it eeeeeeasy).

> 
> But, just exploring the options and ideas
> 
>1. I added another 'demo file': https://github.com/alceu-frigeri/ctikzext/blob/main/doc/ctikzext%20Demo.pdf
>So you can see why I call it a shape-shifting composite, not just because it is made of internal nodes,

Nice! 

> 
>2. Classes, I think this would be the case of a new one (after all it isn't a source, or a resistor or...)... what would be necessary to 'stablish one'? just a separated pgfkey path? all my keys are under `\def\ctikzext@@KeysPath{/tikz/ctikzext}`

Well, basically, classes just need to have defined a scale, a thickness, and a fill color:

https://github.com/circuitikz/circuitikz/blob/edc53449d6e6c4c2d5d06385b8cf92d39540c90f/tex/pgfcirc.defines.tex#L951-L953

so that the class is used to scale and set some basic properties when defining the component, see for example 

https://github.com/circuitikz/circuitikz/blob/edc53449d6e6c4c2d5d06385b8cf92d39540c90f/tex/pgfcircbipoles.tex#L643-L657

(The macros for setting thickness and fill are in 

https://github.com/circuitikz/circuitikz/blob/edc53449d6e6c4c2d5d06385b8cf92d39540c90f/tex/pgfcirc.defines.tex#L359-L377

and the basic dipole definition is in

https://github.com/circuitikz/circuitikz/blob/edc53449d6e6c4c2d5d06385b8cf92d39540c90f/tex/pgfcirc.defines.tex#L550-L557 

and around them; notice that the comment is wrong (the first argument is the class, not the scale factor, my bad...)

> 3. which reminds me of another detail: pgfkeys, my approach was to have a 'search function', {`\SearchValueOf` based on my package `pgfkeysearch.sty`} which searches for a key in a pgfkeyspath (but [...]

This is ok and probably better than the clunky usage of keys in my code, which has a quite low-key approach (I duplicate some keys on `/tikz` level... it was how it was when I started maintaining it, I did know a lot less of TeX programming then, and given that my philosophy is "never break existing code" I never tried to rationalize the thing).

> 
>4. modularization: are you fluent in `expl3`? I know it can be a pain to mix pgf with expl3 in the same file (better keep them separated), but with expl3 properties (and keys) one could 'kind of easily' modularize it, for instance instead of defining shapes the tikz way, have an auxiliary, expl3 function that just assembles parts from a properly constructed property list and generates the shape. That's something I could help ;)

I like `expl3` (although not fluent at all). This is another reason to keep your code separated. Basic `circuitikz` code is format-agnostic (that is: it works in ConTeXt too) and so the `expl3` thing is off-limits.

alceu-frigeri, 2024-01-05:
Hello again,

> Well, basically, classes just need to have defined a scale, a thickness, and a fill color: [...]

I was afraid  you would mention *fill* and ```\pgf@circ@draworfill```,
 specially because given the number of sub-nodes that would be quite a headache.
Reason: some traces are drawn middle ```\backgroundpath``` main culprit: ```\pgfnode```.
And scale as well, I would have to add a scale factor to it all (though that would be easier).

> This is ok and probably better than the clunky usage of keys in my code, which has a quite low-key approach (I duplicate some keys on `/tikz` level... it was how it was when I started maintaining [...]

I fully understand ;) until You get to the point of hacking a hack to keep things working ;)

> I like `expl3` (although not fluent at all). This is another reason to keep your code separated. Basic `circuitikz` code is format-agnostic (that is: it works in ConTeXt too) and so the `expl3` [...]

!! Well, I never used ConTeXt ( I knew the name, but just that)... and just found out it is Lua and TeX based, so I see, That would be another thing to deal with.

OK. for the sake of this open issue, I will publish it as a CircuiTikZ extension (ctikzext package)... and try to see if I can add  fill/not fill in it (besides scale), and then, perhaps, come back 'here' ;)

Cheers, and thanks for your attention on vacation time !

Rmano, 2024-01-05:
Fine with me. One suggestion --- choose a different name from `ctikzext`. There could be more extensions in the future, you never know, even from you.

`ctikzeqblocks` or `ctikzqpoles` would be better, but I'm not so good at choosing names...

alceu-frigeri, 2024-01-06:
I'm more than terrible when it comes to naming. lol. I 'have' time to find one (well, first I have to write down a proper manual), I might go with ```ctikzquads``` or just ```tikzquads``` (I'm not a fan of long package names).

alceu-frigeri, 2024-01-14:
Hi, I managed to write some documentation (and find some long standing bugs in my 'fit to' keys), and it's already live by CTAN as tikzquads and https://github.com/alceu-frigeri/tikzquads

Rmano, 2024-01-19:
Wonderful. If you want, we can add a section/paragraph to the `circuitikz` manual citing the existence of related packages... that would be useful for discoverability.

alceu-frigeri, 2024-01-20:
That would be very nice, indeed ! and taking the opportunity, have you saw tikzdotncross? it borrows an idea from Redaelli (or Yours, Romano, I'm not sure) expands it a little and adds a ```\pathcross``` command that 'solves' the jumping lines problem.

Rmano, 2024-01-21:
@alceu-frigeri  what do you think? Please add/change/edit!

alceu-frigeri, 2024-01-21:
already commented on it ;)

PR #772 (2024-01-22): Add related and enhancement packages to the documentation
Closes #768

[image]

[image]
commit 3fd4be43 (2024-01-22): Add related and enhancement packages to the documentation

Closes #768
closed 2024-01-22 by Rmano