circuitikz/circuitikz #769: [Feature] Additional Nodes for AC\DC DC\DC AC\DC converters

sputeanus, 2024-01-18:
Hey ho, 

I am currently drawing a simple circuit with an AC/DC converter and I am not so happy with the distance of the DC Nodes.

[image]

Intuitively I would place the DC- Node at 10-20% of the height and the DC+ Node at 80-90% of the height.

At the moment it is more like 25% and 75% or even 33% to 66%.
[image]

Are there strong opinions about keeping the current values and if yes might it be possible to introduce a second set of nodes which are further out?

Warm regards
Simon

sputeanus, 2024-01-18:
For example if you want draw a typical setup in the field of power electronics, which is a DC-Link, you get unpleasant results when the nodes are so far in the middle and you loose a lot of vertical space.

``
\begin{tikzpicture}

%% Standard Sizes
% Draw Converters
\draw (0,0)node[tacdcshape] (DC1){};
\draw (5,0)node[tacdcshape,xscale=-1] (DC2){};

% Draw DC Link Connections
\draw (DC1.dc up out) to [short] ++(2,0) coordinate (DC+) to (DC2.dc up out);
\draw (DC1.dc down out) to [short] ++(2,0) coordinate (DC-) to (DC2.dc down out);

% Draw Capacitor
\draw (DC+) to [C, v=$U_{DC}$] (DC-);

%% Increased sizes
% Draw Converters
\draw (0,-4)node[/tikz/circuitikz/bipoles/length=2.5cm,tacdcshape] (DC1){};
\draw (5,-4)node[/tikz/circuitikz/bipoles/length=2.5cm,tacdcshape,xscale=-1] (DC2){};

% Draw DC Link Connections
\draw (DC1.dc up out) to [short] ++(2,0) coordinate (DC+) to (DC2.dc up out);
\draw (DC1.dc down out) to [short] ++(2,0) coordinate (DC-) to (DC2.dc down out);

% Draw Capacitor
\draw (DC+) to [C, v=$U_{DC}$] (DC-);

\end{tikzpicture}
``

[image]

(By the way, is there a better way to scale the converters?)

Rmano, 2024-01-18:
Hmm... 

1. to scale blocks, you can use the class: `\ctikzset{blocks/scale=...}`, and that will apply to every block.
2. for the problem of the position of the anchors: yes, I see... what about if I add a parameter that let the user change the position of the generic `left up` etc. anchors? Now is 0.5 (midway center and border) but it seems easy to change that. 
3. I would write your last example as:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}

%% Standard Sizes
% Draw Converters
\draw (0,0)node[tacdcshape] (DC1){};
\draw (5,0)node[tacdcshape,xscale=-1] (DC2){};

% Draw DC Link Connections
\draw (DC1.right up) to [short] ++(2,0) coordinate (DC+) to (DC2.right up);
\draw (DC1.right down) to [short] ++(2,0) coordinate (DC-) to (DC2.right down);

% Draw Capacitor
\draw (DC+) to [C, v=$U_{DC}$] (DC-);

%% Increased sizes
\ctikzset{blocks/scale=2}
% Draw Converters
\draw (0,-4)node[tacdcshape] (DC1){};
\draw (5,-4)node[tacdcshape,xscale=-1] (DC2){};

% Draw DC Link Connections
\draw (DC1.dc up out) to [short] ++(2,0) coordinate (DC+) to (DC2.dc up out);
\draw (DC1.right down) to [short] ++(2,0) coordinate (DC-) to (DC2.right down);

% Draw Capacitor
\draw (DC+) to [C, v=$U_{DC}$] (DC-);

\end{tikzpicture}
\end{document}
```

[image]

Although I will not really invert the block, just use the corresponding inverse (but that's a matter of taste):

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}

%% Increased sizes
\ctikzset{blocks/scale=2}
% Draw Converters
\draw (0,-4)node[tacdcshape] (DC1){};
\draw (5,-4)node[tdcacshape] (DC2){};

% Draw DC Link Connections
\draw (DC1.right up) to [short] ++(2,0) coordinate (DC+) to (DC2.left up);
\draw (DC1.right down) to [short] ++(2,0) coordinate (DC-) to (DC2.left down);

% Draw Capacitor
\draw (DC+) to [C, v=$U_{DC}$] (DC-);

\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2024-01-18:
@sputeanus  what do you think of the PR #770 ? I think it'll solve the problem.

commit 2ffbb06b (2024-01-19): Add key to configure block's lateral anchors position [skip ci]

Now the position of the lateral anchors (`left up` and similar)
of blocks is configurable.
suggested by @sputeanus, Fixes #769
closed 2024-01-19 by Rmano