circuitikz/circuitikz #773: source/align

jotagah, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jotagah, 2024-01-26:
Is there any way to add a non-rotate to voltage source (american) polarity that is independent of the voltage source rotation? There is a proposed solution on https://tex.stackexchange.com/questions/176296/circuitikz-voltage-source-american-style, but it mess with figure align and seems to disable some features.

Rmano, 2024-01-27:
I'm looking at it. It's quite easy to fix for LaTeX, not so for ConTeXt... I will give you a workaround as soon as I can, but for a final fix/feature I need more time....

Rmano, 2024-01-28:
A temporary solution, with a new TeX (must have `\fpeval`) is:

```latex
\ctikzset{bipoles/vsourceam/inner minus=\rotatebox{\fpeval{\pgf@circ@direction+90}}{$-$}}
\ctikzset{bipoles/vsourceam/inner plus=\rotatebox{\fpeval{\pgf@circ@direction+90}}{$+$}}
\makeatother

\begin{tikzpicture}[american]
    \foreach \a in {0,30,...,360} {
        \draw (0,0) to[V, invert] (\a:4);
    }
\end{tikzpicture}
```

[image]

Although I am not sure I like the not-orthogonal ones...

jotagah, 2024-01-28:
Nice solution, but I guess '\makeatletter' is missing at begin the workaround code.

About the non-orthogonal, why is not possible to draw the minus symbol oriented in the direction of the Iine as your new  horizontal workaround?

Rmano, 2024-01-29:
It's possible, but doing so the minus sign is not horizontal ---  which I supposed it was what you want. In stock circuitikz, the minus is always perpendicular to the line entering. Here, the math symbols are always "straight", in reading terms. At which angle should the orientation change? 30? 45?

jotagah, 2024-01-29:
Sorry for a question not clear. I agree that the question sounds like a self-preference. When I said, "the minus symbol oriented in the line direction", I was referring just to non-orthogonal case, to draw it like your workaround for horizontal usage. 
An output similar to my question can be seen in the answer from a question posted at TeX StackExchange to the non-orthogonal controlled source and answered by J Leon V. at https://tex.stackexchange.com/questions/453356/orientation-of-negative-polarity-notation-in-voltage-source-when-connected-diago/453382#453382)

Rmano, 2024-01-29:
Hmmm. Yes, this is basically manual. I see that the `-` looks good in line for the -45...45 degrees (with the + also inline, so rotated) but I have not clear at all that this is ok for 45...90 degrees. Maybe just adding a manual option to rotate the internal symbols would suffice...

jotagah, 2024-01-29:
That will be an excellent option.

Rmano, 2024-01-29:
The other option is to have a flag that does this (automatically, enabled by a flag --- it's a big change to make it unconditionally):

[image]

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
% \usepackage{circuitikz}
\usepackage[RPvoltages]{circuitikz}
\begin{document}
\makeatletter
\pgfcircdeclarebipolescaled{sources}
{}
{\ctikzvalof{bipoles/vsourceam/height}}
{vsourceAMb}
{\ctikzvalof{bipoles/vsourceam/height}}
{\ctikzvalof{bipoles/vsourceam/width}}
{

    \pgfscope
        \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
        \pgfpathellipse{\pgfpointorigin}{\pgfpoint{0}{\pgf@circ@res@up}}{\pgfpoint{\pgf@circ@res@left}{0}}
        \pgf@circ@draworfill
    \endpgfscope
    \pgf@circ@text@strokecolor
    \pgfgettransformentries\a\b\temp\temp\temp\temp
    \pgfmathsetmacro{\rot}{-atan2(\b,\a)}
    \pgfmathsetlength{\pgf@circ@res@temp}{\ctikzvalof{bipoles/vsourceam/margin}*\pgf@circ@res@up-0.25*\baselineskip}
    \pgfmathsetmacro{\innerrot}{abs(\rot)<46?0:(abs(\rot)>134?0:90)}
    \ifpgf@circ@oldvoltagedirection
        \pgfscope
            \pgftransformxshift{\pgf@circ@res@temp}
            \pgftext[rotate=\innerrot]{\ctikzvalof{bipoles/vsourceam/inner plus}}
        \endpgfscope
        \pgfscope
            \pgftransformxshift{-\pgf@circ@res@temp}
            \pgftext[rotate=\innerrot]{\ctikzvalof{bipoles/vsourceam/inner minus}}
        \endpgfscope
    \else
        \pgfscope
            \pgftransformxshift{\pgf@circ@res@temp}
            \pgftext[rotate=\innerrot]{\ctikzvalof{bipoles/vsourceam/inner minus}}
        \endpgfscope
        \pgfscope
            \pgftransformxshift{-\pgf@circ@res@temp}
            \pgftext[rotate=\innerrot]{\ctikzvalof{bipoles/vsourceam/inner plus}}
        \endpgfscope
    \fi
}
\pgfcirc@activate@bipole@opt{v}{vsourceamb}{vsourceAMb}{american voltage source b}{%
    \circuitikzbasekey/bipole/is voltage=true,
    \circuitikzbasekey/bipole/is voltageoutsideofsymbol=false}

\makeatother

\begin{tikzpicture}[american]
    \foreach \a in {0,30,...,360} {
        \draw (0,0) to[V, invert] ++(\a:4);
    }
    \foreach \a in {0,30,...,360} {
        \draw[color=red] (8,0) to[american voltage source b, invert] ++(\a:4);
    }
\end{tikzpicture}
\end{document}
```

PR #775 (2024-01-30): Vsources sign rotation
Fixes #773 

[image]
closed 2024-01-30 by Rmano