circuitikz/circuitikz #774: Circuit offset

jotagah, 2024-01-28:
In the following circuit, I get an offset to right caused by each one of the line settings I used. Did I make any mistake?

\ctikzset{voltage/american label distance=2pt, voltage/distance from node=.75}
\ctikzset{voltage/bump b=20pt, voltage/european label distance=35pt}

\begin{circuitikz}[american]
	\draw[style=thick][scale=1,transform shape]
	   (0,0) to [R=\SI{4}{\kohm}] ++(0,3)
	   to [V=\SI{6}{\V},invert] ++(2,0) coordinate(1)
	   to [R=$R_x$,*-*] ++(0,-3)
	   (1) to [V=\SI{12}{\V}] ++(3,0)
	   to [open,v=$v_o$,o-o] ++(0,-3) -- (0,0)
	;
\end{circuitikz}

Rmano, 2024-01-29:
Please, try to:

1) use complete examples for your code and mark it as such
2) add an example output
3) for *usage* questions, please use https://tex.stackexchange.com/questions/tagged/circuitikz , it's better suited.

So, your code:

```latex
\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage[siunitx]{circuitikz}
\ctikzset{voltage/american label distance=2pt, voltage/distance from node=.75}
\ctikzset{voltage/bump b=20pt, voltage/european label distance=35pt}
\begin{document}
\begin{circuitikz}[american]
\draw[style=thick][scale=1,transform shape]
(0,0) to [R=\SI{4}{\kohm}] ++(0,3)
to [V=\SI{6}{\V},invert] ++(2,0) coordinate(1)
to [R=$R_x$,-] ++(0,-3)
(1) to [V=\SI{12}{\V}] ++(3,0)
to [open,v=$v_o$,o-o] ++(0,-3) -- (0,0)
;
\end{circuitikz}
\end{document}
```

compiles to:

[image]

You have strange things here: a `scale=1, transform shape` as a lone option in a draw path that makes little sense (in the best case, it's doing nothing and can wreak havoc); you are setting (huge) parameters for internal parameters you should not touch normally and mixing American and European styles... 

I am sorry, but I failed to understand the question here.

jotagah, 2024-01-29:
Thanks for the advice and my apologies about my incomplete example. I note that your circuit has no horizontal offset. So, the issue is from my preamble. I use the qtikz software to draw a circuit, as a prototyped. Something is wrong in my preamble. In qtikz, just comment the above '\ctikzset' lines to fix the offset.

Rmano, 2024-01-29:
The fact that a `\ctikzset` creates problems to another package could be classified as a bug, too. Maybe there is an option clash with `qtikz` at some level? If you have a minimal example showing the problem, I am interested in looking at it.