circuitikz/circuitikz #782: New circuitikz GUI available

cpfannenmueller, 2024-03-28:
At the Institute of Electronics Engineering (where also a few [circuitikz](https://github.com/circuitikz/circuitikz)-developers were located) we designed a open source easy to use [broswer based GUI](https://ussi.e-technik.uni-erlangen.de/tikz/) supporting all currently avaliable symbols. 
[CircuiCanva](https://ussi.e-technik.uni-erlangen.de/tikz/) supports drawing all current symbols, exports the result as TikZ code and has an easy to use interface to add new future symbols when they are integrated in circuitikz.

We are currently searching for feedback on our implemantation and hope to finde contributors for the whole [project](https://ussi.e-technik.uni-erlangen.de). 😊

[image]

During the project also  a Converter Tool was designed. It converts schematics already designed in your EDA design tool to CircuiTikZ. The idea is to avoid redrawing the circuitry for your documentation/paper if you already have it drawn in your simulation tool.

[image]

We would be glad to see you on: https://ussi.e-technik.uni-erlangen.de 😊

bogger33, 2024-04-22:
This tool is very nice, thank you for making it!

My initial feedback:

-When drawing a line, I think right click should act as "finish line", so you don't have to move your mouse into the corner each time. If you right click before drawing a second point, it should act as "cancel line"

-The lines are displayed a bit thin on my screen. It would be good to make them a bit thicker, or possibly to make the grid lighter (or both)

-The zoom steps seem a bit big. Zooming seems to double the resolution. It would be nice to have a step of 150% first

-perhaps a marker at the end of each line, to change its end point by dragging, similarly to how components can be dragged would be useful. Another option would be to automatically have lines snap to components, and move their endpoints alongside them (albeit this would be more complicated and perhaps sometimes undesired)

-it doesn't seem to actually have all symbols, e.g. I'm not seeing electron tubes at all

-it would be nice if right clicking components would allow editing their attributes (e.g. invert for a battery, or labels, etc) - even just manually being able to type in arguments would be enough here, I think

I hope this is helpful feedback. The tool is definitely already very useful as is.

Rmano, 2024-05-04:
I am adding links to the main documentation: 

```
\subsubsection{Related packages}

At the Friedrich-Alexander-Universität, a group of developers are implementing a graphical interface to draw circuits
with \Circuitikz{}. 
You can find more information \href{https://github.com/circuitikz/circuitikz/issues/782}{in this GitHub issue} or, better, in their \href{https://ussi.e-technik.uni-erlangen.de/}{main site}.
```

I hope this helps for finding users/testers!

jmonclard, 2024-07-02:
I just discovered the tool. Already useful and very promising.
Even if not a circuitikz symbol, and very easy to add with a simple tikz command, it could be nice to be able to draw (filled / dashed) background rectangles straight from the GUI.
I think, like bogger33 wrote, that editing symbols attributes (name, value) is really what is missing right now to get a quite complete schematic.
Thank you for the hard work !

cpfannenmueller, 2024-07-03:
> I just discovered the tool. Already useful and very promising. Even if not a circuitikz symbol, and very easy to add with a simple tikz command, it could be nice to be able to draw (filled / dashed) [...]

This is one of our planed tasks currently. See also https://git.lte.e-technik.uni-erlangen.de/circuitikz-gui/circuitikz-pwa/-/issues/18

Currently, our ideas and ToDos here are much more than our time, so we are a bit in trouble. But we kindly invite you to support us and would be if you could help us adding those features :-)

JPWiedemann, 2024-09-02:
@cpfannenmueller and I have updated the tool significantly as well as renamed the project. Please check out a current build [here](https://circuit2tikz.tf.fau.de/designer/). A lot more features are planned (like drawing rectangles and other primitives directly in the GUI like @bogger33 suggested). If you want to provide other feedback or encounter any bugs, please feel free to open an issue on our [github](https://github.com/Circuit2TikZ/CircuiTikZ-Designer/)

Rmano, 2024-09-02:
I merged #826 --- thanks. I will give it a go as soon as I have a bit of time (start of semester here... you know).

Rmano, 2024-11-05:
Hi, maybe you can add an answer to https://tex.stackexchange.com/questions/84890/is-there-an-online-diagram-graphical-editor-that-produces-the-corresponding-late

EEpchi, 2024-12-24:
wow, this tool is great

myzinsky, 2024-12-25:
nice tool, unfortunately useless for me, since it uses absolute placement. for my projects, relative placement would be required, since we change the size of the circuits like a rubber band. How hard would it be use relative placements and not absolute coordinates?

cpfannenmueller, 2024-12-30:
This issue is discussed here: https://github.com/Circuit2TikZ/CircuiTikZ-Designer/issues/41

Rmano, 2025-03-27:
Hi @cpfannenmueller  --- it seems that you have a problem with the SSL certificate: 

[image]

I was using the link: https://ussi.e-technik.uni-erlangen.de/tikz/ of the first post

JPWiedemann, 2025-03-27:
Hi @Rmano, we've moved the website to https://circuit2tikz.tf.fau.de/. We'll update the links in the first post.