circuitikz/circuitikz #783: The spacing of "dashed short" looks a little odd

EEpchi, 2024-04-05:
**Background**: To use a different arrow style, I defined an arrow tip style of my own and use it with current anchors, such as:
```latex
\documentclass{article}
\usepackage{circuitikz}
\newcommand{\iarrow}[3]{\path[iarrow] (#1-Ifrom) -- (#1-Ipos) node[#2]{#3};}
\tikzstyle{iarrow}=[
	tips,arrows={-Triangle[width'=0pt 0.4, length=5pt]},shorten >=-3pt
]
\begin{document}
\begin{circuitikz}
	\scriptsize
	\draw(0,0) to[short,name=sht,i,current/distance=0.2] ++(2,0) node[right]{my own tips};
	\iarrow{sht}{above}{$ I $};
\end{circuitikz}
\end{document}
```
[image]

**Now** I need a dashed line with the current arrow, so I first tried `to[short]`, but I found the spacing looks a little odd, it seems like two pieces put together, the spacing between the shorts appears uneven. Then I found if I use `--` to draw the line, the spacing is good, but then I wouldn't be able to name the components to activate the current anchors and use my own arrow style.
```latex
\documentclass{article}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
	\scriptsize
	\draw[densely dashed] (0,0) to[short] ++(2,0) node[right]{dashed ``short"};
	\draw[densely dashed] (0,-0.5) -- ++(2,0) node[right]{dashed ``-\,-"};
\end{circuitikz}
\end{document}
```
[image]

So why does the dashed `[short]` differs from the dashed `--` 🤔

Rmano, 2024-04-05:
This happens because a `short` is really a bipole of size (almost) zero. So, using `to[short]` is the same as restarting the line midway. 
I don't think it's possible to fix it easily...

EEpchi, 2024-04-05:
oh, I see. Here's my solution  currently:
```latex
\documentclass{article}
\usepackage{circuitikz}
\newcommand{\iarrow}[3]{\path[iarrow] (#1-Ifrom) -- (#1-Ipos) node[#2]{#3};}
\tikzstyle{iarrow}=[
	tips,arrows={-Triangle[width'=0pt 0.4, length=5pt]},shorten >=-3pt
]
\begin{document}
\begin{circuitikz}
	\scriptsize
	\draw[densely dashed] (0,-1) -- ++(2,0);
	\path[iarrow] (0,-1) -- ++(0.3,0) node[above]{$ I $};
\end{circuitikz}
\end{document}
```
[image]

Do you have any advice for me?🌹

Rmano, 2024-04-05:
Yes, your solution is ok. It's basically what the `to[...]` is doing when positioning the current arrow.

EEpchi, 2024-04-05:
> Yes, your solution is ok. It's basically what the `to[...]` is doing when positioning the current arrow.

OK, thank you very much🌹

closed 2024-04-05 by EEpchi