circuitikz/circuitikz #785: [New Component] Request: VFD Tubes

bogger33, 2024-04-21:
**Show the component and its usage**

Similarly to the #781 request for nixie tubes, VFD tubes are also absent from circuitikz. In its most simple form, a VFD tube is just an indirectly heated triode. However, they almost always have multiple anodes for the different segments. More complex tubes also have multiple grids, and different segments may share anodes to reduce the number of pins. Though it is unrealistic to cover every possible use case, it would be good to at least have a component for simple VFD tubes with one grid and multiple anodes, and more complex tubes can be split into multiple of these components in the diagram.

As with the nixie tube request, I have made a collage of representations of VFD tubes I have found:

[image]

Rmano, 2024-04-21:
Probably a possibility is a modification of the triode symbols to add more anodes...

[image]

bogger33, 2024-04-22:
That would work, with either "filament,nocathode" or "fullcathode" it would achieve the desired effect

Rmano, 2024-05-02:
I suppose a multi-anode tube could be used for both this one and Nixie, right? I think I have a bit of time this weekend, and I'll try to see if it is not too complex to implement.

bogger33, 2024-05-02:
nixie tubes actually have multiple cathodes (and they are cold cathode, so do not have any filament), but they're structurally similar in that a variable number of cathodes exist for different types of nixie tubes

Rmano, 2024-05-04:
Heads up: at this moment, I have the following.

[image]

which can be used to draw this (from this issue):

[image]

and this (from the nixie #781):

[image]

at least, more or less... I think they can be used, right?

bogger33, 2024-05-04:
it looks good! Would it be difficult to change the nixie version to have one central pin on the anode, rather than one on each side? other than that, it's perfect, thank you

Rmano, 2024-05-04:
Yes, I have thought that. The only problem is that it will called "cathode"... but maybe that's not a big problem, I can create aliases... maybe "nixie cathode *n*" and "nixie anode".

bogger33, 2024-05-04:
I see. I would be okay with aliases if it makes the code simpler for you. If the alias also changes the look of it to look like an anode, I suppose it's not too confusing anyway.

Rmano, 2024-05-04:
So:

[image]

Do you think it's ok?

bogger33, 2024-05-04:
yes, it looks perfect to me!

Rmano, 2024-05-04:
Ok, I think I'll integrate it today. And the next week I'll release 1.6.8.

Rmano, 2024-05-04:
Thank you for the comments!

bogger33, 2024-05-04:
Thank you for implementing it!

PR #788 (2024-05-04): Multi anode triode
[image]

[image]

Fixes #781 
Fixes #785
commit baa31067 (2024-05-04): Add a multi-anode tube for implementing nixies and vfd

Fixes #785
Fixes #781
closed 2024-05-04 by Rmano