circuitikz/circuitikz #792: [Feature] Filament center tap on tubes

bogger33, 2024-05-13:
**Is your feature request related to a problem? Please describe.**
I wasn't sure whether to file this as a feature request or component request, but some tubes like e.g. the very popular 12AX7 (ECC83) have a center tapped filament, so it can be used with multiple different voltages (either in series or parallel). It would be useful to allow making a connection to the center point of the tube filament in circuitikz to allow for pinout descriptions as well as just more unambiguous circuit diagrams. 

**Describe the solution you'd like**
An option for connecting to the center point of the filament. I apologize if this already exists and I missed it, but I could not find it in the manual.

**Additional context**

Center taps are usually represented in diagrams in one of two ways. 

Either as two filaments (i.e. two triangles each representing one half of the filament):
[image]

Or just a tap in the center of one filament:
[image]

It seems to me that the former is more common, but neither is unusual in my estimation, and the latter would probably be simpler to add.

Rmano, 2024-05-18:
Hi @bogger33  : what do you think of #798?

bogger33, 2024-05-18:
it looks good to me!

commit bef6dc34 (2024-05-19): Merge pull request #798 from Rmano/filament-center-tap

Add a center tap for tube's filament

Closes #792
closed 2024-05-19 by Rmano