circuitikz/circuitikz #793: [New Component] Neon Lamp

bogger33, 2024-05-13:
**Show the component and its usage**

The neon lamp is still often used as an indicator on mains voltage circuits due to its high voltage drop, requiring only a resistor and otherwise no power supply. Due to its negative resistance property it can also be used as an active switching element, which was sometimes done in e.g. older CRT televisions.

The closest component in the circuitikz library right now appears to be the gas filled surge arrester (in particular the american variant has a similar symbol).

**Add references to standards, usage, etc.**
Once again I have prepared a collage of circuit symbols I could find for the neon bulb

[image]

The most common appear to be the two variants at the top, in particular the one on the top left. 

I apologize for all these requests, this should be the last one for a while.

Rmano, 2024-05-13:
😜 no need to apologize. These are quite easy symbols to add (the top right seems to be a solder jumper with a circle around it, and the top left is basically a circled capacitor). Which one do you think it's the best one to bet on?

bogger33, 2024-05-13:
I looked around and it seems that the top left is the one that's used the most in modern documents at least at a cursory glance, although there unfortunately really doesn't seem to be any consensus. I also looked if I could find a standard regarding this, interestingly I did find that IEEE 315-1975 defines a the symbol on the bottom left. You can view it here:

https://www.iprocessmart.com/techsmart/elecsymbols.htm

It's on page 4, at the top right. It doesn't seem to be very commonly used, however, though I've seen it a few times. I think I personally would go with the top left, but I also would be happy with any other variation being added instead.

Rmano, 2024-05-18:
What about these?

[image]

bogger33, 2024-05-18:
Those are nice! Actually seeing these side by side now also makes me think it might be useful to have both for dc and ac circuits respectively (where there either is or isn't a clear anode and cathode respectively). 

Maybe the names could be "double cathode style" and "cathode and anode style" rather than "capacitor" and "tube"?

Rmano, 2024-05-18:
Yes, clearly, I can change the names. Maybe `neonlampcc` and `neonlampca`?

bogger33, 2024-05-18:
makes sense to me

PR #799 (2024-05-19): Add neon lamps, in two versions.
Suggested by @bogger33 in issue #793
Closes #793

[image]
commit c9c1c4f8 (2024-05-19): Add neon lamps, in two versions.

Suggested by @bogger33 in issue #793
Closes #793
closed 2024-05-19 by Rmano