circuitikz/circuitikz #806: Headphone jack connector symbol

anisiobraga, 2024-06-19:
A headphone jack connector is widely used in circuit diagrams related to audio input- output but it is also an input connector easily available at low cost to connect sensors probes like pH sensor with up to 4 pins. Some phone jack connectors has internal switches that allow detection of the pin insertion. Although there is a large number of configurations used for phone jacks the characteristics of the symbols are quite distinctive including a ring pin, a v-notch tip and an arrow tip at 90 degrees representing internal switches. 

**Show the component and its usage**
[image]

A typical usage is the  circuit diagram of an audio connector used in notebooks to switch between internal speakers and a headphone.
 

**References to standards, usage, etc.**
This wikipedia presents the diversity of phone jack connectors and some symbols.
[https://en.wikipedia.org/wiki/Phone_connector_(audio)](url)

This link is a pdf file of General Specifications of 2.5/3.5 Phone Jack Series
https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/6/2.5_2D00_3.5_2D00_phone_2D00_jack.pdf&ved=2ahUKEwjUuL322eeGAxVxppUCHdJgAd0QFnoECBYQAQ&usg=AOvVaw02afOP0K077ZdJG97t5z8K

Rmano, 2024-06-19:
Thanks for the suggestion! One problem I see is that there are tens of different symbols, I do not think it's reasonable (or at least, I do not have the time ) to add all of them.
Maybe there is a way to generalize them? Otherwise, one option is to just add the mono- and stereo- jacks and see if somebody can add the more complex one along the way...
Any idea?

Rmano, 2024-06-19:
I mean, we can start adding these:

[image]

What do you think?

eclecticfluff, 2024-06-19:
I don’t have much use for the symbols myself, but the problem of how to generalize them intriuged me.
[image]
I’m not sure if it is possible to make tikz keys take a set of values like this, but if it is, I think this is a sensible way to allow for customization to cover most if not all cases.

anisiobraga, 2024-06-20:
The idea of adding a mono and a stereo covers much of the applications including audio P3 in notebooks. 
Nevertheless, the code provided by Rmano has the directions to help anyone with a usage of audio jack connectors. 

Anisio R. Braga
COLTEC/UFMG
***@***.***

anisiobraga, 2024-06-20:
The proposed systematization is interesting although  the task of transforming this into a generic configurable component extremely laborious considering the wide variety of configurations. 
However, the idea of ​​creating nodes (tips and arrows as illustrated) that can be added to make up the various configurations can be a less laborious alternative that adds interesting features to Circuitikz. An example illustrating how to compose a mono and a stereo audio jack connectors is already helpful. For example, I've used the code snippet suggested by Rmano to edit a circuit with and without a plug inserted as follows:

￼
%--- Headphone jack
\def\dx{1.25} \def\dy{1.25}
\begin{circuitikz}
        \draw (0,\dy/3*4) to[short, o-] ++(\dx,0) -- ++(-45:0.2) -- ++(45:0.2);
        \draw[-Triangle] (0,\dy) node[ocirc]{} -| ++(\dx/3,\dy/3);
        \draw[-Triangle] (0,\dy/3*2) node[ocirc]{} -| ++(\dx/3,-\dy/3);
        \draw (0,\dy/3) to[short, o-] ++(2*\dx/3,0) -- ++(45:0.2) -- ++(-45:0.2);
        \draw (0,0.0) to[short, o-] ++(1.75*\dx,0) rectangle ++(0.2,4*\dy/3);
\end{circuitikz}

%--- % audio jack with an inserted plug
\def\dx{1} \def\dy{1}
\begin{circuitikz}
        \draw (0,\dy/3*4) to[short, o-] ++(\dx,0.2) -- ++(-45:0.2) -- ++(45:0.2);
        \draw[-Triangle] (0,\dy) node[ocirc]{} -| ++(\dx/3,\dy/3);
        \draw[-Triangle] (0,\dy/3*2) node[ocirc]{} -| ++(\dx/3,-\dy/3);
        \draw (0,\dy/3) to[short, o-] ++(2*\dx/3,-0.2) -- ++(45:0.2) -- ++(-45:0.2);
        \draw (0,0.0) to[short, o-] ++(1.75*\dx,0) rectangle ++(0.2,4*\dy/3);
\end{circuitikz}

Sincerely,

Anisio

Rmano, 2024-06-21:
> The proposed systematization is interesting although the task of transforming this into a generic configurable component extremely laborious considering the wide variety of configurations. 

Yes, that was the problem I was stumbling with. It can be done, but it will be a monster even more complex than the `muxdemux` thing. 

> However, the idea of ​​creating nodes (tips and arrows as illustrated) that can be added to make up the various configurations can be a less laborious alternative that adds interesting features to [...]

Let me repost your code as a compilable snippet with image:

```latex
\documentclass{article}
% \documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}

\begin{document}
%--- Headphone jack
\def\dx{1.25} \def\dy{1.25}
\begin{circuitikz}
        \draw (0,\dy/3*4) to[short, o-] ++(\dx,0) -- ++(-45:0.2) -- ++(45:0.2);
        \draw[-Triangle] (0,\dy) node[ocirc]{} -| ++(\dx/3,\dy/3);
        \draw[-Triangle] (0,\dy/3*2) node[ocirc]{} -| ++(\dx/3,-\dy/3);
        \draw (0,\dy/3) to[short, o-] ++(2*\dx/3,0) -- ++(45:0.2) -- ++(-45:0.2);
        \draw (0,0.0) to[short, o-] ++(1.75*\dx,0) rectangle ++(0.2,4*\dy/3);
\end{circuitikz}\qquad\qquad
%--- % audio jack with an inserted plug
\def\dx{1} \def\dy{1}
\begin{circuitikz}
        \draw (0,\dy/3*4) to[short, o-] ++(\dx,0.2) -- ++(-45:0.2) -- ++(45:0.2);
        \draw[-Triangle] (0,\dy) node[ocirc]{} -| ++(\dx/3,\dy/3);
        \draw[-Triangle] (0,\dy/3*2) node[ocirc]{} -| ++(\dx/3,-\dy/3);
        \draw (0,\dy/3) to[short, o-] ++(2*\dx/3,-0.2) -- ++(45:0.2) -- ++(-45:0.2);
        \draw (0,0.0) to[short, o-] ++(1.75*\dx,0) rectangle ++(0.2,4*\dy/3);
\end{circuitikz}
\end{document}
```
[image]

Maybe just having an arrow tip doing the 
[image]
thing would help? Although it's trivial to put the ` -- ++(45:0.2) -- ++(-45:0.2)` code into a macro, the arrow tip will follow the inclination...

anisiobraga, 2024-06-22:
The arrow tip-v and a code snippet illustrating the usage of mono/stereo audio jack adds  a very nice starting point to this kind of application, and simplifies it. Interesting is to realize that after considering a hard and complex task of generating a configurable component symbol, providing a hint of how to draw an arrow tip-v is sufficient to accomodate nicely an audio jack symbol.
 
Anisio

Rmano, 2024-07-04:
So, I learned how to design arrow tips (I think...). For example, these are the plain tip v-style. I do not know if it makes sense to add more...

```latex
\documentclass[border=2.78pt]{standalone}
\usepackage{tikz}
\usetikzlibrary{arrows.meta}
\pgfdeclarearrow{name=Jack Tap,
    parameters = {
    \the\pgfarrowlength,%
    \the\pgfarrowwidth%
    },
    setup code = {
        \pgfarrowssettipend{.5\pgfarrowlength}
        \pgfarrowssetlineend{-.6\pgfarrowlength}
        \pgfarrowssetvisualbackend{-.6\pgfarrowlength}
        \pgfarrowssetbackend{-.6\pgfarrowlength}
        % hull
        \pgfarrowshullpoint{.5\pgfarrowlength}{0pt}
        \ifpgfarrowswap
            \pgfarrowshullpoint{0pt}{-\pgfarrowwidth}
        \else
            \pgfarrowshullpoint{0pt}{\pgfarrowwidth}
        \fi
        \pgfarrowshullpoint{-.6\pgfarrowlength}{0pt}
        % Saves: Only the length:
        \pgfarrowssavethe\pgfarrowlength
        \pgfarrowssavethe\pgfarrowwidth
    },
    drawing code = {
        \pgfpathmoveto{\pgfqpoint{.5\pgfarrowlength}{0pt}}
        \ifpgfarrowswap
            \pgfpathlineto{\pgfqpoint{0pt}{-\pgfarrowwidth}}
        \else
            \pgfpathlineto{\pgfqpoint{0pt}{\pgfarrowwidth}}
        \fi
        \pgfpathlineto{\pgfqpoint{-.5\pgfarrowlength}{0pt}}
        \pgfpathlineto{\pgfqpoint{-.6\pgfarrowlength}{0pt}}
        \pgfusepathqstroke
    },
    defaults = {length = 0.2cm, width=0.1cm }
}

\begin{document}
\begin{tikzpicture}[]
\draw [-{Turned Square}] ( 0,0) -- ++(2,0);
\draw [-{Jack Tap}] ( 0,.3) -- ++(2,0);
\draw [-{Jack Tap[swap]}] ( 0,.6) -- ++(2,0);
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2024-07-05:
So the thing will become this: what do you think?  

```latex
\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}

\pgfdeclarearrow{name=Jack Tap,
    parameters = {%
    \the\pgfarrowlength,%
    \the\pgfarrowwidth,
    \ifpgfarrowswap s\fi%
    },
    setup code = {
        \pgfarrowssettipend{.5\pgfarrowlength}
        \pgfarrowssetlineend{-.6\pgfarrowlength}
        \pgfarrowssetvisualbackend{-.6\pgfarrowlength}
        \pgfarrowssetbackend{-.6\pgfarrowlength}
        % hull
        \pgfarrowshullpoint{.5\pgfarrowlength}{0pt}
        \pgfarrowshullpoint{0pt}{\pgfarrowwidth}
        \pgfarrowshullpoint{-.6\pgfarrowlength}{0pt}
        % Saves: Only the length:
        \pgfarrowssavethe\pgfarrowlength
        \pgfarrowssavethe\pgfarrowwidth
    },
    drawing code = {
        \pgfpathmoveto{\pgfqpoint{.5\pgfarrowlength}{0pt}}
        \pgfpathlineto{\pgfqpoint{0pt}{\pgfarrowwidth}}
        \pgfpathlineto{\pgfqpoint{-.5\pgfarrowlength}{0pt}}
        \pgfpathlineto{\pgfqpoint{-.6\pgfarrowlength}{0pt}}
        \pgfusepathqstroke
    },
    defaults = {length = 0.3cm, width=0.15cm},
    % cache=false, % breaks everything
}
\tikzset{v/.tip={Jack Tap[swap]}, ^/.tip={Jack Tap}}

\newcommand\dx{1.5} \newcommand\dy{1}
\begin{document}

\begin{circuitikz}
    \draw[-v] (0,\dy/3*4) to[short, o-] ++(\dx,0);
    \draw[-Triangle] (0,\dy) node[ocirc]{} -| ++(\dx/3,\dy/3);
    \draw[-Triangle] (0,\dy/3*2) node[ocirc]{} -| ++(\dx/3,-\dy/3);
    \draw[-^] (0,\dy/3) to[short, o-] ++(2*\dx/3,0);
    \draw (0,0.0) to[short, o-] ++(1.75*\dx,0) rectangle ++(0.2,4*\dy/3);
\end{circuitikz}\qquad\qquad
%--- % audio jack with an inserted plug
\begin{circuitikz}
    \draw[-v] (0,\dy/3*4) to[short, o-] ++(\dx,0.2);
    \draw[-Triangle] (0,\dy) node[ocirc]{} -| ++(\dx/3,\dy/3);
    \draw[-Triangle] (0,\dy/3*2) node[ocirc]{} -| ++(\dx/3,-\dy/3);
    \draw[-^] (0,\dy/3) to[short, o-] ++(2*\dx/3,-0.2);
    \draw (0,0.0) to[short, o-] ++(1.75*\dx,0) rectangle ++(0.2,4*\dy/3);
\end{circuitikz}
\end{document}
```

[image]

anisiobraga, 2024-07-05:
That’s an elegant solution that adds simplicity to draw audio jack connectors. Thanks. Anisio R. Braga

Rmano, 2024-07-09:
I added more options, see https://github.com/circuitikz/circuitikz/pull/810#issuecomment-2218402740 !

anisiobraga, 2024-07-09:
> I added more options, see #810 (comment) <https://github.com/circuitikz/circuitikz/pull/810#issuecomment-2218402740> !

Great. With these new styles with fill and width options most types of audio connectors can be drawn easily and consistently.

Anisio R. Braga
COLTEC/UFMG
***@***.***

Rmano, 2024-07-10:
I am waiting to see if there is an answer to https://github.com/pgf-tikz/pgf/issues/1348, to see if the option is to be called `filled` or `closed` or whatever.

Rmano, 2024-07-19:
Ultimately, the TikZ decision made things a bit complicated... on the nice side, I learned a lot from the process.
Thanks @Skillmon !

I will merge now; I hope to be able to release 1.6.10  before going on vacation...

commit 7cd7e732 (2024-07-19): Add `Jack Tap` arrow tip (#810)

Add a new type of arrow to help drawing Jack connectors

Fixes #806 

See also  https://github.com/pgf-tikz/pgf/pull/1352
Thanks to @Skillmon
https://github.com/pgf-tikz/pgf/issues/1348#issuecomment-2231681032
closed 2024-07-19 by Rmano