circuitikz/circuitikz #807: [New Component(may be)]A different form for current probe

EEpchi, 2024-06-27:
**Show the component and its usage**
It's just another form of current probe, I think it's widely used in my country.

Add an image of the new component and typical usage in a circuit.

[image]

**references**
Someone had already asked about this form of current probe in [https://tex.stackexchange.com/questions/521103/tikz-diagram-for-current-sensor-measurement](url), and I have drawn a circuit with it using existing version of `circuitikz`. 
```latex
\documentclass[UTF-8,a4paper]{ctexart}
\usepackage{unicode-math}
\setmainfont{TeX Gyre Termes}
\setmathfont{TeX Gyre Termes Math}
\setsansfont{TeX Gyre Termes}
\usepackage[europeanresistors,americaninductors,americanvoltages,siunitx]{circuitikz}
\usetikzlibrary{shapes.geometric}
\begin{document}
\pagestyle{empty}
\ctikzset{tripoles/mos style/arrows, transistors/arrow pos=end}
\ctikzset{resistors/scale=0.6, capacitors/scale=0.6, inductors/scale=0.6, capacitors/height=0.4,capacitors/width=0.15,diodes/scal
\begin{circuitikz}[circuitikz/.cd,nodes width=0.025,bipoles/thickness=1.5,thyristor style=compact,diodes/scale=0.4,resistors/scale=0.4,transistors/scale=0.7,tripoles/thickness=1.5,muxdemuxes/thickness=1.5]
\tikzset{op/.style={muxdemux,muxdemux def={NL=2, Lh=4, NR=1, Rh=0, NB=0,NT=0,w=3.5}}}
	\small
	\draw (0,0) to[short,-*,i,name=s] ++(1.5,0) to[D-,mirror,l=$\mathrm{VD_1}$] ++(0,1) to[short,-*] ++(0.8,0) to[D-,invert,l=$\mathrm{VD_3}$] ++(0,-1) to[short,-*] ++(0,-0.8) to[D-,invert,-*,l=$\mathrm{VD_4}$] ++(0,-1) -- ++(-0.8,0) to[D-,mirror,l=$\mathrm{VD_2}$] ++(0,1) -- ++(0,0.8);
	\draw (0,0) to[open,o-o,v,font=\small,name=o] ++(0,-0.8) -- ++(2.3,0);
	\draw (o-Vfrom) ++(0,-0.05) node{\small $+$} (o-Vto) ++(0,-0.05) node{\small $-$} (o-Vlab) ++(0,-0.05) node{\small $u$};
	\draw (2.3,1) to[short,n=ss,i] ++(1.05,0) arc(180:0:0.15) -- (3.8,1) to[L,l=$L$,-*,label distance=-2pt] ++(1.5,0) to[Tnigfete,n=Q,invert,-*] ++(0,-2.8) -- ++(-3,0) (Q) node[right]{Q};
	\draw (5.3,1) to[D-] ++(1.5,0) to[C,l_=$C$,name=C,i,*-*] ++(0,-2.8) -- ++(-1.5,0);
	\draw (6.8,1) -- ++(1,0) to[R,l_=$R$,i,name=R,v^=$u_\mathrm{d}$,voltage shift=-0.9] ++(0,-2.8) -- ++(-1,0);
	\path[iarrow] (s-Ifrom) -- (s-Ipos) node[below]{$i$};
	\path[iarrow] (ss-Ifrom) -- (ss-Ipos) node[above]{$i_\mathrm{L}$};
	\path[iarrow] (R-Ifrom) -- (R-Ipos) node[left]{$i_\mathrm{R}$};
	\path[iarrow] (C-Ifrom |- R-Ifrom) -- (C-Ipos |- R-Ipos) node[left]{$i_\mathrm{C}$};
	\draw (1.5,-5) node(u){$\lvert u \rvert$} ++(2,0) node[draw](times){\scriptsize multiplier};
	\draw (C-Ipos |- u) node[circle,draw](c1){} ++(0,1) node(ud){$u_\mathrm{d}$} ++(1,-1) node(ur){$u_\text{ref}$} ($(c1)!0.5!(times.east)$) node[draw,minimum width=.5cm,minimum height=.5cm](PI1){\footnotesize PI};
	\draw ($(c1.north)!0.5!(c1.center)$) node{\tiny\bfseries --} ($(c1.east)!0.5!(c1.center)$) node{\tiny\bfseries +};
	\draw (times) ++(0,1) node[circle,draw](c2){} ($(c2.north)!0.5!(c2.center)$) node{\tiny\bfseries --} ($(c2.south)!0.5!(c2.center)$) node{\tiny\bfseries +};
	\draw (c2) ++(0,1) node[draw,align=center,font=\scriptsize,inner sep =2pt](i){Input current \\[-2pt] detection};
	\draw (3.5,1) node[circ,circuitikz/nodes width=0.05](node){} -- (i.north);
	\draw (Q.G) -- ++(0,-1.4) node[op,anchor=rpin 1,rotate=90,scale=0.35,draw only left pins={}](op){} (op.blpin 1) node[above=-2pt]{\footnotesize\bfseries +} (op.blpin 2) node[above=-2pt]{\rotatebox{-90}{\footnotesize\bfseries --}};;
	\draw (c2 -| op.blpin 1) node[draw,minimum width=.5cm,minimum height=.5cm](PI2){\footnotesize PI};
	\draw (op.blpin 2 |- i.south) node(uc){$ u_\mathrm{c} $};
	\foreach \i/\j in {u.east/times.west,
	times.north/c2.south,
	c2.east/PI2,
	i.south/c2.north,
	ur.west/c1.east,
	ud.south/c1.north,
	c1.west/PI1.east,
	PI1.west/times.east,
	PI2.north/op.blpin 1,
	uc.north/op.blpin 2}{
		\draw[farrow] (\i) -- (\j);}	
\end{circuitikz}
\end{document}
```
[image]

I think it's easier to draw if I can use existing components, so I'd be appreciated if  I can have it in this package🌹.

Rmano, 2024-06-27:
This should be quite easy to add. In which country do you say it's used? It's a quite "non-electrical" symbol (unless the ground connection is implicitly assumed),  but I think I'll add it when I have a bit of time.

Rmano, 2024-06-27:
For example, this would be ok? 

[image]

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}

\makeatletter
\ctikzset{bipoles/currtap/height/.initial=0.4}
\ctikzset{bipoles/currtap/dot size/.initial=0.5}
%% currtap
\pgfcircdeclarebipolescaled{instruments}
{
    \savedmacro{\@@dotsize}{\def\@@dotsize{\ctikzvalof{bipoles/currtap/dot size}}}
    \anchor{tap}{\northeast\pgf@x=0pt\pgf@y=\@@dotsize\pgf@y\pgf@y=-\pgf@y}
}
{\ctikzvalof{bipoles/currtap/height}}
{currtap}
{\ctikzvalof{bipoles/currtap/height}}
{\ctikzvalof{bipoles/currtap/height}}
{
    \pgfscope
        \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{0pt}}
        \pgfpatharc{180}{0}{\pgf@circ@res@right}
        \pgf@circ@draworfill
    \endpgfscope
    \pgfpathcircle{\pgfpoint{0pt}{0pt}}{\@@dotsize*\pgf@circ@res@up}
    \pgfusepath{fill}
}
\pgfcirc@activate@bipole@simple{l}{currtap}
\makeatother

\begin{document}
\section{Dummy sec}
\subsection{Dummy subsec}
\subsubsection{Dummy subsubsec}

\begin{groupdesc}
    \circuitdescbip*{currtap}{Current tap (probe)}{}(tap/-90/0.2)
\end{groupdesc}

% \geolrcoord{currtapshape}

\begin{LTXexample}[varwidth=true, basicstyle=\small\ttfamily]
\begin{circuitikz}[european]
    \draw (0,0) to[currtap=I, *-*, name=ct] ++(3,0);
    \draw (ct.tap) -- ++(0,-1);
\end{circuitikz}
\end{LTXexample}

\end{document}

```

EEpchi, 2024-06-28:
> For example

wow, that's very nice.

> In which country do you say it's used?

I'm in China. Our textbook has it:
[image]
maybe they learned it from PSIM😂：

[image]

And I think it would be perfect if I could freely change the size of the solid circle in the middle：

[image]

Anyway, thank you for your efforts🌹

Rmano, 2024-06-28:
> And I think it would be perfect if I could freely change the size of the solid circle in the middle：

You can: 

[image]

(It's a relative size). My doubt is if to make it fillable: 
[image]
Does it make any sense, and if it does, should it fill just the semicircle or a full circle around the inner dot?

EEpchi, 2024-06-28:
> if to make it fillable

I don't think it makes much sense to me to fill in(or just I can't get it). But I think if it's for emphasis or whatever, it's okay to fill just the semicircle

Rmano, 2024-07-02:
I am removing the fill option for now. PR in a few minutes...

Dr4UX, 2024-07-25:
I've just stumbled across this component. Funnily enough, I designed exactly the same component two days ago and wanted to propose it in the next days.

I am used to this component with the circle in the middle only drawn and not filled. Maybe it would be an option to use the `fill`-option to fill the inner circle? Or add a `not filled` parameter to realise this?

Rmano, 2024-07-25:
What about: 

[image]

Dr4UX, 2024-07-25:
looks perfect. Thank you!

Rmano, 2024-07-26:
Going full-monty:

[image]

PR #808 (2024-07-02): Add a different current probe (currtap)
Fixes #807 

[image]
PR #815 (2024-07-26): More configurable currdot component
Thanks to Dr4UX for hints; closes #807

[image]
commit d3908855 (2024-07-26): More configurable currdot component

Thanks to Dr4UX for hints; closes #807
closed 2024-07-26 by Rmano