circuitikz/circuitikz #809: [New Component] Transient generator

jmonclard, 2024-07-02:
Idea for a future release : a source with an inner symbol representing transient waveform. Would be of interest in schematics and drawings relative to device EMC qualification tests (indirect lightning surge, ESD, electrical fast transient burst,...).

Typical waveforms used in industrial/aeronautic/military applications include unipolar and damped sinusoid. So two different symbols would be nice.

Such tests are for instance defined in IEC EN 61000-4 standards (61000-4-2, 61000-4-5, 61000-4-4)

**Example of usage in a specification document for aero applications**

[image]

**Example of symbol used in LT Spice circuit simulator**
[image]

**Hand drawing example of a damped sinusoid generator symbol**
[image]

Rmano, 2024-07-03:
Hm... I do not know if that one is really needed. Those are typically one-shot components, in the sense that you have an off-one in a circuit, so it can be easily built by drawing on an empty source.

```latex
\documentclass[border=3mm]{standalone}
\usepackage{circuitikz}
\newcommand{\addVpulse}[1]{%
    \draw (#1.center) ++(-0.3,-0.2) -- ++(0.1,0) -- ++(0,0.4)
        to[bend right] ++(0.4,-0.4) -- ++(0.1,0);
    }
\begin{document}
\begin{circuitikz}
    \draw (0,0) to[esource, v=$V_p$, name=vp] ++(0,2);
    \addVpulse{vp}
\end{circuitikz}
\end{document}
```

[image]

An option could be to add some mechanism for including arbitrary drawings to a source so that we can define a style similar to what is done for oscilloscopes (Look for "Oscilloscope waveform" in the manual, around page 81)

Rmano, 2024-11-10:
I will close this for now --- please feel free to reopen if you think it is not answered!

closed 2024-11-10 by Rmano