circuitikz/circuitikz #811: [Bug] bad line joint at the thermocouple element

Dr4UX, 2024-07-23:
I noticed that the connection of the `thermocouple` element is bad since the line caps are not lining up. You can see it in the picture below.

```latex
\documentclass{standalone}
\usepackage{circuitikz}

\begin{document}
\begin{tikzpicture}
    \draw (0,0) to [thermocouple] ++(1,0);
\end{tikzpicture}
\end{document}
```

[image]

It gets better when you correct the left and right points of the element in its definition, but I have no idea, how you can adjust the joints for the thick part. 

```latex
\pgfcircdeclarebipolescaled{misc}
{}
{\ctikzvalof{bipoles/thermocouple/height 2}}
{thermocouple}
{\ctikzvalof{bipoles/thermocouple/height}}
{\ctikzvalof{bipoles/thermocouple/width}}
{
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left-\pgfhorizontaltransformationadjustment*0.5*\pgflinewidth}{0}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@left-\pgfhorizontaltransformationadjustment*0.5*\pgflinewidth}{.5\pgf@circ@res@down}}
    \pgfusepath{draw}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left-\pgfhorizontaltransformationadjustment*0.5*\pgflinewidth}{.5\pgf@circ@res@down}}
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfpathlineto{\pgfpoint{0}{\pgf@circ@res@down}}
    \pgfusepath{draw}
    \pgfsetlinewidth{\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpoint{0}{\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right+\pgfhorizontaltransformationadjustment*0.5*\pgflinewidth}{.5\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right+\pgfhorizontaltransformationadjustment*0.5*\pgflinewidth}{0}}
    \pgfusepath{draw}
}
```

[image]

Rmano, 2024-07-23:
Ouch! Yes, this is a problem when you stop a drawing and start a new one. In this case, I would draw the whole element in one go and then superpose the thick part. I'll try (and check the resistor too). As you say, a perfect solution is basically impossible unless the component starts with a horizontal line. You can probably make it less noticeable with `line cap=round` or something similar.

Rmano, 2024-07-23:
Like:

```latex
\documentclass{standalone}
\usepackage{circuitikz}
\makeatletter
\pgfcircdeclarebipolescaled{misc}
{}
{\ctikzvalof{bipoles/thermocouple/height 2}}
{thermocouple}
{\ctikzvalof{bipoles/thermocouple/height}}
{\ctikzvalof{bipoles/thermocouple/width}}
{
    \pgfsetrectcap
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{0}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@left}{.5\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{0}{\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{.5\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0}}
    \pgfusepath{draw}
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfsetroundcap
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{.5\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{0}{\pgf@circ@res@down}}
    \pgfusepath{draw}
}
\makeatother
\begin{document}
\begin{tikzpicture}
    \draw (0,0) to [thermocouple] ++(1,0);
\end{tikzpicture}
\end{document}

```

[image]

Rmano, 2024-07-23:
BTW, I do not see any special problem with the resistor...

[image]

Rmano, 2024-07-24:
Like, maybe, this:

[image]

Rmano, 2024-07-24:
We can continue the discussion in #814

PR #812 (2024-07-24): Change the drawing of thermocouple for better joins
Fixes #811
commit ffdcefaa (2024-07-24): Change the drawing of thermocouple for better joins

Fixes #811
closed 2024-07-24 by Rmano