circuitikz/circuitikz #821: [Bug] Open voltage arrow and label overlap

wallscheid, 2024-08-15:
**Describe the bug**
The automatically generated voltage label of an open circuit voltage arrow can overlap with the arrow, see
[image]

**To Reproduce**
```latex
\documentclass{article}
\usepackage[european, smartlabels]{circuitikz}
\begin{document}
\begin{circuitikz}[]
            \draw (0,2) to [open, o-o, voltage = straight] ++(0,-2)
            to ++(-4,0)
            to [vsource, v<=$u_\mathrm{in}(t)$, i=$i_\mathrm{in}(t)$] ++(0,2)
            to [variable european resistor] ++(4,0) ;
            \draw (0,2) to [open, v = $ u_\mathrm{in}(t)$, voltage = straight] (0,0);
 \end{circuitikz}
\end{document}
```

**Operating system, LaTeX distribution**
Win 11 pro, MikTeX-pdfTeX 4.19

**Expected behavior**
Label is properly left or right aligned without overlapping - compare voltage source on the left hand side of the above screenshot, where the behavior works as expected

Rmano, 2024-08-15:
Seen, I'll look at it as soon as I can --- on vacations now!

Rmano, 2024-08-20:
I can confirm the bug, that can be reproduced with the minimal example:

```latex
\documentclass[]{article}
\usepackage{circuitikz}
\ctikzset{voltage=straight}
\begin{document}
    \begin{circuitikz}[]
        \draw (2,0) to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(0,2);
    \end{circuitikz}
\end{document}
```

[image]

I'll look into it. As a workaround, you can use (adjusting the shift by tries)

```latex
\draw (2,0) to [open, o-o, v = $ u_\mathrm{in}(t)$, voltage shift=2] ++(0,2);
```
[image]

Rmano, 2024-08-20:
So, it seems that these lines:

https://github.com/circuitikz/circuitikz/blob/47c19c27bf7174eb80332c044df4a0c85ffadf3e/tex/pgfcircvoltage.tex#L194-L200

are the culprits for the bug, and removing them will fix it. The puzzling thing is that **I** added those lines, and I can't remember why.  I will give it some more regression checks before applying; in the meantime, you can use the live patch below.

```latex
\documentclass[]{article}
\usepackage[european, smartlabels]{circuitikz}
\usepackage{etoolbox}
\makeatletter
\patchcmd{\pgf@circ@drawvoltagegeneric}{%
        \ifpgf@circuit@bipole@voltage@straight
            \ifx\@@kind\@@open
                \def\pgf@circ@bipole@voltage@label@anchor{center}
                \pgfmathsetlength{\absvshift}{\absvshift+sign(\absvshift)*height{"Q"}} % with the current font.
            \fi
    \fi}{}{}{\typeout{Patching of void straight voltage failed}}
\makeatother
\begin{document}
\begin{circuitikz}[]
    \draw (0,2) to [open, o-o, voltage = straight] ++(0,-2)
    to ++(-4,0)
    to [vsource, v<=$u_\mathrm{in}(t)$, i=$i_\mathrm{in}(t)$] ++(0,2)
    to [variable european resistor] ++(4,0) ;
    \draw (0,2) to [open, v = $ u_\mathrm{in}(t)$, voltage = straight, voltage shift=2] (0,0);
\end{circuitikz}

\vspace{1cm}

\begin{circuitikz}[]
    \ctikzset{voltage=straight}
    \draw (0,0) to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(0,2)
        to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(2,0);
    \ctikzset{voltage=raised}
    \draw (3,0) to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(0,2)
        to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(2,0);
    \ctikzset{voltage=american}
    \draw (6,0) to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(0,2)
        to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(2,0);
    \ctikzset{voltage=european}
    \draw (9,0) to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(0,2)
        to [open, o-o, v = $ u_\mathrm{in}(t)$] ++(2,0);
\end{circuitikz}
\end{document}
```

[image]

PR #823 (2024-08-23): Possible fix for the bug #871
The original code has been left and commented out --- I added it in the past, so there must have been a reason. Let's see what happens.

Fixes #821
closed 2024-08-23 by Rmano