circuitikz/circuitikz #825: [Bug] Voltage Directions Battery

myzinsky, 2024-08-28:
[image]

Looking in the manual, should the volatage arrow of the battery also not be inverted like for diode and resistors? Seems that it it not influenced at all. 

And another question: 

[image]

Why is the arrow much smaller compared to the other elements? This looks a bit weird :-)

Rmano, 2024-08-28:
Do you mean the arrow in `EF` ? Maybe... that is a real mess of options there; with the hindsight of all these years of maintenance, what I think would be better is to forget voltage direction, draw everything in the direction of the bipole, and let the user adjust with `<` `>` or whatever. I think I'll not touch that code with a three-meter pole until I decide to go with `circuitikz-ng` and break backward compatibility (don't hold your breath; that is a retirement project, almost eight years to go...)

About the second question, I don't understand what you think is "smaller"... the arrow size can be configured (even styled, I think)...

myzinsky, 2024-08-28:
Yeah you're right. I have almost 1000 pictures drawn with circuit tikz. If we change this basis, it will break a lot of pictures. I just stumbled over this. :-)

Rmano, 2024-08-28:
Ok, I'll close as "can't-fix" ;-)

closed 2024-08-28 by Rmano