circuitikz/circuitikz #827: [New Component] Four-channel oscilloscope

steigies, 2024-09-04:
Four channel oscilloscopes are becoming more common, I would like to see them as a component in circuitikz:
[TDS3000C](https://download.tek.com/datasheet/TDS3000C-Digital-Phosphor-Oscilloscope-Datasheet-41W1248226.pdf)

I have tried to add two inputs the existing oscope, but this results in: Package PGF Math Error: Unknown function `in' (in 'in 3').

With four inputs, maybe the component needs to be added to pgfcircquadpoles.tex instead of bipoles, so:
- I tried to copy oscope as fourscope and
- I tried to copy the fourport coupler and coupler2 as coupler4
but I am getting pgfkeys errors, so I must be missing something.

I am sorry that I can not provide a sample implementation, but it should not be that difficult to add two more inputs to a component?

Rmano, 2024-09-04:
Well, in reality the oscilloscope symbol is intended as a *symbolic* representation that the test point is connected to a track in an oscilloscope --- it is **not** meant to be a physical representation. So basically, I'd use 4 separated symbols.

I recon that there is an example of physical connection in the manual --- on page 87  with the current one. If you want, you can use a similar thing --- you can use calculated coordinates to add "points" on the symbols; I don't think that adding an oscilloscope with arbitrary number of anchors is useful, given that it is very easy with the native Ti*k*Z  tools: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\tikzset{big oscope/.style={oscopeshape,
    circuitikz/bipoles/oscope/width=1.2,
    circuitikz/instruments/scale=1.5}
}
\begin{document}
\begin{tikzpicture}[]
    \draw (1,-1) node[big oscope](O2){};
    \draw (O2.in 2) to[short, *-] ++(0,-0.5);
    \draw ($(O2.in 2)!0.33333!(O2.in 1)$) to[short, *-] ++(0,-0.5);
    \draw ($(O2.in 2)!0.66667!(O2.in 1)$) to[short, *-] ++(0,-0.5);
    \draw (O2.in 1) to[short, *-] ++(0,-0.5);
\end{tikzpicture}
\end{document}
```

[image]

Or, if you want to use more of the space,

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\tikzset{big oscope/.style={oscopeshape,
    circuitikz/bipoles/oscope/width=1.2,
    circuitikz/instruments/scale=1.5}
}
\begin{document}
\begin{tikzpicture}[]
    \draw (1,-1) node[big oscope](O2){};
    \draw ($(O2.in 2)!-0.25!(O2.in 1)$) to[short, *-] ++(0,-0.5);
    \draw ($(O2.in 2)!0.25!(O2.in 1)$) to[short, *-] ++(0,-0.5);
    \draw ($(O2.in 2)!0.75!(O2.in 1)$) to[short, *-] ++(0,-0.5);
    \draw ($(O2.in 2)!1.25!(O2.in 1)$) to[short, *-] ++(0,-0.5);
\end{tikzpicture}
\end{document}
```

[image]

steigies, 2024-09-04:
I don't want an arbitrary number of inputs, just four. But your example looks just like what I want, thank you!

Rmano, 2024-09-04:
> I don't want an arbitrary number of inputs, just four. 

Yes, I see ;-) But adding a 4-inputs (5? with GND?) would be a bad design --- somebody can ask for 4 or 6 or... so the logical thing would be to change to have a (backward-compatible) arbitrary number of inputs.

> But your example looks just like what I want, thank you!

Happy to help! I will close this issue now, please feel free to reopen if you need it.

closed 2024-09-04 by Rmano