circuitikz/circuitikz #828: [New Component] Chopper block and chopper amplifier

PJansky, 2024-11-05:
I would be interested in seeing chopper modulator and a general chopper amplifier added to circuitikz.

[image]
In this image, CH1, CH2P and CH2P represent chopping modulator comprised of 4 crossed out switches. In this use case with chopper blocks are added to simplify the circuit as opposed to showing the 4 separate transistors.

A symbol for a chopped opamp would also be nice. I've seen symbols for chopper amplifiers like the image below, with a small 
modulator block placed at the input and output
[image]

Here is about what I would imagine. The connection points are indicated with red, but honestly I don't see the connection points on the top and bottom of choppers used that often. As in the previous images, they indicate with which signal the signals are modulated. Unfortunately, I'm not aware of any formal definition of the symbols used as they are quite rarely used and rarely found in any library. 
[image]

I would be willing to pay for the creation of these symbols as I require them for my master thesis.

Rmano, 2024-11-05:
The problem I see is that these symbols can't be of fixed size: in your examples, the vertical ones are different from the horizontal ones. There is no mechanism to stretch or shrink devices based on the position of anchors, so it will be at the end a matter of guessing. 

I think that a well-thought macro could do the job better than a component. I am in a very bad moment now, with little time on my hands, but I'll try to see if I can suggest something.

PJansky, 2024-11-05:
No worries, I'm currently in no rush and could find a different way to create the drawings requiring these symbols if needs be.

Rmano, 2024-11-09:
One solution could be a macro like this. You need to specify the center point (you can use all the methods Ti*k*Z has to build it) and the two connections of the first port (P2A and P2B here) and the thing builds itself, adding coordinates for the connection points: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\NewDocumentCommand{\ChopperByTerminals}{O{} m m m m}{%
    % #1 -> up extension
    % #2 -> center
    % #3 -> term 1
    % #4 -> term 2
    % #5 -> name
    \coordinate(#5-C) at (#2);
    \coordinate(#5-P1A) at (#3);
    \coordinate(#5-P1B) at (#4);
    \coordinate(#5-P2B) at ($(#5-P1A)!2!(#5-C)$);
    \coordinate(#5-P2A) at ($(#5-P1B)!2!(#5-C)$);
    \coordinate(#5-T1A) at ($(#5-P1B)!1.2!(#5-P1A)$);
    \coordinate(#5-T1B) at ($(#5-P1A)!1.2!(#5-P1B)$);
    \coordinate(#5-T2A) at ($(#5-P2B)!1.2!(#5-P2A)$);
    \coordinate(#5-T2B) at ($(#5-P2A)!1.2!(#5-P2B)$);
    \coordinate(#5-CKA) at ($(#5-T1A)!0.5!(#5-T2A)$);
    \coordinate(#5-CKB) at ($(#5-T1B)!0.5!(#5-T2B)$);
    \draw [draw, thick, fill=white, #1] (#5-T1A) rectangle (#5-T2B);
    \draw [draw, thick, line join=round, #1] (#5-P1A) -- (#5-P2B) -- (#5-P1B) -- (#5-P2A) -- (#5-P1A);
}
\begin{document}
\begin{tikzpicture}[]
    \node[op amp](A) at (0,0) {};
    \ChopperByTerminals[draw=red]{[xshift=-0.5cm]A.leftedge}{A.-}{A.+}{chopper in}
    \draw [red, <-] (chopper in-CKA) -- ++(0,.5) node[above]{$f_c$};
    \ChopperByTerminals[draw=red]{[xshift=1.2cm]A.leftedge}{[xshift=1.7cm]A.-}{[xshift=1.7cm]A.+}{chopper out}
    \draw [red, <-] (chopper out-CKA) -- ++(0,.5) node[above]{$f_c$};
    %Try another one
    \draw (4,1) node[npn](Q1){} (7,1) node[npn, xscale=-1](Q2){};
    \coordinate (Qcenter) at ($(Q1.E)!0.5!(Q2.E)$);
    \ChopperByTerminals[]{[yshift=-0.25cm]Qcenter}{Q1.E}{Q2.E}{cE}
    \draw [<-] (cE-CKA) -- ++(-0.5,0) node[left]{$f_c$};
    % pseudo-anchors
    \ChopperByTerminals[draw=blue]{5.5,-1.5}{4,-1.25}{7,-1.25}{C}
    \foreach \a in {P1A, P1B, P2A, P2B, CKA, CKB}
        \node[circle, draw, inner sep=1pt, pin=60:\a] at (C-\a) {};
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2024-11-10:
One solution could be a macro like this. You need to specify the center point (you can use all the methods Ti*k*Z has to build it) and the two connections of the first port (P2A and P2B here) and the thing builds itself, adding coordinates for the connection points: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\NewDocumentCommand{\ChopperByTerminals}{O{} O{0.2cm} m m m}{%
    % #1 -> options to the draw command
    % #2 -> half width
    % by default is drawn to the right (if vertical), to the left if #2 is negative
    % If horizontal, by default is above, below if #2 is negative
    % #3 -> term 1
    % #4 -> term 2
    % #5 -> name for the "anchors"
    \coordinate(#5-P1A) at (#3);
    \coordinate(#5-P1B) at (#4);
    % the center is midway P1A and P1B, distance #2 at 90°
    \coordinate(#5-C1) at ($(#5-P1A)!0.5!(#5-P1B)$);
    \coordinate(#5-C) at ($(#5-C1)!#2!90:(#5-P1B)$);
    \coordinate(#5-P2B) at ($(#5-P1A)!2!(#5-C)$);
    \coordinate(#5-P2A) at ($(#5-P1B)!2!(#5-C)$);
    \coordinate(#5-T1A) at ($(#5-P1B)!1.2!(#5-P1A)$);
    \coordinate(#5-T1B) at ($(#5-P1A)!1.2!(#5-P1B)$);
    \coordinate(#5-T2A) at ($(#5-P2B)!1.2!(#5-P2A)$);
    \coordinate(#5-T2B) at ($(#5-P2A)!1.2!(#5-P2B)$);
    \coordinate(#5-CKA) at ($(#5-T1A)!0.5!(#5-T2A)$);
    \coordinate(#5-CKB) at ($(#5-T1B)!0.5!(#5-T2B)$);
    \coordinate(#5-C2) at ($(#5-P2A)!0.5!(#5-P2B)$);
    \draw [draw, thick, fill=white, #1] (#5-T1A) rectangle (#5-T2B);
    \draw [draw, thick, line join=round, #1] (#5-P1A) -- (#5-P2B) -- (#5-P1B) -- (#5-P2A) -- (#5-P1A);
}
\begin{document}
\begin{tikzpicture}[]
    \node[op amp](A) at (0,0) {};
    \ChopperByTerminals[draw=red][-0.2cm]{A.-}{A.+}{chopper in}
    \draw [red, <-] (chopper in-CKA) -- ++(0,.5) node[above]{$f_c$};
    \ChopperByTerminals[draw=red]{[xshift=1.2cm]A.-}{[xshift=1.2cm]A.+}{chopper out}
    \draw [red, <-] (chopper out-CKA) -- ++(0,.5) node[above]{$f_c$};
    %Try another one
    \draw (4,1) node[npn](Q1){} (7,1) node[npn, xscale=-1](Q2){};
    \coordinate (Qcenter) at ($(Q1.E)!0.5!(Q2.E)$);
    \ChopperByTerminals[][-0.2cm]{Q1.E}{Q2.E}{cE}
    \draw [<-] (cE-CKA) -- ++(-0.5,0) node[left]{$f_c$};
    % pseudo-anchors
    \ChopperByTerminals[draw=blue][-1cm]{4,-1.25}{7,-1.25}{C}
    \foreach \a in {P1A, P1B, P2A, P2B, CKA, CKB, C, C1, C2}
        \node[circle, draw, inner sep=1pt, pin={[font=\tiny, fill=white]60:C-\a}] at (C-\a) {};
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2024-11-10:
Hi! I closed this issue with the addition of documentation; if you need more information, please feel free to reopen it!

closed 2024-11-10 by Rmano

PJansky, 2024-11-17:
Thanks a lot! It seems to work well :)

PR #829 (2024-11-10): Adding an example macro (choppers) to the manual
Fixes #828

[image]

[image]
commit 93c88007 (2024-11-10): Adding an example macro (choppers) to the manual

Fixes #828