circuitikz/circuitikz #830: [Bug] muxdemux label not working

FelixSelter, 2024-11-14:
**Describe the bug**
If i try to add labels to my muxdemux pins, I get an error that muxdemux labels is undefined.
`Package pgfkeys: I do not know the key '/tikz/muxdemux label', to which you passed 'L1=left1, L2=left2, R1=r1, B2=b2, T1=t1, T2=t2', and I am going to ignore it. Perhaps you misspelled it.`

**To Reproduce**
This an example from the docs 4.24.4.1

```latex
\documentclass{article}
\usepackage{circuitikz}

\tikzset{myICwl/.style={muxdemux,
            muxdemux def={Lh=4, Rh=4, w=4,
                    NR=1, NL=2, NB=2, NT=2,},
            muxdemux label={L1=left1, L2=left2,
                    R1=r1, B2=b2, T1=t1, T2=t2},
        }
}

\begin{document}

Circuitikz: \pgfcircversion{} (\pgfcircversiondate)\par
Ti\emph{k}Z: \pgfversion{} (\pgfversiondate)\par\bigskip

\begin{tikzpicture}[]
    \node[myICwl]{};
\end{tikzpicture}
\end{document}
```
**Screenshots**
[image]

**Operating system, LaTeX distribution**
I am using ubuntu
```
pdfTeX 3.141592653-2.6-1.40.25 (TeX Live 2023)
kpathsea version 6.3.5
Copyright 2023 Han The Thanh (pdfTeX) et al.
There is NO warranty.  Redistribution of this software is
covered by the terms of both the pdfTeX copyright and
the Lesser GNU General Public License.
For more information about these matters, see the file
named COPYING and the pdfTeX source.
Primary author of pdfTeX: Han The Thanh (pdfTeX) et al.
Compiled with libpng 1.6.39; using libpng 1.6.39
Compiled with zlib 1.2.13; using zlib 1.2.13
Compiled with xpdf version 4.04
```

**Expected behavior**
I expected my pins to be labeled like in the docs

Rmano, 2024-11-14:
Can you please compile this snippet and give the results? 

```latex
\documentclass[border=5pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{pgfplots}
\pgfplotsset{compat=newest}
\begin{document}

\today{}:
\begin{tabular}{lll}
    Tikz:& \pgfversion (\pgfrevisiondate)& \pgfversiondate\\
    Pgfplots:& \pgfplotsversion & \pgfplotsversiondate \\
    Circuitikz:& \pgfcircversion & \pgfcircversiondate\\
\end{tabular}

\end{document}
```

Rmano, 2024-11-14:
Hmmm,  Something pesky here; your example works ok on my Ubuntu machine with TeXLive.

Can you run `kpsewhich pgfcircquadpoles.tex` and report the output?

FelixSelter, 2024-11-14:
Sure:

```
kpsewhich pgfcircmultipoles.tex
/usr/local/texlive/2023/texmf-dist/tex/generic/circuitikz/pgfcircmultipoles.tex
```

But I have on line 1172: `\ctikzset{multipoles/flipflop/t3/.initial={}}` and `\tikzset{muxdemux label/.code=\pgfqkeys{\circuitikzbasekey/multipoles/muxdemux/label}{#1}}` can nowhere be found in the document.

I can confirm that the line is present in the zip file that i downloaded from ctan. There must have been an issue copying the files over. I try again and then report. I opened the folder with sudo, deleted everything inside it and copied the contents from the zip over. I cannot imagine where something could have gone wrong

FelixSelter, 2024-11-14:
Oh I see now. I replaced the files inside `/usr/local/texlive/2023/texmf-dist/tex/latex/circuitikz` but not the ones in `/usr/local/texlive/2023/texmf-dist/tex/generic/circuitikz`. Everything works fine now and I am really sorry for my mistake

Rmano, 2024-11-14:
Notice that it is *much* better not to touch your LaTeX distribution from Ubuntu. You should create a `~/texmf` directory in your home, and then unzip there all the content of the `.tds.zip` file you get from CTAN or from here. That way, your new version will not be overwritten when ubuntu decide to make a minor update...

See https://tex.stackexchange.com/questions/73016/how-do-i-install-an-individual-package-on-a-linux-system

FelixSelter, 2024-11-14:
> Notice that it is _much_ better not to touch your LaTeX distribution from Ubuntu. You should create a `~/texmf` directory in your home, and then unzip there all the content of the `.tds.zip` file [...]

Thank you for the great tip.