circuitikz/circuitikz #833: [Bug] Inconsistent line widths of barrier components

schtandard, 2024-12-10:
**Describe the bug**
The `barrier` component includes a line along the bipole axis in the same width as the junction, i.e. thicker than the normal line width used for connections. This looks really weird; the barrier should be only the cross. It is also inconsistent with the `openbarrier` component when its `gap` is set to 0.

[image]

I believe that the `barrier` should look just like an `openbarrier` with no `gap`, i.e. the bold stroke along the wire should be removed.

**To Reproduce**
The image above can be produced with the following MWE

```latex
\documentclass{article}

\usepackage{circuitikz}

\begin{document}

\begin{circuitikz}
  \draw (0,0)
    to [barrier] ++(1,0)
    to [openbarrier, bipoles/openbarrier/gap=0] ++(1,0);
\end{circuitikz}

\end{document}
```

**Operating system, LaTeX distribution**

I am on Windows with TeX Live 2024. I have Ti*k*Z verion 3.1.10 (2023-01-15) and Circuitikz version 1.7.0 (2024/08/03).

```
>pdflatex --version
pdfTeX 3.141592653-2.6-1.40.26 (TeX Live 2024)
kpathsea version 6.4.0
Copyright 2024 Han The Thanh (pdfTeX) et al.
There is NO warranty.  Redistribution of this software is
covered by the terms of both the pdfTeX copyright and
the Lesser GNU General Public License.
For more information about these matters, see the file
named COPYING and the pdfTeX source.
Primary author of pdfTeX: Han The Thanh (pdfTeX) et al.
Compiled with libpng 1.6.43; using libpng 1.6.43
Compiled with zlib 1.3.1; using zlib 1.3.1
Compiled with xpdf version 4.04
```

**Expected behavior**
The `barrier` should look just like an `openbarrier` with no `gap`, i.e. the bold stroke along the wire should be removed.

schtandard, 2024-12-10:
(I'm happy to make a PR if this change is accepted.)

Rmano, 2024-12-10:
Oh, yes, I think you are right. A PR would be welcome (remember to add a CHANGELOG.md  entry with your attribution!).

PR #835 (2024-12-16): Fix barrier shapes
The first commit closes #833 by drawing the barrier wire with normal linewidth instead of the bolder component linewidth.

However, there really shouldn't be any wire at all, just as a capacitor symbol doesn't contain any lead wires. The following MWE shows two problems arising due to that wire: The first tries to use barriers on circuit segments shorter than the component length (but not shorter than the cross, which is the intuitive length of the component). The second uses the barrier node shapes and tries to place curved wires.
```latex
\documentclass{article}

\usepacka [... cut]
closed 2024-12-16 by Rmano