circuitikz/circuitikz #836: [New Component] German TVS diode

myzinsky, 2024-12-25:
[image]

[image]

German TVS diodes have different shapes. It would be nice if both could be implemented.

I tried it myself but diodes seem a bit complex in the code base ...

Rmano, 2024-12-25:
What about a (global) option? So that the TVS diode switches between "45-degree whiskers" to "90-degree whiskers"?

myzinsky, 2024-12-26:
Sounds reasonable

Rmano, 2024-12-29:
@myzinsky, the single (one-way) TVS is really a Zener diode, correct?

myzinsky, 2024-12-29:
They told me that would be the German TVS ^^

myzinsky, 2024-12-29:
https://de.wikipedia.org/wiki/Suppressordiode

Rmano, 2024-12-29:
They seem fast zener diodes. My doubt is whether to add a flag that changes the aspect of both ZZener and TVS or create new shapes. But I think I'll go for the flag...

Rmano, 2024-12-29:
[image]

@myzinsky what do you think

PR #837 (2024-12-29): Add a flag for German-type TVS
Moreover, fix footnotes and stroke-tvs
Fixes #836

[image]
commit 86cb0d3a (2024-12-29): Add a flag for German-type TVS

Moreover, fix footnotes and stroke-tvs
Fixes #836
closed 2024-12-29 by Rmano