circuitikz/circuitikz #852: [New Component] Autotransfomer

m-borjesson, 2025-03-09:
To complete the suite of transformers (currently oosourcetrans and ooosource) it would be usefull to also have the autotransformer (with the centerpoint for ground and also the wye, delta and zig connection as in oosourcetrans and ooosource).

**Show the component and its usage**
[image]
[image]

**Add references to standards, usage, etc.**
https://www.iso.org/obp/ui#iec:grs:60417:5941
https://www.iso.org/obp/ui#iec:grs:60417:6018

Rmano, 2025-03-09:
This is interesting. I'll have a look to see if it's possible to add it in a coherent way with the ootrans &co things.

Rmano, 2025-03-11:
I am not sure if I should maintain the size equal  to the `oosourcetrans` or I should make it smaller in the horizontal direction, and so more similar to the one you posted...

Rmano, 2025-03-11:
Forget that: this 

[image]

is better looking and much more similar to the ISO one. What do you think?

m-borjesson, 2025-03-11:
Looks great.

Rmano, 2025-03-11:
Ok. I'll add it as soon as I manage to solve a bit of problems/incoherencies I found while doing it.

PR #856 (2025-03-11): Add oo-style auto transformer
Suggested by @m-borjesson
Closes #852

[image]

[image]
closed 2025-03-11 by Rmano