circuitikz/circuitikz #865: Undefined control sequence I.691

cemiller4, 2025-05-08:
I'm a beginner user to Latex, MiKTex, and Circuitikz package.  I keep getting error for undefined control sequence.

**example:**
\documentclass[11pt, a4paper]{book}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
\tikz \draw (0,0) to[R=$R_1$] (2,0);
\end{circuitikz}
\end{document}   

**results in:**  
\AppData\Local\Programs\MiKTeX\tex/generic/circuitikz\pgfcircsha
pes.tex
! Undefined control sequence.
l.691 \pgfcirc@activate@bipole@simple
                                     {l}{iecconn}

Additional info:  Several years ago I was able to get my MiKTex setup to work with Circuitikz examples I found on web.  I suspect they are no longer working either from MiKTex or computer upgrade.  I may have been using Linux setup at the time but can't remember.

Rmano, 2025-05-08:
To have an error on that madro you should have a very old, or broken, installation. Try to update MikTeX both in user and admin mode. Anyway, try to compile the following code and report the versions (it's in the issue template...)

```latex
\documentclass{article}
\usepackage{circuitikz}
\begin{document}

Circuitikz: \pgfcircversion{} (\pgfcircversiondate)\par
Ti\emph{k}Z: \pgfversion{} (\pgfversiondate)\par\bigskip

\begin{tikzpicture}[]
    \draw (0,0) to[R] ++(2,0);
\end{tikzpicture}
\end{document}
```

cemiller4, 2025-05-09:
Thank you for helping.  But I'm still not having any luck.    
1. I uninstalled and reinstalled using basic-miktex-24.1-x64.exe
2. updated miktex for local only 
3. updated miktex for all users
4. tried ccleaner to eliminate registry problems
5. ran your code several times in each of the above situations
still get same 'undefined control sequence'
---------------
tried to find the version but I'm confused about were 'issue template' is.  I did find this version in log file:
Package: circuitikz 2020/02/11{} The CircuiTikz circuit drawing package version
 1.0.1
---------
Here is my MikTex package version for Circuitikz:
[image]

Rmano, 2025-05-09:
Ok. So you have a stray, ancient version of some of the circuitikz files in your personal directories, because the MikTeX data is correct. Where did you find the file saying version 2020/02/11? Try to add `\listfiles` to your latex file and paste the list that will be generated in the log file.

closed 2025-05-09 by cemiller4

Rmano, 2025-05-09:
Yes, clearly so. You should also remove all possible files in the format `pgfcirc-something.tex`, and surely the `circuitikzmanual.pdf` which is ancient too.